fwupd/fwupd #2197: fwupdmgr file read error on RTL810xE PCI Express Fast Ethernet controller

BillDietrich, 2020-06-21:
**Describe the bug**
Alarming error message when probably it should say "operation not supported by this device".

**Steps to Reproduce**
sudo fwupdmgr verify [DEVICEID]

gives

failed to verify RTL810xE PCI Express Fast Ethernet controller: failed to read firmware: Error reading from file: Input/output error

Is there some file-permission error in the system, or the fwupd definition for this device is wrong, or something ? The device is not updatable according to fwupdmgr; maybe that is causing the error ?

**Expected behavior**
I'm told this device doesn't support this operation.  How about changing the error message to something like "this device doesn't support this operation" ?

**fwupd version information**
Please provide the version of the daemon and client.
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed by default in Ubuntu 20.04 GNOME.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

Inspiron N5010
│
├─Core Processor Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     ilm19941.d15
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               70eb0a46-3f89-595b-81a2-01851eef9efd
│                          b36a8fab-195a-5871-be0b-9aefd1837d6c
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─RTL810xE PCI Express Fast Ethernet controller:
│     Device ID:           a1d3f119b7f41e1f6fb1cc96e27da2d329911cbd
│     Current version:     02
│     Vendor:              Realtek Semiconductor Co., Ltd. (PCI:0x10EC)
│     GUIDs:               4cd761f5-a33a-5231-b70a-00f3a78beaf7
│                          22ea45ed-4dd3-5754-b6f5-20e7a67250a5
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─WDC WD3200BEVT-75ZCT2:
      Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
      Summary:             ATA Drive
      Current version:     11.01A11
      GUIDs:               c520b79c-b72b-5ab4-a454-e0afd1994efe
                           c73442c6-d2d3-5613-95e5-db8f8b41a310
                           fb0ac991-48f7-5756-a564-23dfaf9106e9
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

**Additional questions**
- Operating system and version:  Ubuntu 20.04 GNOME
- Have you tried rebooting?  Yes
- Is this a regression?  Don't know

hughsie, 2020-06-21:
>  How about changing the error message to something like "this device doesn't support this operation"

Well, the kernel is saying "here's a file you can read to get the firmware" and then it's returning with "Input/output error" when we actually try to read the firmware.

BillDietrich, 2020-06-21:
Can you tell if it's a permission error (I should report to Ubuntu) or a driver error (I should file to kernel/driver) ?

hughsie, 2020-06-21:
It might be a result of kernel lockdown. I'll as the kernel developers and see if they know. @superm1 any ideas?

superm1, 2020-06-21:
If it was a permissions error related to lockdown you would see something in `dmesg`, and I would also expect the error to be `-EPERM`

BillDietrich, 2020-06-21:
It turns out that I do get this in dmesg:

```
r8169 0000:13:00.0: Invalid PCI ROM header signature: expecting 0xaa55, got 0x0000
```

superm1, 2020-06-21:
So I feel fwupd is showing the correct messaging for this circumstance from what it got from the kernel.

BillDietrich, 2020-06-21:
So this is a driver bug ?  Or a device that does not support reading the firmware ?  Is there an API for the driver to say "operation not supported" instead of returning bogus data ?

superm1, 2020-06-21:
That's something you'll have to bring to the kernel folks to decide.

BillDietrich, 2020-06-21:
Is there an API for the driver to say "operation not supported" instead of returning bogus data ?

superm1, 2020-06-21:
I would _expect_ if it's not supported by the kernel driver, the kernel driver wouldn't offer the sysfs attribute to try to access the data.  But this is something you'll have to take up with the kernel maintainers for that part.

superm1, 2020-06-22:
if the kernel people say that there is something else uesrland needs to be doing before accessing ROM, re-open this.  but for now this is in camp of kernel bug.  it shouldn't export the attribute unless it works.

closed 2020-06-22 by superm1

BillDietrich, 2020-06-23:
Okay, what is the name of the attribute being exported ?  Thanks.

superm1, 2020-06-23:
`"rom"` https://github.com/fwupd/fwupd/blob/f50ff2c27e37a2914703de09a39e1fa5a8296a7b/plugins/optionrom/fu-optionrom-device.c#L51