fwupd/fwupd #2198: System BIOS is not listed as a device

BillDietrich, 2020-06-21:
**Describe the bug**
System (motherboard) BIOS is not listed as a device, so no way to verify or update it.

**Steps to Reproduce**
On my Dell Inspiron N5010, doing

fwupdmgr get-devices --show-all-devices

shows 3 devices: integrated GPU, disk drive, and network controller.
It does not show the motherboard / system as a device that contains firmware.

**Expected behavior**
Show motherboard / system as a device, and give info about firmware in it.

The system BIOS does show up in output of "dmidecode", as in for example "sudo dmidecode -s bios-version".

**fwupd version information**
Please provide the version of the daemon and client.
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed by default in Ubuntu 20.04 GNOME.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

Inspiron N5010
│
├─Core Processor Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     ilm19941.d15
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               70eb0a46-3f89-595b-81a2-01851eef9efd
│                          b36a8fab-195a-5871-be0b-9aefd1837d6c
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─RTL810xE PCI Express Fast Ethernet controller:
│     Device ID:           a1d3f119b7f41e1f6fb1cc96e27da2d329911cbd
│     Current version:     02
│     Vendor:              Realtek Semiconductor Co., Ltd. (PCI:0x10EC)
│     GUIDs:               4cd761f5-a33a-5231-b70a-00f3a78beaf7
│                          22ea45ed-4dd3-5754-b6f5-20e7a67250a5
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─WDC WD3200BEVT-75ZCT2:
      Device ID:           602b0a6cc821d155208724f0e22f8d111542b74c
      Summary:             ATA Drive
      Current version:     11.01A11
      GUIDs:               c520b79c-b72b-5ab4-a454-e0afd1994efe
                           c73442c6-d2d3-5613-95e5-db8f8b41a310
                           fb0ac991-48f7-5756-a564-23dfaf9106e9
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

**Additional questions**
- Operating system and version:  Ubuntu 20.04 GNOME
- Have you tried rebooting?  Yes
- Is this a regression?  Don't know

hughsie, 2020-06-21:
Are you booted in UEFI mode, or compat BIOS mode?

BillDietrich, 2020-06-21:
Legacy BIOS.

hughsie, 2020-06-21:
> Legacy BIOS.

So that's expected, we can only flash system firmware using the UEFI runtime services (UpdateCapsule). If you re-install / change to UEFI boot you should have system devices appear. There's nothing we can do to fix this in legacy mode I'm afraid.

closed 2020-06-21 by hughsie

BillDietrich, 2020-06-21:
Somehow dmidecode is able to read the entire contents of the BIOS firmware, and the metadata (such as version number) about it.  So it seems fwupd should at least be able to do a verify, if the vendor supplied an image in the database.

Is fwupd limited to UEFI ?  I don't see that in the man page for fwupdmgr.

It seems to me that the motherboard BIOS should always be listed as a device, even if the description just says "not UEFI so we can't do anything with it".

hughsie, 2020-06-21:
> dmidecode is able to read the entire contents of the BIOS firmware

It really can't.

> Is fwupd limited to UEFI

No, the UEFI firmware update plugin is limited to UEFI systems. The other plugins (e.g. nvme, ata, logitech, synaptics etc) work fine.

BillDietrich, 2020-06-21:
Re: dmidecode:

Well, at least dmidecode shows BIOS metadata.  On my system:
```
$ sudo dmidecode -t bios
[sudo] password for user1: 
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 2.6 present.

Handle 0x001C, DMI type 13, 22 bytes
BIOS Language Information
	Language Description Format: Long
	Installable Languages: 1
		en|US|iso8859-1
	Currently Installed Language: en|US|iso8859-1

Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Dell Inc.
	Version: A15
	Release Date: 07/19/2011
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 2048 kB
	Characteristics:
		MCA is supported
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		ESCD support is available
		Boot from CD is supported
		Selectable boot is supported
		BIOS ROM is socketed
		EDD is supported
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		CGA/mono video services are supported (int 10h)
		ACPI is supported
		USB legacy is supported
		ATAPI Zip drive boot is supported
		BIOS boot specification is supported
		Targeted content distribution is supported
	BIOS Revision: 1.5
	Firmware Revision: 1.5
```

Maybe you'd then have to read 64KB from /dev/mem at address 0xF0000 to get the actual contents of the firmware.

BillDietrich, 2020-06-21:
Re: UEFI firmware update plugin:

So, is the issue that the software lacks a "legacy BIOS firmware update plugin" ?  Should I file a request to create that ?  Then fwupdmgr would show a BIOS firmware device and metadata, even if it can't update it ?

superm1, 2020-06-21:
> Re: UEFI firmware update plugin:
> 
> So, is the issue that the software lacks a "legacy BIOS firmware update plugin" ? Should I file a request to create that ? Then fwupdmgr would show a BIOS firmware device and metadata, even if it can't update it ?

You have to understand the architecture of how this whole suite works expects that the EFI system resource table is populated, and the firmware supports the UEFI runtime services associated with capsules.  If the firmware doesn't provide it, there is no open source friendly way to flash the firmware.

FWIW _new_ Dell systems sold these days **can't be installed** in legacy mode.  You can boot a USB disk in legacy mode, but that's it.  So you can understand how it is a total waste of effort for the fwupd project to reverse engineer legacy BIOS update on Dell systems.  For your specific case, I would recommend you try to download the BIOS from support.dell.com, save it to a FAT32 formatted USB disk and try to flash it from the F12 BIOS POST screen.  I don't know for sure a model as old as yours supports that, but you can at least try.

BillDietrich, 2020-06-21:
My Dell system is from 2011.

I think fwupdmgr should show legacy BIOS as a device but mark it "not upgradable through fwupdmgr" or something.  I accept that there is no way to flash the firmware if there is not EFI.

If you're unwilling to do that, then at least the fwupdmgr man page should say "EFI-only" or "non-EFI motherboard not shown".  When the system is EFI, does fwupdmgr list the motherboard as a device ?  I think it should.

Yes, I have upgraded the firmware a couple of times using FreeDOS or similar.  The issue here is making fwupdmgr better, not other ways to see/flash the firmware, and also not just my machine.

superm1, 2020-06-22:
Added this, you should end up with a read only view as a result: https://github.com/fwupd/fwupd/commit/91e27e145a1f7cb97ec78875d08f1b9a9496dadf

BillDietrich, 2020-06-23:
Okay, thanks very much !