fwupd/fwupd #2200: fwupd not working to update Dell DW5821e

wicadmin, 2020-06-21:
**Describe the bug**
fwupdmgr not finding updates for my LTE modem (Dell dw5821e). Tried Ubuntu snap 1.4.4 and Fedora 32 1.4.4

**Steps to Reproduce**
fwupdmgr get-updates --allow-reinstall
No updatable devices

**Expected behavior**
fwupdmgr to reapply the firmware from LVFS.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt (Ubuntu) 
yum (Fedora)

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
Standard PC (i440FX + PIIX, 1996)
│
├─Intel Core Processor (Haswell, no TSX, IBRS):
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x1
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│     Device Flags:        • Internal device
│   
├─QXL paravirtual graphic card (QEMU Virtual Machine):
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     04
│     Vendor:              Red Hat, Inc. (PCI:0x1B36)
│     GUIDs:               6e76e104-5378-5cf7-8d89-9abcd551626e ← PCI\VEN_1B36&DEV_0100&REV_04
│                          48108f63-5e54-5627-b89c-73a36a9c49ae ← PCI\VEN_1B36&DEV_0100
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─Virtio network device:
      Device ID:           8476ae9bb1ff458ade59e6e1bca9e4cfd101e26d
      Vendor:              Red Hat, Inc. (PCI:0x1AF4)
      GUIDs:               62deba3e-b70b-51b5-b66a-919b6727f5cf ← PCI\VEN_1AF4&DEV_1000&REV_00
                           21c85fac-5270-576f-a84e-04969f8cf75a ← PCI\VEN_1AF4&DEV_1000
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available
    
[cloud@cloud-00 ~]$ lsusb
Bus 001 Device 005: ID 413c:81d7 Dell Computer Corp. DW5821e Snapdragon X20 LTE
Bus 001 Device 002: ID 0627:0001 Adomax Technology Co., Ltd QEMU USB Tablet
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

```

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2020-06-21:
That's s virtual machine...

wicadmin, 2020-06-21:
> That's s virtual machine...

Yes, but the exact same issue on a non virtual Ubuntu 18.04 with snap 1.4.1. I tried Fedora 32 just in case I saw something different, but exact same issue.

superm1, 2020-06-21:
I think you'll need to share the daemon log in verbose mode so we can see what the modem manager plugin is doing.  Please with 1.4.4.

wicadmin, 2020-06-21:
> I think you'll need to share the daemon log in verbose mode so we can see what the modem manager plugin is doing. Please with 1.4.4.

Sure...what are the steps for this? I will do it on the Ubuntu snap 1.4.4 as it is native and avoid passing the card to kvm Fedora.

wicadmin, 2020-06-21:
```
(fwupdmgr:875): FuMain-DEBUG: 16:38:54.314: DW5821e Snapdragon X20 LTE
  DeviceId:             fa707b9af86ff44bc17316b6c3e5ea82aab3ce86
  Guid:                 64da2d58-8d1b-5e5b-b793-f88ba5a25a8f <- USB\VID_413C&PID_81D7&REV_0318
  Guid:                 761d6124-0002-5185-b767-9adf67bf1a5e <- USB\VID_413C&PID_81D7
  Guid:                 795e079d-093b-5503-aa59-35b832480e95 <- USB\VID_413C
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Flags:                registered|use-runtime-version
  Vendor:               Dell Inc.
  Version:              T77W968.F1.0.0.4.2.GC.010
  VersionFormat:        plain
  Icon:                 network-modem
  Created:              2020-06-21

```

superm1, 2020-06-21:
`fwupdmgr modify-config VerboseDomains *`
and then reproduce and share the journal
`journalctl -u fwupd.service -b`

https://github.com/fwupd/fwupd/wiki/Getting-involved

superm1, 2020-06-21:
Oh, I notice in there:

```│     Update Error:        No vendor ID set```

That might be the root of the problem.

wicadmin, 2020-06-21:
> Oh, I notice in there:
> 
> `│ Update Error: No vendor ID set`
> 
> That might be the root of the problem.

Yes, that's what I'm suspecting also. Aleksander also thinks it is the same. Was there something changed with fwupd that caused this?

superm1, 2020-06-21:
This commit: 
https://github.com/fwupd/fwupd/commit/d8ea8da902493638bbe776ca854c1c17ca0fbcee

wicadmin, 2020-06-21:
> This commit:
> [d8ea8da](https://github.com/fwupd/fwupd/commit/d8ea8da902493638bbe776ca854c1c17ca0fbcee)

Is there a way to workaround this for now?

wicadmin, 2020-06-21:
> `fwupdmgr modify-config VerboseDomains *`

fwupdmgr modify-config VerboseDomains *
Invalid arguments: KEY VALUE expected

superm1, 2020-06-21:
Oh darn, I forgot that doesn't work with snap because of something funny snap does in argument handling.  There is a config file in `/root/snap/fwupd/` somewhere called daemon.conf.  You can change the key manually there and then restart the daemon.

`sudo systemctl stop snap.fwupd.fwupd`

wicadmin, 2020-06-21:
OK, I sudo to root and set the domain modem_manager to verbose without the error. However, nothing interesting from the journal after doing the get-devices.

```
Jun 16 07:59:37  systemd[1]: Starting Firmware update daemon...
Jun 16 07:59:37  systemd[1]: Started Firmware update daemon.
Jun 17 08:03:07  systemd[1]: Starting Firmware update daemon...
Jun 17 08:03:08  systemd[1]: Started Firmware update daemon.
Jun 18 04:10:38  systemd[1]: Starting Firmware update daemon...
Jun 18 04:10:38  systemd[1]: Started Firmware update daemon.
Jun 18 08:02:43  systemd[1]: Starting Firmware update daemon...
Jun 18 08:02:44  systemd[1]: Started Firmware update daemon.
Jun 18 09:05:33  systemd[1]: Stopping Firmware update daemon...
Jun 18 09:05:33  systemd[1]: Stopped Firmware update daemon.
Jun 20 20:54:19  systemd[1]: Starting Firmware update daemon...
Jun 20 20:54:20  systemd[1]: Started Firmware update daemon.
Jun 20 20:54:40  systemd[1]: Stopping Firmware update daemon...
Jun 20 20:54:40  systemd[1]: Stopped Firmware update daemon.

```

wicadmin, 2020-06-21:
> Oh darn, I forgot that doesn't work with snap because of something funny snap does in argument handling. There is a config file in `/root/snap/fwupd/` somewhere called daemon.conf. You can change the key manually there and then restart the daemon.
> 
> `sudo systemctl stop snap.fwupd.fwupd`

Same journal output as above.

superm1, 2020-06-21:
Looking at the code I think this is something we forgot.  The vendor is supposed to be set from `mm_modem_get_manufacturer`, but nothing sets `vendor_id`. @hughsie what should we set here?

aleksander0m, 2020-06-22:
What is the expected `vendor_id`? Is it the USB vid or some other thing?

wicadmin, 2020-06-22:
Hi...sorry for the impatience, but while this is being fixed and release, is there a work around I can use in the meantime to either ignore this check or force in somewhere what it is looking for?

Thank you.

superm1, 2020-06-22:
I think for now adding `VendorId = USB:0x413C` into the quirks file calling out the device.

superm1, 2020-06-22:
Like this:
```
diff --git a/plugins/modem-manager/modem-manager.quirk b/plugins/modem-manager/modem-manager.quirk
index b110d7c6..6dacb9fc 100644
--- a/plugins/modem-manager/modem-manager.quirk
+++ b/plugins/modem-manager/modem-manager.quirk
@@ -3,21 +3,25 @@
 [DeviceInstanceId=USB\VID_413C&PID_81D7]
 Summary = Dell DW5821e LTE modem
 CounterpartGuid = USB\VID_413C&PID_81D6
+VendorId = USB:0x413C

 # DW5821e in fastboot mode
 [DeviceInstanceId=USB\VID_413C&PID_81D6]
 Summary = Dell DW5821e LTE modem (fastboot)
 CounterpartGuid = USB\VID_413C&PID_81D7
+VendorId = USB:0x413C

 # DW5821e/eSIM
 [DeviceInstanceId=USB\VID_413C&PID_81E0]
 Summary = Dell DW5821e/eSIM LTE modem
 CounterpartGuid = USB\VID_413C&PID_81E1
+VendorId = USB:0x413C

 # DW5821e/eSIM in fastboot mode
 [DeviceInstanceId=USB\VID_413C&PID_81E1]
 Summary = Dell DW5821e/eSIM LTE modem (fastboot)
 CounterpartGuid = USB\VID_413C&PID_81E0
+VendorId = USB:0x413C

 # T77W968
 [DeviceInstanceId=USB\VID_0489&PID_E0B4]
````

superm1, 2020-06-22:
@aleksander0m who is the vendor for those other modules you added in e66e0b7114a190569e9b57b36372b889a7c975f8?  I guess Foxconn?

wicadmin, 2020-06-22:
Where can I update that file on my snap install. I did this, but the only file with that name is on a read-only file-system:

```
sudo find /snap -name modem-manager.quirk
/snap/fwupd/2164/share/fwupd/quirks.d/modem-manager.quirk
```

superm1, 2020-06-22:
That's correct - it's readonly in snap.  You can't change it.  It seems that we considered this and it will load from localstatedir if you put it there.  Copy that file to the correct place in `/root/snap/fwupd` and it should pick it up.  I don't know off hand the right path, I'd guess `/root/snap/fwupd/current/var/lib/fwupd`.  You'll know it's the right place if it has a `pending.db`

wicadmin, 2020-06-22:
@superm1  - Do I need to recreate the full path (quirks.d etc.)?

superm1, 2020-06-22:
Yes I think so.

wicadmin, 2020-06-22:
> Yes I think so.

I tried it and did `systemctl stop snap.fwupd.fwupd` but still get-devices shows No Vendor ID set.

superm1, 2020-06-22:
Can you run the daemon in verbose, and show the whole log?  It will show in the logs that it actually tried to use it.

wicadmin, 2020-06-22:
hmm...still not seeing much with `journalctl -u fwupd.service -b`

I did set the verbose to *

```
journalctl -u fwupd.service -b
-- Logs begin at Mon 2019-07-15 23:50:10 EDT, end at Mon 2020-06-22 16:23:52 EDT. --
Jun 16 07:59:37  systemd[1]: Starting Firmware update daemon...
Jun 16 07:59:37  systemd[1]: Started Firmware update daemon.
Jun 17 08:03:07  systemd[1]: Starting Firmware update daemon...
Jun 17 08:03:08  systemd[1]: Started Firmware update daemon.
Jun 18 04:10:38  systemd[1]: Starting Firmware update daemon...
Jun 18 04:10:38  systemd[1]: Started Firmware update daemon.
Jun 18 08:02:43  systemd[1]: Starting Firmware update daemon...
Jun 18 08:02:44  systemd[1]: Started Firmware update daemon.
Jun 18 09:05:33  systemd[1]: Stopping Firmware update daemon..
```

superm1, 2020-06-22:
That's not in verbose mode, did you make the right change to your daemon.conf?  It should be super verbose

wicadmin, 2020-06-22:
```
cat /root/snap/fwupd/2164/etc/fwupd/daemon.conf

[fwupd]

# Allow blacklisting specific devices by their GUID
# Uses semicolons as delimiter
BlacklistDevices=

# Allow blacklisting specific plugins
# Uses semicolons as delimiter
BlacklistPlugins=test;invalid

# Maximum archive size that can be loaded in Mb, with 0 for the default
ArchiveSizeMax=0

# Idle time in seconds to shut down the daemon -- note some plugins might
# inhibit the auto-shutdown, for instance thunderbolt.
#
# A value of 0 specifies 'never'
IdleTimeout=7200

# Comma separated list of domains to log in verbose mode
# If unset, no domains
# If set to FuValue, FuValue domain (same as --domain-verbose=FuValue)
# If set to *, all domains (same as --verbose)
VerboseDomains=*

# Update the message of the day (MOTD) on device and metadata changes
UpdateMotd=true

# For some plugins, enumerate only devices supported by metadata
EnumerateAllDevices=false

```

superm1, 2020-06-22:
Is 2164 your latest version?  That looks correct..

wicadmin, 2020-06-22:
yes

```
ls -l /root/snap/fwupd/
total 8
drwxr-xr-x 4 root root 4096 Jun 20 20:56 2164
drwxr-xr-x 3 root root 4096 Feb  9 12:49 common
lrwxrwxrwx 1 root root    4 Jun 18 10:30 current -> 2164
```

superm1, 2020-06-22:
Oh! You're looking at the wrong journal.

Look at `journalctl -u snap.fwupd.fwupd -b`

wicadmin, 2020-06-22:
[log.txt](https://github.com/fwupd/fwupd/files/4815562/log.txt)

superm1, 2020-06-22:
It seems that is loading from `/root/snap/fwupd/2164/var/lib/fwupd/quirks.d/` successfully in one of the boots but was gone later?

wicadmin, 2020-06-22:
[log.txt](https://github.com/fwupd/fwupd/files/4815789/log.txt)

I put the file back in that location.

wicadmin, 2020-06-22:
Noticed this:
`device fa707b9af86ff44bc17316b6c3e5ea82aab3ce86 [DW5821e Snapdragon X20 LTE] does not define an update protocol`

superm1, 2020-06-23:
The missing protocol I put in a PR on #2211

superm1, 2020-06-23:
once that PR goes in, please check with the edge snap.  I'm not convinced that the quirk file is `/root/snap` is overriding the one in the snap.

aleksander0m, 2020-06-23:
@superm1 sorry for the late reply; yes, Foxconn is the vendor for those modules. I find your suggested PR perfectly fine; thanks for that!

wicadmin, 2020-06-23:
So, my modem does have the latest firmware already, however @aleksander0m and I have exhausted options why the two modems I have will not turn on the software radio.

With the fixes in fwupd edge, I see the device now shows up, but I cannot get it to reinstall the firmware even after downloading the CAB file and using the `--allow-reinstall` option.

```
fwupdmgr --verbose get-updates                                    
• ADATA SX6000NP has no available firmware updates
• DW5821e Snapdragon X20 LTE has no available firmware updates
• Samsung SSD 850 EVO 500GB has no available firmware updates
• System Firmware has no available firmware updates
• USB3.0 Hub has no available firmware updates
• WDC WD10EZEX-00ZF5A0 has no available firmware updates

```
```
fwupdmgr get-devices                    
All Series
│
│   
├─DW5821e Snapdragon X20 LTE:
│     Device ID:           fa707b9af86ff44bc17316b6c3e5ea82aab3ce86
│     Summary:             Mobile broadband device
│     Current version:     T77W968.F1.0.0.4.2.GC.010
│     Vendor:              Dell Inc. (USB:0x413c
│     )
│     GUIDs:               64da2d58-8d1b-5e5b-b793-f88ba5a25a8f
│                          761d6124-0002-5185-b767-9adf67bf1a5e
│                          795e079d-093b-5503-aa59-35b832480e95
│     Device Flags:        • Updatable
```
```
fwupdmgr --verbose install --allow-reinstall fa707b9af86ff44bc17316b6c3e5ea82aab3ce86 ~/Downloads/beedb25554099ba1cb06d217b728512f06590468-20190926-dw5821e-T77W968.F1.0.0.4.2.cab
failed to open fa707b9af86ff44bc17316b6c3e5ea82aab3ce8
```

wicadmin, 2020-06-23:
@superm1  - It appears one of the GUID is missing per the log statements above. Not sure if that is an issue.

wicadmin, 2020-06-23:
OK, looks like I had the parameters in the wrong order. But not it is saying to supported devices.

```
fwupdmgr --verbose install ~/Downloads/beedb25554099ba1cb06d217b728512f06590468-20190926-dw5821e-T77W968.F1.0.0.4.2.cab fa707b9af86ff44bc17316b6c3e5ea82aab3ce86
(fwupdmgr:23311): Fwupd-DEBUG: 14:22:29.425: Emitting ::status-changed() [decompressing]
Decompressing…           [            /                          ](fwupdmgr:23311): Fwupd-DEBUG: 14:22:33.002: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
No supported devices found

fwupdmgr --verbose reinstall fa707b9af86ff44bc17316b6c3e5ea82aab3ce86                                                                                             
No releases found
(fwupdmgr:23699): FuMain-DEBUG: 14:24:35.401: No releases found

```

wicadmin, 2020-06-23:
@superm1  @aleksander0m  - The GUIDs seems off?

superm1, 2020-06-23:
So looking at the XML in the archive it seems the generic carrier and carrier default GUIDs aren't getting exported by modem-manager to me.  None of these correspond to the ones you got out of modem manager.  So I would ask is your modem manager on the host side new enough to export them?  If you don't know  @aleksander0m would probably have to answer that.

```
  <provides>
    <!-- USB\VID_413C&PID_81D7&REV_0318&CARRIER_GCF -->
    <firmware type="flashed">c1d92c58-db79-5ad9-9eef-0ca8e5e49562</firmware>
    <!-- USB\VID_413C&PID_81D7&REV_0318&CARRIER_DEFAULT -->
    <firmware type="flashed">1d9644ca-c94b-5309-ab1b-6742250f7f4a</firmware>
  </provides>
```

wicadmin, 2020-06-23:
ModemManger is compiled from very recent sources:
```
mmcli 1.12.10
Copyright (2011 - 2019) Aleksander Morgado

```

wicadmin, 2020-06-23:
None of the GUIDs match for any carrier.

```
fwupdmgr get-details ~/Downloads/beedb25554099ba1cb06d217b728512f06590468-20190926-dw5821e-T77W968.F1.0.0.4.2.cab
Decompressing…           [***************************************]
All Series
│
├─Unknown Device:
│ │   Description:         Updating the firmware on your DW5821e device improves stability and adds new features.
│ │   GUID:                1960a6a8-4782-53f8-913e-e53c579ac7fc
│ │   Device Flags:        • Supported on remote server
│ │ 
│ └─DW5821e (FirstNet):
│       New version:       T77W968.F1.0.0.4.2.A2.020
│       Remote ID:         lvfs
│       Summary:           Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
│       License:           Proprietary
│       Size:              70.8 MB
│       Urgency:           High
│       Vendor:            Dell Inc.
│       Flags:             trusted-payload
│       Description:       Firmware release T77W968.F1.0.0.4.2 packaged for LVFS.
│     
├─Unknown Device:
│ │   Description:         Updating the firmware on your DW5821e device improves stability and adds new features.
│ │   GUID:                ba3c9774-4f24-55c6-b07b-a8f0d363fec2
│ │   Device Flags:        • Supported on remote server
│ │ 
│ └─DW5821e (AT&T):
│       New version:       T77W968.F1.0.0.4.2.AT.020
│       Remote ID:         lvfs
│       Summary:           Firmware for the Dell DW5821e LTE modem (AT&T carrier)
│       License:           Proprietary
│       Size:              70.8 MB
│       Urgency:           High
│       Vendor:            Dell Inc.
│       Flags:             trusted-payload
│       Description:       Firmware release T77W968.F1.0.0.4.2 packaged for LVFS.
│     
├─Unknown Device:
│ │   Description:         Updating the firmware on your DW5821e device improves stability and adds new features.
│ │   GUID:                21b3f782-bdb0-5c67-a75c-c919e84d4551
│ │   Device Flags:        • Supported on remote server
│ │ 
│ └─DW5821e (NTT DoCoMo):
│       New version:       T77W968.F1.0.0.4.2.DO.010
│       Remote ID:         lvfs
│       Summary:           Firmware for the Dell DW5821e LTE modem (NTT DoCoMo carrier)
│       License:           Proprietary
│       Size:              70.8 MB
│       Urgency:           High
│       Vendor:            Dell Inc.
│       Flags:             trusted-payload
│       Description:       Firmware release T77W968.F1.0.0.4.2 packaged for LVFS.
│     
├─Unknown Device:
│ │   Description:         Updating the firmware on your DW5821e device improves stability and adds new features.
│ │   GUIDs:               c1d92c58-db79-5ad9-9eef-0ca8e5e49562
│ │                        1d9644ca-c94b-5309-ab1b-6742250f7f4a
│ │   Device Flags:        • Supported on remote server
│ │ 
│ └─DW5821e:
│       New version:       T77W968.F1.0.0.4.2.GC.010
│       Remote ID:         lvfs
│       Summary:           Firmware for the Dell DW5821e LTE modem (Generic)
│       License:           Proprietary
│       Size:              70.8 MB
│       Urgency:           High
│       Vendor:            Dell Inc.
│       Flags:             trusted-payload
[... 90 lines cut ...]
│       Summary:           Firmware for the Dell DW5821e LTE modem (Vodafone carrier)
│       License:           Proprietary
│       Size:              70.8 MB
│       Urgency:           High
│       Vendor:            Dell Inc.
│       Flags:             trusted-payload
│       Description:       Firmware release T77W968.F1.0.0.4.2 packaged for LVFS.
│     
└─Unknown Device:
  │   Description:         Updating the firmware on your DW5821e device improves stability and adds new features.
  │   GUID:                0db1d91e-b90f-51be-a85c-ed3c788663f5
  │   Device Flags:        • Supported on remote server
  │ 
  └─DW5821e (Verizon):
        New version:       T77W968.F1.0.0.4.2.VZ.011
        Remote ID:         lvfs
        Summary:           Firmware for the Dell DW5821e LTE modem (Verizon carrier)
        License:           Proprietary
        Size:              70.8 MB
        Urgency:           High
        Vendor:            Dell Inc.
        Flags:             trusted-payload
        Description:       Firmware release T77W968.F1.0.0.4.2 packaged for LVFS.

```

superm1, 2020-06-23:
Yeah looking at your earlier view, you don't have any carrier GUIDs at all! 

```
  Guid:                 64da2d58-8d1b-5e5b-b793-f88ba5a25a8f <- USB\VID_413C&PID_81D7&REV_0318
  Guid:                 761d6124-0002-5185-b767-9adf67bf1a5e <- USB\VID_413C&PID_81D7
  Guid:                 795e079d-093b-5503-aa59-35b832480e95 <- USB\VID_413C
```

So if you have a modern MM, I would question if we have an older version of one of the libraries contained in the snap that needs revving?  From the view outside of fwupd, it's all a black box to me..

The stuff that we have in the snap:
* libmm (modem manager library only) 1.10.0
* libmbim 1.18.0
* libqmi 1.23.0
*

wicadmin, 2020-06-23:
@superm1 - Oh....yes, that would be the issue most likely.

```
mmcli 1.12.10
qmicli 1.25.900
mbimcli 1.24.0
```

superm1, 2020-06-23:
Absolutely, please advise what versions of the various tags to move to that are stable/safe and I'll get a PR in.

wicadmin, 2020-06-23:
```
MM: 1.14.0 (https://www.freedesktop.org/software/ModemManager)
libmbim: 1.24.0 (https://www.freedesktop.org/software/libmbim)
libqmi: 1.26.0 (https://www.freedesktop.org/software/libqmi)
```

wicadmin, 2020-06-23:
@superm1  - seems the snap build failed.

superm1, 2020-06-23:
Yeah, looks like it needs some more deps.  This is part of why we stick to tagged releases in the snap build for deps, to avoid the yo-yo from other projects moving too quickly for us to keep up with.  I respun it and i'll keep an eye to see if it's enough.

wicadmin, 2020-06-23:
Thanks...and it will be in edge again right once successful?

superm1, 2020-06-23:
Yeah once it's successful and merged it will promote to edge.  Depends on how busy the build farm is, give it an hour or so after merged to be promoted.

wicadmin, 2020-06-23:
OK, will do...yes, @aleksander0m  has kept it pretty much dependency free, not much has changed in that area since I've started using it a couple years ago. Did have a `runtime` dependency of PPP once that was elusive, but that was about it.

superm1, 2020-06-23:
Hmm...

```
FAILED: plugins/modem-manager/21c01c5@@fu_plugin_modem_manager@sha/fu-qmi-pdc-updater.c.o 
 
cc -Iplugins/modem-manager/21c01c5@@fu_plugin_modem_manager@sha -Iplugins/modem-manager -I../plugins/modem-manager -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -Isubprojects/libxmlb/src -I../subprojects/libxmlb/src -Isubprojects/gusb -I../subprojects/gusb -Isubprojects/gusb/gusb -I../subprojects/gusb/gusb -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/gio-unix-2.0/ -I/usr/include/libusb-1.0 -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/gudev-1.0 -I/usr/include/json-glib-1.0 -I/root/project/stage/include/libmm-glib -I/root/project/stage/include/ModemManager -I/root/project/stage/include/libqmi-glib -I/root/project/stage/include -I/root/project/stage/usr/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -g -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wdiscarded-qualifiers -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmaybe-uninitialized -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-unknown-pragmas -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 '-DMM_REQUIRED_VERSION="1.10.0"' '-DQMI_REQUIRED_VERSION="1.23.1"' -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginMm"' -MD -MQ 'plugins/modem-manager/21c01c5@@fu_plugin_modem_manager@sha/fu-qmi-pdc-updater.c.o' -MF 'plugins/modem-manager/21c01c5@@fu_plugin_modem_manager@sha/fu-qmi-pdc-updater.c.o.d' -o 'plugins/modem-manager/21c01c5@@fu_plugin_modem_manager@sha/fu-qmi-pdc-updater.c.o' -c ../plugins/modem-manager/fu-qmi-pdc-updater.c
 
In file included from /usr/lib/x86_64-linux-gnu/glib-2.0/include/glibconfig.h:9:0,
 
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
 
                 from /usr/include/glib-2.0/glib/galloca.h:32,
 
                 from /usr/include/glib-2.0/glib.h:30,
 
                 from /usr/include/glib-2.0/gobject/gbinding.h:28,
 
                 from /usr/include/glib-2.0/glib-object.h:23,
 
                 from /usr/include/glib-2.0/gio/gioenums.h:28,
 
                 from /usr/include/glib-2.0/gio/giotypes.h:28,
 
                 from /usr/include/glib-2.0/gio/gio.h:26,
 
                 from ../libfwupdplugin/fu-common.h:9,
 
                 from ../plugins/modem-manager/fu-qmi-pdc-updater.c:11:
 
/usr/include/glib-2.0/glib/gmacros.h:441:43: error: redefinition of ‘glib_autoptr_cleanup_QmiMessagePdcLoadConfigInput’
 
 #define _GLIB_AUTOPTR_FUNC_NAME(TypeName) glib_autoptr_cleanup_##TypeName
 ```

wicadmin, 2020-06-23:
Let me try compiling these versions.

superm1, 2020-06-23:
In the 1.26.0 changelog I see
```
   ** Added g_autoptr() support to all public types.
```

Which probably explains what's happening

superm1, 2020-06-23:
for `libqmi-glib`

wicadmin, 2020-06-23:
I am able to get libqmi to install.

```
qmicli -V
qmicli 1.26.0
Copyright (C) 2012-2020 Aleksander Morgado
```

[70 more replies]

commit f0b82736 (2020-06-22): trivial: add Vendor ID into Modem manager devices

Fixes: #2200
commit 5521e475 (2020-06-22): trivial: add Vendor ID into Modem manager devices

Fixes: #2200
commit d40eee27 (2020-06-23): trivial: snap: move to newer modem manager stuff
Fixes: #2200
PR #2217 (2020-06-23): trivial: snap: move to newer modem manager stuff
Fixes: #2200

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
closed 2023-09-23 by hughsie