fwupd/fwupd #2205: Wiki: Getting involved - What does this sentence mean?

mendel5, 2020-06-22:
In the paragraph https://github.com/fwupd/fwupd/wiki/Getting-involved#guids it says:

> In this case STORAGE-DELL-108788 was used to build the GUID f5396862-8816-5437-823e-d7899a32e667 Placing this GUID in the XML file to match that string.

The last sentence `Placing this GUID in the XML file to match that string.` does not seem fitting here.

Maybe you @superm1 can help?

superm1, 2020-06-22:
sure, made a small adjustment.  Go ahead and change further yourself, or raise more questions if you think it's still unclear.

closed 2020-06-22 by superm1

mendel5, 2020-07-25:
Hi @supermario, now I know why I was wondering about this sentence.

The sentence
> In this case STORAGE-DELL-108788 was used to build the GUID f5396862-8816-5437-823e-d7899a32e667.

makes more sense when you know that the code uses `uuid.uuid5` to build the GUID. See https://lvfs.readthedocs.io/en/latest/metainfo.html#using-guids

As described on Wikipedia:
> Version-3 and version-5 UUIDs are generated by hashing a namespace identifier and name. Version 3 uses MD5 as the hashing algorithm, and version 5 uses SHA-1.
https://en.wikipedia.org/wiki/Universally_unique_identifier#Versions_3_and_5_(namespace_name-based) .

At first I thought the UUID/GUID was created using version 4 which would be randomly generated. But using version 5 with SHA-1 makes sure that for the same input x the output y will always be the same.