fwupd/fwupd #2212: Dell WD19TB failing to update

bulletmark, 2020-06-23:
**Describe the bug**
Trying to update Dell dock WD19TB but the update always stays "Updates will complete ..".

**Steps to Reproduce**
1. sudo fwupdmgr enable-remote lvfs-testing
2. sudo fwupdmgr refresh
3. sudo fwupdmgr get-devices
4. sudo fwupdmgr get-updates
5. sudo fwupdmgr update

Unplugged laptop from dock and waited 15 mins. Also, power recycled the dock later.

**Expected behavior**
Expected firmware to update.

**fwupd version information**
```shell
lt:~ sudo fwupdmgr --version
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4
```
Installed by pacman on Arch Linux.

**fwupd device information**
```shell
$ fwupdmgr get-devices --filter=~internal
lt:~ sudo fwupdmgr get-devices --filter=~internal
XPS 13 9300
│
├─WD19TB:
│ │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.04
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   Serial Number:       4DN5523/3159128927644594
│ │   Update Message:      Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │ 
│ ├─Thunderbolt controller in Dell dock:
│ │     Device ID:         5e8cd9f97ee07a0af3a5e0c70fffb713aa32362e
│ │     Summary:           Thunderbolt controller
│ │     Current version:   43.00
│ │     Vendor:            Dell Inc. (TBT:0x00D4)
│ │     Install Duration:  22 seconds
│ │     GUID:              c94770ca-1773-592c-b20a-e87243bc7cd0 ← TBT-00d4b070
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │   
│ ├─Package level of Dell dock:
│ │     Device ID:         073624c16dd99abe01ba1da223a70321e4f29beb
│ │     Summary:           A representation of dock update status
│ │     Current version:   01.00.14.01
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  5 seconds
│ │     GUID:              8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:      • Updatable
│ │                        • Supported on remote server
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:         0acceea54e71c4d8002593822afe0f4705616613
│ │     Summary:           USB 3.1 Generation 1 Hub
│ │     Current version:   01.21
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  14 seconds
│ │     GUIDs:             86fb40c0-8bf5-5a8b-a4ad-3156cf6bfaf4 ← USB\VID_413C&PID_B06F&REV_0101
│ │                        b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                        ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:         96969b82d977fdcbf607df084ef0dcf10119409b
│ │     Summary:           USB 3.1 Generation 2 Hub
│ │     Current version:   01.47
│ │     Vendor:            Dell Inc. (USB:0x413C)
│ │     Install Duration:  3 seconds
│ │     GUIDs:             707c63d2-e597-5c40-84db-9b1bb4c48d96 ← USB\VID_413C&PID_B06E&REV_0101
│ │                        acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
│ │                        568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device is usable for the duration of the update
│ │   
│ └─VMM5331 in Dell dock:
│       Device ID:         228ac6846b64d532dd1ea1b7e651e8f5f55fbc34
│       Summary:           Multi Stream Transport controller
│       Current version:   05.04.03
│       Vendor:            Dell Inc. (USB:0x413C)
│       Install Duration:  6 minutes
│       GUID:              89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
│       Device Flags:      • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│                          • Device is usable for the duration of the update
│     
├─Integrated Webcam HD:
│     Device ID:           4295296d98b3ba38c72f6baa33d24f03a1d428f6
│     Current version:     91.58
│     Vendor:              CN0PW36V8LG0099B00DPX02 (USB:0x0BDA)
│     Serial Number:       200901010001
│     GUIDs:               bba32a58-5e20-5d64-bc15-287159bdaf46 ← USB\VID_0BDA&PID_5511&REV_9158
│                          14e94ffe-f85a-5645-9af7-705464a66c8c ← USB\VID_0BDA&PID_5511
│     Device Flags:        • Updatable
│   
├─Unifying Receiver:
│     Device ID:           394573a9ff378daf9103489552ab3d4f0e1b98ce
│     Summary:             A miniaturised USB wireless receiver
│     Current version:     RQR24.11_B0036
│     Bootloader Version:  BOT03.02_B0009
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
│                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:        • Updatable
│                          • Supported on remote server
│   
└─VMM3200:
      Device ID:           47a0754f6a2c9a7eb5f2883e4b78834a64451577
      Summary:             Multi-Stream Transport Device
      Current version:     0.00.00
      Vendor:              Synaptics (DRM_DP_AUX_DEV:0x06CB)
      GUIDs:               1beb64bd-6dc8-5db2-84b9-f32964abc876 ← MST-leaf-vmm3200-0
                           b57f0a20-424c-51e0-b91b-575f3d74ed3a ← MST-leaf-0
                           d6072785-6fc0-5f83-9d49-11376e7f48b1 ← MST-leaf
      Device Flags:        • Updatable
```

**Dock SKU**
-  WD19TB (Thunderbolt)

**Peripherals connected to the dock**
- 2 Dell displays connected via DP
- USB wired Keyboard
- USB Mouse transceiver
- USB webcam
- Ethernet cable to router
- 3.5 mm audio

**Verbose daemon logs**
```shell
$ fwupdmgr modify-config "VerboseDomains" "*"
$ sudo fwupdmgr get-devices
$ sudo fwupdmgr reinstall 570867f4ddda9915445da59bd85cd0e0e507270e
# Seemed to complete OK. Was prompted to upload results which I did.
# Unplugged laptop cable to dock and waited 3+ mins
$ sudo fwupdmgr get-devices
# still says:
─WD19TB:
│ │   Device ID:           570867f4ddda9915445da59bd85cd0e0e507270e
│ │   Summary:             High performance dock
│ │   Current version:     01.00.00.04
│ │   Minimum Version:     01.00.00.00
│ │   Vendor:              Dell Inc. (USB:0x413C)
│ │   Install Duration:    1 minute
│ │   Serial Number:       4DN5523/3159128927644594
│ │   Update Message:      Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.
│ │   GUID:                cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Device stages updates
│ │                        • Device can recover flash failures
│ │                        • Device is usable for the duration of the update

$ journalctl -b -u fwupd.service --since "15 mins ago"
# Log too big. Get log from https://www.dropbox.com/s/995ushlokza1ywa/systemd.log

```

**Additional questions**
- Arch Linux updated to current.
- Yes, unplugged dock and later tried power cycling.
- This is new laptop and dock. fwupd did update bios correctly but not the dock.

superm1, 2020-06-23:
Everything looks updated to me. I'll double check your life tomorrow for anything that stands out, but it looks to me you're on the latest firmware now.

bulletmark, 2020-06-23:
So why does it keep saying "Updates will complete after the dock is unplugged"?

bulletmark, 2020-06-23:
Also, while you are here, I am having a lot of problems with this dock and my XPS 13 9300. The laptop locks up if the dock is connected and the laptop sleeps (either s2idle or deep). Without the dock connected it works fine, hence why I am trying to get latest TB firmware. So I am particularly keen to get the latest TB controller firmware in the laptop. The output is:
```
XPS 13 9300
│
├─Thunderbolt Controller:
│     Device ID:           4dfdfb15692ebe398660902d377dc876bff94c7b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     80.00
│     Update Error:        Missing non-active nvmem
│     GUID:                e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:        • Internal device
│                          • Requires AC power
```
Is that the latest version? What does "Update Error: Missing non-active nvmem" mean?

superm1, 2020-06-23:
Oh I see what you mean. We might have a plumbing problem with the history database. I'll look through your logs tomorrow and see if I can repro it.

bulletmark, 2020-06-23:
@superm1, sorry but I hit you up with 2 quick questions above. Can you please look at my [second question](https://github.com/fwupd/fwupd/issues/2212#issuecomment-647895909)?

superm1, 2020-06-23:
>│ │   Update Message:      Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.

I can reproduce this.  It happens with no updates applied, no `pending.db`.  It's once you run `fwupdmgr refresh` that message is applied to the device.  I'll see if I can come up with a way to prevent it but still let the message show when appropriate (pending updates).

Your dock is fully up to date, so the message is meaningless in your situation.

>Is that the latest version? What does "Update Error: Missing non-active nvmem" mean?

The TBT firmware is distributed as part of the `System Firmware` in the XPS 9300.  The fwupd device version isn't super useful in this case.  We explicitly didn't make mention this because there are other reasons for not being able to flash from Linux, and this is just one of them.  Can you advise a message that you think would have made more sense in Update Error?  We can certainly discuss to change it to something more sensical.

>The laptop locks up if the dock is connected and the laptop sleeps (either s2idle or deep

Try this patch: https://people.canonical.com/~acelan/bugs/x_crash/0001-drm-i915-icl-add-missing-GT-workaround.patch

bulletmark, 2020-06-23:
Your response here makes perfect sense and your explanation of that error is clear but surely you can see that reporting "Update Error: Missing non-active nvmem" is blatantly unclear to a simple user? In fact, it is worse than that because it implies a system/user error causing a problem and is one of the reasons I came to this issue tracker. Saying something like "Update not possible - update included in System Firmware" would make more sense.

Regarding the output of `get-devices`, all a user really wants to know is have I got the latest version but the output does not really tell him that. It lists the "Current Version" but not an indicator if this is the latest available. Also, if I go to the [Dell Site](https://www.dell.com/support/home/en-us/drivers/driversdetails?driverid=p20vd&oscode=wt64a&productcode=dell-wd19tb-dock) the only version I see there for WD19TB is 01.00.14.01, A03 but how do I correlate this with anything displayed in the `get-devices` output?

Again, while you are here, the bug instructions say to do a `fwupdmgr modify-config "VerboseDomains" "*"` before sending log data. I have looked all over the place but how do I turn this off? There is no `list-config`, where is this config stored? Because I set that option, fwupd.service is spamming my journal. For that matter, why does that service need to run all the time? I really only want it when I want to check for and install updates. I could not work out how to disable it (other than masking the service I guess) and could not work out how to lower all that spam so my only solution was to uninstall fwupd(!).

Another note. I had a number of cases where `fwupdmgr get-devices` would work and eventually output a message about incompatible versions (sorry can not remember the exact text) but the problem was that the fwupd.service was locked up so I just rebooted to clear the fault.

superm1, 2020-06-23:
> Your response here makes perfect sense and your explanation of that error is clear but surely you can see that reporting "Update Error: Missing non-active nvmem" is blatantly unclear to a simple [...]

You know, maybe we should just exclude this device in that circumstance.  The firmware version doesn't really mean anything to anyone, and it's not upgradable from Linux.

> Regarding the output of get-devices, all a user really wants to know is have I got the latest version but the output does not really tell him that. It lists the "Current Version" but not an [...]

This correlates to "Package version of Dell dock".  Because it's a composite update, lots of versions are displayed for different components.

> Again, while you are here, the bug instructions say to do a fwupdmgr modify-config "VerboseDomains" "*" before sending log data. I have looked all over the place but how do I turn this off? There is [...]

I think `fwupdmgr modify-config "VerboseDomains" ""` should unset it.  If that doesn't work (which it might not like the empty parameter) then modify `/etc/fwupd/daemon.conf` to unset it.

> For that matter, why does that service need to run all the time? I really only want it when I want to check for and install updates. I could not work out how to disable it (other than masking the [...]

It will actually shut down if not in use, by a timeout configurable in that same configuration file.

> Another note. I had a number of cases where fwupdmgr get-devices would work and eventually output a message about incompatible versions (sorry can not remember the exact text) but the problem was that the fwupd.service was locked up so I just rebooted to clear the fault.

If you can ever reproduce this, certainly interested to hear more in another bug report.  It sounds like a missing timeout somewhere.

bulletmark, 2020-06-23:
Regarding that last comment, there are 2 bugs.
1. The service is locking up sometimes. Obviously you need more info about this and yes if I see it again I will provide that.
2. The client is reporting a "version mismatch" for the case where the reply to the daemon times out. It reports something like server version (null), client version 1.X. This should probably be fixed to say something like "service did not respond".

superm1, 2020-06-24:
I've submitted another PR related to that error message you mentioned for the problem #"2" you encountered.

bulletmark, 2020-06-24:
FYI, [that patch](https://people.canonical.com/~acelan/bugs/x_crash/0001-drm-i915-icl-add-missing-GT-workaround.patch) you suggested didn't make any difference :( I think that is the right track though as I suspect it is related to the i915 video driver dealing with the 2 external screens connected via DP on the TB dock. The machine appears to hang but I find I can actually still ssh to it, via the network port on the TB dock.

superm1, 2020-06-24:
> FYI, [that patch](https://people.canonical.com/~acelan/bugs/x_crash/0001-drm-i915-icl-add-missing-GT-workaround.patch) you suggested didn't make any difference :( I think that is the right track [...]

In that case I think it's best to follow https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs to file a bug report with your exact circumstances.  I do know similar things happened on the older kernel that the XPS 9300 was certified with, but if that patch isn't helping probably need have a separate report raised.

commit 060eb68f (2020-06-25): fu-util: Only show UpdateMessage when state is success

This will show it only when an update was sucessful.
Fixes: #2212
PR #2218 (2020-06-25): fu-engine: only set update message from metadata on upgrades
After a release finishes installing, this isn't useful metadata.

Fixes: #2212

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit c2721c86 (2020-06-25): fu-util: Only show UpdateMessage when state is success

This will show it only when an update was sucessful.
Fixes: #2212
commit 71cea593 (2020-06-26): fu-util: Only show UpdateMessage when state is success

This will show it only when an update was sucessful.
Fixes: #2212
closed 2020-06-25 by superm1