fwupd/fwupd #2219: Dell XPS 9570 System Firmware - Update errors

ericonr, 2020-06-23:
**Describe the bug**

https://fwupd.org/lvfs/devices/com.dell.uefie4d7bdc4.firmware (as well as Dell's own page) shows that the 1.16.2 version of the system firmware for this laptop is available. My system is running the 1.15.0 one, but I get update errors when running `get-devices`, and the update isn't even shown. I have already refreshed and force refreshed the database.

```
$ fwupdmgr get-devices
[...]
├─System Firmware:
│     Device ID:           xxxxxx
│     Current version:     1.15.0
│     Minimum Version:     1.15.0
│     Vendor:              Dell Inc. (DMI:Dell Inc.)
│     Update Error:        missing signed bootloader for secure boot: /usr/libexec/fwupd/efi/fwupdx64.efi.signed cannot be found
│     GUID:                xxxxxx
│     Device Flags:        • Internal device
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
[...]

$ fwupmgr update
• Thunderbolt Controller has no available firmware updates
• PC401 NVMe SK hynix 512GB has no available firmware updates
• Touchpad has no available firmware updates
No updatable devices
```

I have signed the bootloader with `sbsign` (the same error appeared before and after that):

```
$ cd cd /usr/libexec/fwupd/efi/
$ sudo sbsign --cert /etc/refind.d/keys/refind_local.crt --key /etc/refind.d/keys/refind_local.key fwupdx64.efi
$ ls
fwupdx64.efi
fwupdx64.efi.signed
```

I don't know if it's related, but I get this info about the TPM module:

```
├─TPM 2.0:
│     Device ID:           xxxxxxx
│     Summary:             Platform TPM device
│     Current version:     7.2.0.1
│     Vendor:              Dell Inc. (PCI:0x1028)
│     Update Error:        Updating disabled due to TPM ownership
│     GUIDs:               xxxxx 
[...]
│     Device Flags:        • Internal device
│                          • Requires AC power
```

I have erased all the codes because I don't know how unique they are, but if you need any of them, let me know.

**Steps to Reproduce**

Explained above, I think.

**Expected behavior**

The update would show up regardless of the presence of a signed bootloader. Ideally, some mechanism to create the signed bootloader should be available as well.

**fwupd version information**

Please provide the version of the daemon and client.

```
$ fwupdmgr --version
client version:	1.4.2
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `xbps-install`, from Void Linux.

**System UEFI configuration**

Please provide the output of the following commands:

```shell
$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0006
Boot0000* VoidLinux	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,AC-E4-2E-00-95-10-DF-61)/HD(1,GPT,0364df65-c719-437b-ad5b-2e373e6eec3c,0x800,0x800000)/File(\EFI\void\vmlinuz-1.2_1.EFI)
Boot0006* rEFInd Boot Manager	HD(1,GPT,0364df65-c719-437b-ad5b-2e373e6eec3c,0x800,0x800000)/File(\EFI\refind\refind_x64.efi)
```

**Additional questions**

- Operating system and version: Void Linux, up to date (rolling release)
- Have you tried rebooting? No.
- Is this a regression? Not sure.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled? Yes.

ericonr, 2020-06-23:
As it turns out, killing the daemon that was spawned before I generated the signed bootloader was enough for it to detect the signed bootloader. So the daemon is probably checking for its presence during startup and caching that, which seems like the wrong behavior.

So, the actual bugs for this issue are:

- If a signed bootloader can't be found, `fwupdmgr` doesn't even warn the user about existing updates.
- If a signed bootloader is created after the daemon has been launched, the daemon doesn't know about it and doesn't check for it.
- The program only complains about the lack of a signed bootloader, and doesn't have a way to sign it (I'm not sure if this one is within the scope of this program, specially given that there are so many ways of managing SB keys).

superm1, 2020-06-23:
>I have erased all the codes because I don't know how unique they are, but if you need any of them, let me know.

JFYI these are not private data when in hash form.  If they were in the form before they were hashed, that's up to your security policy.

> As it turns out, killing the daemon that was spawned before I generated the signed bootloader was enough for it to detect the signed bootloader. So the daemon is probably checking for its presence during startup and caching that, which seems like the wrong behavior.
> 

That's correct - it is cached at startup.  We do actually have another check later on `fu_uefi_get_built_app_path` that will run at update time.  I think we can probably drop the coldplug check.  @hughsie what do you think?

> So, the actual bugs for this issue are:
>
> * If a signed bootloader can't be found, `fwupdmgr` doesn't even warn the user about existing updates.
> * If a signed bootloader is created after the daemon [...]

If we drop that check, then in your case you can try to update it, and if the bootloader was signed it would work, and if it wasn't then you would get an error.  And then you can sign it and try again without restarting the daemon.

ericonr, 2020-06-23:
> JFYI these are not private data when in hash form. If they were in the form before they were hashed, that's up to your security policy.

I see, thanks!

What's the behavior with your linked PR? `get-devices` would show that an update is available and, and `update` would error out because it can't find the signed bootloader?

Thanks for the quick response!

superm1, 2020-06-23:
The behavior is that you won't see any errors until you try to run the update, correct.  And when you run it you'll see something like
`/usr/lib/fwupd/efi/fwupdx64.efi.signed` not found.  If you or @hughsie  don't think that's verbose enough, there is an opportunity to prefix it.

ericonr, 2020-06-24:
I think it would make sense to have something similar to the current one: `Update Error: missing signed bootloader for secure boot: /usr/libexec/fwupd/efi/fwupdx64.efi.signed cannot be found`

superm1, 2020-06-24:
OK, pushed an updated commit into PR that will accomplish something like that.  It won't show up in `UpdateError` but it will be in the error string you get when you try to run the update.

closed 2020-06-24 by superm1

ericonr, 2020-06-24:
Thanks a lot for the fix!

commit c3a01371 (2020-06-24): trivial: uefi: drop secure boot check at coldplug

We'll instead check this when the user tries to run an update.  This
allows them to sign a bootloader after the daemon starts (or remove
a signed bootloader after starting)

Fixes: #2219
commit 13207f3c (2020-06-24): trivial: uefi: drop secure boot check at coldplug

We'll instead check this when the user tries to run an update.  This
allows them to sign a bootloader after the daemon starts (or remove
a signed bootloader after starting)

Fixes: #2219
PR #2220 (2020-06-24): trivial: uefi: drop secure boot check at coldplug
We'll instead check this when the user tries to run an update.  This
allows them to sign a bootloader after the daemon starts (or remove
a signed bootloader after starting)

Fixes: #2219

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 4669dd59 (2020-06-24): trivial: uefi: drop secure boot check at coldplug

We'll instead check this when the user tries to run an update.  This
allows them to sign a bootloader after the daemon starts (or remove
a signed bootloader after starting)

Fixes: #2219