fwupd/fwupd #2230: Flatpak help text seems to tell the user the wrong command for invoking the executable

LinuxOnTheDesktop, 2020-06-26:
**Describe the bug**

`sudo flatpak run org.freedesktop.fwupd` produces text that starts:

> Usage:
>   fwupdtool [OPTION…]

Yet: (1) the non-flatpak version of fwupd is run by the command `fwupdmgr`: (2) the flatpak version is run - on my systems at least - via the (not very friendly) `sudo flatpak run org.freedesktop.fwupd`. Contrast the 'fwupdtool' in the text I quoted.

**Steps to Reproduce**

1. Install Flatpak version ([which there are no instructions for doing](https://github.com/fwupd/fwupd/issues/2156)).
2. `sudo flatpak run org.freedesktop.fwupd`.

**Expected behavior**
The flatpak version should tell you the actual command that you need to run it. Failing that, it should tell you the command that would work on a non-flatpak installation. Or perhaps I am confused: if so, I submit that the problem is that the program confuses.

**fwupd version information**
```
$ flatpak list
Ref                                                 Options       
org.freedesktop.fwupd/x86_64/stable                 system,current
[. . .]
$ sudo flatpak run org.freedesktop.fwupd --version
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.3
	efivar:	37

```

**Additional questions**
- Operating system and version: Mint 19.3 x64 Cinnamon
- Have you tried rebooting? Yes
- Is this a regression? I do not know

hughsie, 2020-06-26:
> the non-flatpak version of fwupd is run by the command fwupdmgr

the fwupd project has three main deliverables:

1. the fwupd daemon, runs as root automatically in the background
2. the fwupdmgr front end util, run as user in the foreground
3. the fwupdtool power user tool which runs as root in the foreground (without the daemon)

for flatpak we only package the last one as it's not easy to run a daemon as root without the sandbox. This is why you're seeing references to fwupdtool and not fwupdmgr for flatpak. Of course, without the daemon it takes a few seconds to probe the hardware each time, rather than relying on a cached list of properties.

LinuxOnTheDesktop, 2020-06-26:
That clarifies some things that have long confused me. _However_, there is still the following oddity. You write that the flatpak version of your software comprises 'fwupdtool'. Yet, the user is to run that application via the command 'fwupdmgr', despite (1) that difference in name and even though (2) the help text produced by  via the command 'fwupdmgr' tells one to run `fwupdtool`?

>  Of course, without the daemon it takes a few seconds to probe the hardware each time, rather than relying on a cached list of properties.

That does not account for thirty seconds before the help text is displayed (see my previously linked issue report) or does it?

hughsie, 2020-06-26:
> Yet, the user is to run that application via the command 'fwupdmgr

Were does it say that? Thanks.

superm1, 2020-06-26:
> That does not account for thirty seconds before the help text is displayed (see my previously linked issue report) or does it?

That's the bug you are tracking upstream with flatpak and isn't a fwupd bug.

LinuxOnTheDesktop, 2020-06-26:
> That's the bug you are tracking upstream with flatpak and isn't a fwupd bug.

Yes; I mentioned it only because you mentioned a delay ('without the daemon it takes a few seconds to probe the hardware each time').

**On running via `fwupdmgr`**: my mistake. As my own report above shows, one runs the flatpak via `sudo flatpak run org.freedesktop.fwupd`. I got confused because I aliased that command to `fwupdmgr`.  (I did so in an effort, er, to reduce confusion when moving to the flapak version of the tool from the apt, or actually the snap, version.) _However_: it remains the case that the flatpak version of the tool tells the user to issue (and seems to present itself as resulting from issuing) what seems to be the wrong command, namely, `fwupdtool`.

superm1, 2020-06-26:
I do think it would be nice for flatpak to offer an aliasing functionality that lets you say put something into $PATH that is a wrapper to your binary.  But it doesn't exist there.  Maybe it's a feature request to the flatpak folks.
This is something that snap does (you can request aliases from the snap store).

superm1, 2020-06-26:
Within the flatpak container the binary is absolutely called fwupdtool (see https://github.com/flathub/org.freedesktop.fwupd/blob/master/org.freedesktop.fwupd.json#L7).  So this is completely not a fwupd bug.  If you think that it should be executable via fwupdtool in $PATH, then please raise it to flatpak guys as a feature request as I mentioned previously.

closed 2020-06-26 by superm1

superm1, 2020-06-26:
Actually it looks like there is some work for this:
https://github.com/flatpak/flatpak/pull/1254
https://github.com/flatpak/flatpak/issues/1188

They're put in `$flatpakdir/exports/bin` and you can add to PATH if you want to.

LinuxOnTheDesktop, 2020-06-26:
I am sorry to bang on but I will do so because, to put it metaphorically, the quagmire deepens though there is a ray of light. Let me explain.

I added that directory to my path via a config file. I rebooted and verified that the path change had been made. It had. I ran `~ $ fwupdtool` and `~ $ fwupd` and both returned 'command not found'. _However_, `~  $ org.freedesktop.fwupd` works _and without any delay_. _Yet_ - can you believe this? -  `~  $ sudo org.freedesktop.fwupd` fails, as does e.g.  `~  $ org.freedesktop.fwupd refresh`. The latter fails because . . it needs sudo. 

All this strengthens the case that someone who actually understands these things should write a guide to using the flatpak version of fwupd.

superm1, 2020-06-26:
@hughsie what about if we asked flathub people to rename it to `org.freedesktop.fwupd.fwupdtool`?  That might be more obvious?

hughsie, 2020-06-26:
I think I'm the flathub maintainer for fwupd. I'm not super keen on changing the name, it's a lot of work. I think I'd rather just sit down and write some really good documentation....

superm1, 2020-06-26:
I think the most important thing in documentation that's coming across to me is that it needs to be front and center at the top that 
1. the flatpak is a supplement, not a replacement.  You should be using your distro provided version for most stuff, and if there is a plugin you need only in the new shiny, use it for that.
2. it's a power user tool, you can't use a GUI, need root to actually launch it etc.

LinuxOnTheDesktop, 2020-06-26:
> the flatpak is a supplement, not a replacement. You should be using your distro provided version for most stuff

This is complete news to me and frankly it seems to add yet another layer of as-yet-undocumented complexity! Moreover, I need a current version of fwup and not, so far as I am aware, because of a 'plugin' but rather because the old version in my distro's repos fails to work, in various ways, with my hardware. The _snap_ version works well, but my distro - Mint - is abandoning snap, as of, probably, tomorrow - which is when Mint 20 comes out. (Admittedly Mint 20 will have a newer version of fwupd, and hopefully that version will work with my hardware, but, clearly, I and others might need access to a newer version.)

superm1, 2020-06-26:
>This is complete news to me and frankly it seems to add yet another layer of as-yet-undocumented complexity

The flatpak doesn't compile in support for the UEFI plugin, which I would argue is the most important plugin of all in fwupd. 

>Mint - is abandoning snap, as of, probably, tomorrow - which is when Mint 20 comes out

It's not like Mint is a proprietary OS.  You can certainly go and install snap yourself even if it's not supported.  Also if Mint really is based off Ubuntu 20.04, it will have fwupd 1.3.9, which is a fairly modern version.

LinuxOnTheDesktop, 2020-06-27:
i> The flatpak doesn't compile in support for the UEFI plugin, which I would argue is the most important plugin of all in fwu

So, wait, do I need to install some other version of fwupd if I wish to use that plugin, then? This is exactly the sort of complexity I was complaining of. Frankly, it seems to be that the flatpak version of fwupd is not usable by non-experts. I for one have spent more than enough time on - trying to use - that version. I will remove it and use the version of fwupd in Mint 20 or, if that fails, I suppose I will have to contravene a design decision of my distro and install the snap version. All of this is a pain.

superm1, 2020-06-28:
Like I said it's for power users that need a newer version for some plugin or fix.

UEFI can't be supported there because flatpak can't write to the EFI system partition.