fwupd/fwupd #2233: Possibilities to downgrade Dell WD15 Dock firmware

vsht, 2020-06-28:
**Describe the question**
The question may not be precisely related to the development of fwupd,
so I apologize in advance if it's completely off-topic.

I'm using Fedora 32 on a Dell Latitude 5480 (iGPU only). At home and
at work the laptop is connected to a Dell WD 15  dock via USB C.

At work my setup consists of two FHD screens (one of them connected
via VGA) and I have no issues so far. 

At home the WD15 is connected to a Dell Ultrasharp U2515H (WQHD). That 
particular WD15 dock which I bought used from eBay seemed to have 
a pretty old firmware (unfortunately, I didn't look up the version number).
Nevertheless, it worked flawlessly with my Dell screen. Some time ago I 
was stupid enough to update the WD15 firmware (I know, never touch the 
running system). I did that via Windows 10 and everything worked flawlessly.

However, that firmware seems to have a nasty Linux-only bug: Once I cold boot
the system or connect already booted laptop to the dock, the screen will start 
flickering every 5-10 minutes, i.e. it blacks out for 3-4 seconds, until the image 
comes back and then after some minutes the pattern repeats. This doesn't 
happen when running Windows 10, so the bug really appears to be related
to Linux. However, `dmesg` doesn't show any output when those black outs
happen, which makes it virtually impossible to debug. 

I just can say, that this doesn't happen with FHD screens, so for some reason the combination
of WQHD+Linux+newer WD 15 firmware results into an unusable setup.

For me, the only option seems to be to perform a downgrade to a firmware
version that used to work. I don't particularly care if there is a risk to brick the dock,
as I got the current one rather cheap on eBay and I if needed I could probably 
get another one for a similar price (<<50 €).

As far as I understand, DELL's Windows utilities do not support downgrading. At least 
I wasn't able to make `DellDockingStationFwUp_1.0.4_IE_03222018.exe` flash the firmware
after having installed the latest one from `DellDockingStationFwUp_1.0.6_12052018_WD15.exe`
before.

I had more success with `fwupdmgr --allow-older install` and managed to downgrade 
`Dell Inc. WD15 VMM3332 MST` to `3.10.002` via `mst_3_10_002.cab`. However, this version
seems to be "too new" as my screen flickering problems are still there. 

I would appreciate to learn what options I still have. Is there perhaps a possibility to scavenge
right firmware blobs from the `DellDockingStationFwUp` files and ``feed'' them to fwupdmgr?
Or perhaps there is some magic trick to make `DellDockingStationFwUp` flash an older firmware
version anyway?

I'm asking this question here since I noticed that there are some Linux- and tech-savvy  
DELL employees around, so the hope is to get more useful information than via the official forums.
As I've already stated, I'm fully aware of potential risks to brick the dock by doing these kind
of manipulations, but I'd like to try it anyway. I'm certainly not going to blame anyone if things
go wrong.

**fwupd version information**
```shell
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
```

superm1, 2020-06-29:
>However, that firmware seems to have a nasty Linux-only bug: Once I cold boot
the system or connect already booted laptop to the dock, the screen will start
flickering every 5-10 minutes, i.e. it blacks out for 3-4 seconds, until the image
comes back and then after some minutes the pattern repeats. This doesn't
happen when running Windows 10, so the bug really appears to be related
to Linux. However, dmesg doesn't show any output when those black outs
happen, which makes it virtually impossible to debug.

Are you absolutely sure this was the only part that changed ? As in you dual booted, rebooted to Windows upgraded the firmware and came back to Linux and hit these problems?  Or is it _possible_ there was a kernel regression?

>I just can say, that this doesn't happen with FHD screens, so for some reason the combination
of WQHD+Linux+newer WD 15 firmware results into an unusable setup.

But you're sure that WQHD did work before?

> I had more success with fwupdmgr --allow-older install and managed to downgrade
Dell Inc. WD15 VMM3332 MST to 3.10.002 via mst_3_10_002.cab. However, this version
seems to be "too new" as my screen flickering problems are still there.

I was going to agree if any firmware is causing this it's probably the MST, which you already downgraded with fwupd.

> I'm asking this question here since I noticed that there are some Linux- and tech-savvy
DELL employees around, so the hope is to get more useful information than via the official forums.
As I've already stated, I'm fully aware of potential risks to brick the dock by doing these kind
of manipulations, but I'd like to try it anyway. I'm certainly not going to blame anyone if things
go wrong.

To be honest - this _sounds_ like an i915 regression not a firmware bug.  I would follow this stuff to get a bug filed and test with latest i915.
https://gitlab.freedesktop.org/drm/intel/-/wikis/How-to-file-i915-bugs

vsht, 2020-06-29:
> Are you absolutely sure this was the only part that changed ? As in you dual booted, rebooted to Windows upgraded the firmware and came back to Linux and hit these problems? Or is it possible there was a kernel regression?

Yes, I'm sure. The issues started immediately after coming back to Linux after the firmware upgrade. I didn't update
the kernel in between. If it would have been a kernel regression, I would have just gone back to the last working kernel version.

> But you're sure that WQHD did work before?

Of course, since I've been using it for many months without any issues.

> I was going to agree if any firmware is causing this it's probably the MST, which you already downgraded with fwupd.

As I wrote, I don't know what is the original firmware version the dock had. If I understand you correctly, the firmware upgrade 
DellDockingStationFwUp_1.0.4_IE_03222018.exe corresponds to `mst_3_10_002.cab`. However, this also means that
the original factory firmware must haven even an earlier date. Perhaps it was the one that worked for me.

Another possibility would be that using mst_3_10_002.cab doesn't really donwgrade everything to the level of
the 1.0.4 firmware package. This I obviously can't know.

> To be honest - this sounds like an i915 regression not a firmware bug. I would follow this stuff to get a bug filed and test with latest i915.

I wouldn't really call this regression, since with the old firmware everything worked flawlessly. Moreover, I've already
experienced some i915 regressions in the past (not related to WD15) and there one always could see some dmesg
output appearing simultaneously with graphical artifacts. In this case, whenever the screen goes black and then
comes back to life, dmesg remains empty. Otherwise I would have already reported that. So it looks more like 
a hardware issue.

superm1, 2020-06-29:
>As I wrote, I don't know what is the original firmware version the dock had. If I understand you correctly, the firmware upgrade
DellDockingStationFwUp_1.0.4_IE_03222018.exe corresponds to mst_3_10_002.cab. However, this also means that
the original factory firmware must haven even an earlier date. Perhaps it was the one that worked for me.

There are other firmwares in the dock that would be modified at the same time, but those are not currently supported in Linux.
Knowing the hardware architecture of the WD15 is why I was suggesting it's most likely to be the MST.

>Another possibility would be that using mst_3_10_002.cab doesn't really donwgrade everything to the level of
the 1.0.4 firmware package. This I obviously can't know.

I can suggest a workaround you can try though if you want to force a downgrade in some of the other firmwares.  I don't know this will actually work though.
Look at the README.md here: https://github.com/fwupd/firmware-dell.  Try to use the "old" exe file as an argument to that and keep your dock connected across the reboot.  See if that downgrades the other firmwares.  You can check from fwupdmgr get-devices output the other firmware versions (they should show up as fwupd devices I expect).

>I wouldn't really call this regression, since with the old firmware everything worked flawlessly. Moreover, I've already
experienced some i915 regressions in the past (not related to WD15) and there one always could see some dmesg
output appearing simultaneously with graphical artifacts. In this case, whenever the screen goes black and then
comes back to life, dmesg remains empty. Otherwise I would have already reported that. So it looks more like
a hardware issue.

I still do feel that the described behavior could be related to training or timing problems.  You might not see errors come up in dmesg at the logging level you're set.  Bumping up to a debugging level log level might show more information about what the driver is doing at this time.
I guess let's see if your downgrade works (fingers crossed).

vsht, 2020-06-29:
> There are other firmwares in the dock that would be modified at the same time, but flashing those are not currently supported in Linux.
Knowing the hardware architecture of the WD15 is why I was suggesting it's most likely to be the MST.

I see. I suppose that Windows utilities do not support any kind of downgrade, so `fwupd` is my only chance.

> I can suggest a workaround you can try though if you want to force a downgrade in some of the other firmwares. I don't know this will actually work though.
Look at the README.md here: https://github.com/fwupd/firmware-dell. Try to use the "old" exe file as an argument to that and keep your dock connected across the reboot. See if that downgrades the other firmwares. You can check from fwupdmgr get-devices output the other firmware versions (they should show up as fwupd devices I expect).

Many thanks, this sounds as something that might work. I'll try it once I get back home.

> I still do feel that the described behavior could be related to training or timing problems. You might not see errors come up in dmesg at the logging level you're set. Bumping up to a debugging level log level might show more information about what the driver is doing at this time.

Well, if the old firmware was supported via a quirk in i915 that fails with the new firmware, this might be so. I can try to 
increase the logging level, if nothing else helps. However, even in this case, this bug would be probably too rare
and low priority to be fixed asap. Effectively, downgrading the dock or getting another one with a sufficiently old firmware is the fastest way to get a working setup.

vsht, 2020-06-29:
Oh and by the way, I've just found that there is also `DellDockingStationFwUp_1.0.0.8_20161220_WD15.exe`
which is much earlier than `DellDockingStationFwUp_1.0.4_IE_03222018.exe`. With your useful advice regarding
`firmware-dell` I might have a chance to downgrade to that version from 2016. Let's see :)

superm1, 2020-06-29:
>I see. I suppose that Windows utilities do not support any kind of downgrade, so fwupd is my only chance.

They do support a downgrade function, but I don't know it's publicly documented. You can try to look at `/?` or `--help` output to see maybe I'm wrong.  It's not my place to publicly mention it if it's not already done so.

>Many thanks, this sounds as something that might work. I'll try it once I get back home.

The part that might fail is what ESRT GUID is put into your capsule header and the BIOS might not accept it.  So I'll be interested to hear your results.

>fastest way to get a working setup.

Sure.  Makes sense.  But for the greater good I would rather see a bug filed with all the relevant details to figure out what about i915 doesn't like changes in the dock firmware.

vsht, 2020-06-29:
> They do support a downgrade function, but I don't know it's publicly documented. You can try to look at /? or --help output to see maybe I'm wrong. It's not my place to publicly mention it if it's not already done so.

You are right. One can downgrade MST easily via `mst.exe -fd` although your other advice using
```
./contrib/firmware_packager/install_dell_bios_exe.py .DellDockingStationFwUp_1.0.0.8_20161220_WD15.exe
```
worked as well, no issues encountered. The problem is, however, that the MST firmware version 3.10.002 
(which doesn't work for me) seems to be the earliest  binary one can grab via the official channels 
(i.e. DELL website). In the MST release notes (the file inside SYNAPTICS dir when one unpacks the exe) this 
version is listed as 

```
[Ver3.10.002] 
   Generated on 10:35:42.65 Mon 04/25/2016 
   Modified items: 
	1. Query the DP link status if HPD long pulse received and aux is alive
	2. Change the configuratoin file based DPCD 5h setting for MAC system to enable 4K output
	3. Fix the 3 streams on single DP MST TX maximal bandwidth issue
```

Since WD15 was released somewhere around March 2016 or so, I suppose that the factory
firmware was something like 3.10.000  or 3.09.xxx. But there is obviously no way I can 
get it. Even if I had another WD15 with the factory firmware, I have no idea how to grab it,
although I'm now able to flash older firmware versions via `mst.ext`. Well, tough luck.

> Sure. Makes sense. But for the greater good I would rather see a bug filed with all the relevant details to figure out what about i915 doesn't like changes in the dock firmware.

I agree. I guess I'll put it on my TODO list. For the moment, the workaround is to connect the screen directly via
HDMI/DP. Not very cool given that I need to handle both cables (HDMI and USB-c) when docking the machine,
but it works.

Anyway, many thanks for your help. I guess  there is nothing else one can do on the level of fwupd. Should I just
close the issue?

superm1, 2020-06-29:
> worked as well, no issues encountered.

When you say this worked well - can you please describe your experience?  On the reboot you saw the BIOS flasher take over, and it took a long time?

>The problem is, however, that the MST firmware version 3.10.002
(which doesn't work for me) seems to be the earliest binary one can grab via the official channels
(i.e. DELL website). In the MST release notes (the file inside SYNAPTICS dir when one unpacks the exe) this
version is listed as

I don't think WD15 launched with something less than `3.10.002`.

>I agree. I guess I'll put it on my TODO list. For the moment, the workaround is to connect the screen directly via
HDMI/DP. Not very cool given that I need to handle both cables (HDMI and USB-c) when docking the machine,
but it works.

I'm assuming you've already tried the simple debugging steps like to remove power from the dock?

> Anyway, many thanks for your help. I guess there is nothing else one can do on the level of fwupd. Should I just
close the issue?

I'd be interested if you could share your `fwupdmgr get-devices` output from before/after the upgrade/downgrade if possible.  I'd like to see what actually changed in your upgrade/downgrade.

superm1, 2020-06-29:
>   Generated on 10:35:42.65 Mon 04/25/2016 

Hmm, I'll look around to see if there was anything earlier I can find.

vsht, 2020-06-29:
> When you say this worked well - can you please describe your experience? On the reboot you saw the BIOS flasher take over, and it took a long time?

Exactly, BIOS flasher with the message 
```

Updating your firmware
Do not power down your system
Flash Progrress [---------------------------------]
(Part 1 of 1) Wired Dock EC
```
I think that it took something like 10 minutes or so. The experience was very smooth, no issues, no freezes.

The only weird thing is that the UEFI capsule is marked as a firmware update for the laptop, not the dock

```
$ fwupdmgr get-details /tmp/tmphrjiehgo/firmware.cab                                                                                                                                                     

Entpacken …              [***************************************]
Latitude 5480
│
└─System Firmware:
  │   Device ID:           348fa58069868c96025a2416f1a1024667045bdd
  │   Current version:     1.17.1
  │   GUID:                49e03513-7afe-498a-9d03-89e5ba1da865
  │   Device Flags:        • Internal device
  │                        • Updatable
  │                        • Requires AC power
  │                        • Supported on remote server
  │                        • Needs a reboot after installation
  │                        • Cryptographic hash verification is available
  │                        • Device is usable for the duration of the update
  │ 
  └─New firmware:
        New version:       1.17.1
        Zusammenfassung:   Unknown
        License:           Proprietary
        Size:              2,2 MB
        Created:           2020-06-29
        Vendor:            Dell Inc
```

But I guess that an average user won't notice that anyway.

> I'm assuming you've already tried the simple debugging steps like to remove power from the dock?

Yes. I've heard that the dock is quite complex inside and needs to "boot" when attached to a power supply.
So power cycling was one of the first things I tried.

> I don't think WD15 launched with something less than 3.10.002.

That's interesting. So you are thinking that my issues might come from a different component of the dock?
The stuff inside the `INTEL` folder seems to be related to Thunderbolt, so I don't think that it's relevant. 
ASMedia is probably all about the USB and ethernet, as far as I understand.

> I'd be interested if you could share your fwupdmgr get-devices output from before/after the upgrade/downgrade if possible. I'd like to see what actually changed in your upgrade/downgrade.

Sure. This is the current status after the downgrade via `DellDockingStationFwUp_1.0.0.8_20161220_WD15.exe`

```
Latitude 5480
│
├─Dell WD15:
│ │   Device ID:           23dac160329bd372d6e661813cc277f0205753ee
│ │   Zusammenfassung:     A USB type-C docking station
│ │   Vendor:              Dell Inc. (PCI:0x1028)
│ │   GUID:                e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:        • Requires AC power
│ │ 
│ ├─Dell WD15 EC:
│ │     Device ID:         c8ca45eaa670a8cca0123ac84e23188a8907d432
[... 47 lines cut ...]
│     GUID:                49e03513-7afe-498a-9d03-89e5ba1da865
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM 2.0:
│     Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│     Zusammenfassung:     Platform TPM device
│     Current version:     1.3.2.8
│     Vendor:              Dell Inc. (PCI:0x1028)
│     Update Error:        Updating disabled due to TPM ownership
│     GUIDs:               8ced72be-20bc-5e7f-a4f3-47838492f8a5
│                          ff71992e-52f7-5eea-94ef-883e56e034c6
│                          dd853888-f2c5-5a5d-b0c0-a068bdd2e6c8
│                          dc37fd1f-2db4-5bae-ab0a-20e268a2e918
│     Device Flags:        • Internal device
│                          • Requires AC power
```

I can try to upgrade using `DellDockingStationFwUp_1.0.8_WD15_20200207.exe` tomorrow to provide more output. After
all, upgrading/downgrading seems to be pretty "painless" with this dock.

superm1, 2020-06-30:
> I think that it took something like 10 minutes or so. The experience was very smooth, no issues, no freezes.

Well that's great to hear, thanks!

> The only weird thing is that the UEFI capsule is marked as a firmware update for the laptop, not the dock

Yeah this is expected based on how it actually ran the update. 

>That's interesting

I might be wrong, because it might have been a shortly after launch thing, or a near day 0 update.  Are there any logs saved anywhere from the Windows updater you can find still?  In the root of `C` or in the `dell` directory anywhere?  I would really like to know what version you started with.

> That's interesting. So you are thinking that my issues might come from a different component of the dock?
The stuff inside the INTEL folder seems to be related to Thunderbolt, so I don't think that it's relevant.
ASMedia is probably all about the USB and ethernet, as far as I understand.

That updater is both for WD15 and TB16.  Some of the components and architecture are shared between them.

> I can try to upgrade using DellDockingStationFwUp_1.0.8_WD15_20200207.exe tomorrow to provide more output. After
all, upgrading/downgrading seems to be pretty "painless" with this dock.

That would be great, especially considering you have the same problem on both versions.  Let me explain what is actually happening:
* MST update happens in "OS".  This is both for Windows and Linux.  The MST update you saw on LVFS is the same one in that EXE.
* EC, Passive cable, and port controller all update in the UEFI updater.

vsht, 2020-06-30:
> I might be wrong, because it might have been a shortly after launch thing, or a near day 0 update. Are there any logs saved anywhere from the Windows updater you can find still? In the root of C or in the dell directory anywhere? I would really like to know what version you started with.

Unfortunately, I managed to overwrite the original logs when trying to downgrade some time ago by running various
versions of `DellDockingStationFwUp_XYZ`. Otherwise, I would have stated the working firmware versions at the very beginning.

It appears that

`./contrib/firmware_packager/install_dell_bios_exe.py ../DellDockingStationFwUp_1.0.8_WD15_20200207.exe`

doesn't really work. The laptop reboots, but the UEFI screen shows only "Firmware Update Failed. Rebooting your system".
The dock still works, and the firmware versions remain unchanged, so I no harm is done. Updating via W10
worked flawlessly after this failed attempt.

It seems that what I managed to downgrade so far were only MST and the EC
[image]
The port controller and the cable firmware remained unaffected. I'm wondering if they could be the culprits here. 

The current output of `fwupdmgr get-devices` is

```
Latitude 5480
│
├─Dell WD15:
│ │   Device ID:           23dac160329bd372d6e661813cc277f0205753ee
│ │   Zusammenfassung:     A USB type-C docking station
│ │   Vendor:              Dell Inc. (PCI:0x1028)
│ │   GUID:                e7ca1f36-bf73-4574-afe6-a4ccacabf479
│ │   Device Flags:        • Requires AC power
│ │ 
│ ├─Dell WD15 EC:
│ │     Device ID:         c8ca45eaa670a8cca0123ac84e23188a8907d432
│ │     Zusammenfassung:   A USB type-C docking station
│ │     Current version:   0.0.16
│ │     Vendor:            Dell Inc. (PCI:0x1028)
│ │     GUID:              e8445370-0211-449d-9faa-107906ab189f
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │   
│ ├─Dell WD15 Passive Cable:
│ │     Device ID:         c3069dadc9611c8733685eaf20ca2097afc98391
│ │     Zusammenfassung:   A USB type-C docking station
│ │     Current version:   0.3.18
│ │     Vendor:            Dell Inc. (PCI:0x1028)
│ │     GUID:              fece1537-d683-4ea8-b968-154530bb6f73
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │   
│ ├─Dell WD15 Port Controller 1:
│ │     Device ID:         75770fca48031e08c23ec31cf16ca8082a1d9932
│ │     Zusammenfassung:   A USB type-C docking station
│ │     Current version:   1.1.8
│ │     Vendor:            Dell Inc. (PCI:0x1028)
│ │     GUID:              8ba2b709-6f97-47fc-b7e7-6a87b578fe25
│ │     Device Flags:      • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │   
│ └─VMM3332 inside Dell WD15/TB16/TB18 wired Dock:
│       Device ID:         d762543f8c20f636e6fff031a000078d3e10c600
│       Zusammenfassung:   Multi-Stream Transport Device
│       Current version:   3.12.02
│       Vendor:            Synaptics (DRM_DP_AUX_DEV:0x06CB)
│       GUIDs:             c1e81f96-234e-597d-b9cb-6477d0652868
│                          78c73924-821a-5e6f-af82-86df6f67b182
│                          c1e2bea5-4cf2-5f84-b965-533e8e5fea45
│                          02b77ccb-03ab-521f-b98e-ef05395e9fa2
│                          82460515-30d3-5cdf-be4c-571340842a10
│                          a6f00ca7-54b7-5d51-9d6e-543d8df85376
[... 41 lines cut ...]
│                          ff71992e-52f7-5eea-94ef-883e56e034c6
│                          dd853888-f2c5-5a5d-b0c0-a068bdd2e6c8
│                          dc37fd1f-2db4-5bae-ab0a-20e268a2e918
│     Device Flags:        • Internal device
│                          • Requires AC power
│   
└─TS240GMTS420S:
      Device ID:           612faa85309fec62ddfdce9f4635cdfef838d74e
      Zusammenfassung:     ATA Drive
      Current version:     Q1121A0
      Vendor:              Transcend (ATA:0x8564)
      GUIDs:               8c8e6158-ac10-5921-8127-2001c45057ec
                           38dd6336-d97f-5f0e-88b2-107e52ae425a
                           64f21124-e915-5820-821d-13dab60d2aef
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

```

I also noticed that lowering the resolution on Dell U2515H to FullHD doesn't help to remove
the glitches. It really looks like the monitor itself cannot properly communicate with the grahpics/dock
on Linux after the firmware update.

superm1, 2020-07-01:
>./contrib/firmware_packager/install_dell_bios_exe.py ../DellDockingStationFwUp_1.0.8_WD15_20200207.exe doesn't really work. 

Did it work once and then not another time?  That's super confusing, because you described a 10 minute update which is kinda what I expected with the bus and protocol is uses to do the update.

>It seems that what I managed to downgrade so far were only MST and the EC
>The port controller and the cable firmware remained unaffected. I'm wondering if they could be the culprits here.

It's also possible the other devices didn't change.

>I also noticed that lowering the resolution on Dell U2515H to FullHD doesn't help to remove
the glitches. It really looks like the monitor itself cannot properly communicate with the grahpics/dock
on Linux after the firmware update.

Shot in the dark - you already tried replacing with another cable?  I know it worked well and all before and in Windows, but it could be a more sensitive timing configuration by Linux i915.

If that shot in the dark didn't help and neither did all these upgrade and downgrades, I think it's time to file an issue against i915 with all the details you can.  Doing the test on DRM tip and with verbose DRM logging and such will hopefully be helpful.

Sorry there wasn't much we could do from fwupd side.  Please mention the bug number though if/when you file it, I would certainly follow along.

closed 2020-07-01 by superm1

vsht, 2020-07-03:
> Did it work once and then not another time? That's super confusing, because you described a 10 minute update which is kinda what I expected with the bus and protocol is uses to do the update.

In the meantime I did some more experiments and here's what I found.

First of all, it's pretty easy to initiate a downgrade using Dell's official firmware update package. Just run it from a PowerShell
(with admin permissions) and add the switch '/f'. For example 
```
DellDockingStationFwUp_1.0.6_12052018_WD15.exe /f
```
forces a downgrade using firmware binaries contained in the v. 1.0.6 update package. BTW, this is [documented](https://www.dell.com/support/article/de-de/sln82511/command-line-switches-for-dell-bios-updates?lang=en) so the information is free to share without fearing legal consequences.

My problem is, however, that I cannot do a complete downgrade using `DellDockingStationFwUp_1.0.6_12052018_WD15.exe`
or `DellDockingStationFwUp_1.0.4_IE_03222018.exe`. In both cases during the UEFI phase there are 3 firmware update steps to be completed

1. Wired Dock EC
2. Wired Dock USB PD
3. Cable USB PD

Steps 1. and 2. finish successfully but when it comes to step 3, I get `Cable USB PD update failed` and then immediately
`Extended EC update failed`. In the case of the `Cable USB PD` the progress indicator doesn't even change from 0%, it just 
fails immediately. This way the firmware versions remain as they are and nothing changes.

I'm still curious whether it would be possible to downgrade the `Wired Dock USB PD` separately which is something I didn't do yet (in my previous attempts I got only to `Wired Dock EC` and the `MST`). Or perhaps initiate a downgrade and then remove 
the `Cable USB PD` update capsule from the efi partition so that the downgrade can succeed. I tried this several times (also cold-booted) and the outcome is always the same.

> If that shot in the dark didn't help and neither did all these upgrade and downgrades, I think it's time to file an issue against i915 with all the details you can. Doing the test on DRM tip and with verbose DRM logging and such will hopefully be helpful.

I'm planning to submit a bug report, but even then having an ability to switch between a faulty and a working configuration would help to resolve the bug much faster.

It's interesting that when both the laptop and WD15 are  cold-booted, the situation looks good (no blackouts, very rare glitches) for the first 5-10 minutes, but then it deteriorates and turns to the usual unusable behavior. So to me it looks like some variables/arrays are overflowing and this become worse the longer the system is working.

vsht, 2020-08-07:
Just as a hint for people facing the same issue:

@superm1 was right that this behavior is caused by the i915 driver and _not_ WD15.

In fact one of the recent kernel updates few weeks ago even broke my other setup at work, with two
Full HD monitors connected to WD15. They started to flicker the same way as my U2515H
at home, unless I connect them directly to the laptop. 

However, few days ago I switched to a ThinkPad T14 AMD (not because of this issue) and was very surprised to
see that all issues with WD15 were gone. Now I again have all my screens (both at work
and at home) connected to a WD15 that is connected to my T14 via a single USB-C cable.
Everything is working as expected without flickering and/or black screens.

There seem to be [some](https://gitlab.freedesktop.org/search?utf8=%E2%9C%93&search=flickering&group_id=2642&project_id=4519&scope=issues&search_code=false&snippets=false&repository_ref=&nav_source=navbar) bugs in the i915 bugtracker that might be related, but since I'm planning to 
sell the Latitude 5480 soon, I can't help here. 

Thanks again to @superm1 for all his help.