fwupd/fwupd #2237: "efivars support missing: /sys/firmware/efi/efivars" on a BIOS computer

joakim-tjernlund, 2020-06-30:
**Describe the bug**
Multiple warnings on a BIOS only machine:

```
FuPluginUefi         kernel efivars support missing: /sys/firmware/efi/efivars
```
**Steps to Reproduce**
Steps to reproduce the behavior.

fwupdtool install <some fw cab>

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4

**Additional questions**
- Operating system and version: Gentoo
- Have you tried rebooting? No
- Is this a regression? Not sure

hughsie, 2020-06-30:
>  Multiple warnings on a BIOS only machine:
> FuPluginUefi         kernel efivars support missing: /sys/firmware/efi/efivars

What were you expecting?

joakim-tjernlund, 2020-06-30:
> > Multiple warnings on a BIOS only machine:
> > FuPluginUefi         kernel efivars support missing: /sys/firmware/efi/efivars
> 
> What were you expecting?

I guess I should have provided more context ...
As I was upgrading an USB device which has nothing to do with UEFI, I figured I should not have to see these.
See https://github.com/fwupd/jabra-firmware/pull/3

superm1, 2020-06-30:
Note; this is fixed in 1.5.0+, it is not planned to be backported to 1.4.x right now since 1.4.x actually makes a dummy device to tell you to switch modes from the uefi plugin, 1.5.x has a BIOS plugin for that.

closed 2020-06-30 by superm1

joakim-tjernlund, 2020-07-01:
> Note; this is fixed in 1.5.0+, it is not planned to be backported to 1.4.x right now since 1.4.x actually makes a dummy device to tell you to switch modes from the uefi plugin, 1.5.x has a BIOS plugin for that.

Sure, I am happy with that. Thanks.

PR #2240 (2020-06-30): uefi: disable plugin if efivar is not supported
Don't even try to coldplug the device.

Fixes: #2237

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 6a6029f1 (2020-06-30): uefi: disable plugin if efivar is not supported

Don't even try to coldplug the device.

Fixes: #2237