fwupd/fwupd #2261: Flatpak: will not run on Mint 20

LinuxOnTheDesktop, 2020-07-10:
To (try to?) reproduce: install Mint 20; after that, do and see the following.

```
$ flatpak install fwupd
Looking for matches…
Found similar ref(s) for ‘fwupd’ in remote ‘flathub’ (system).
Use this remote? [Y/n]: y
Found ref ‘app/org.freedesktop.fwupd/x86_64/stable’ in remote ‘flathub’ (system).
Use this ref? [Y/n]: y
Info: org.freedesktop.Platform.Icontheme.Paper is end-of-life, with reason: The Icontheme extensions are not used anymore.

org.freedesktop.fwupd permissions:
    network      devices      file access [1]      system dbus access [2]

    [1] /boot, /sys, xdg-download
    [2] org.freedesktop.UPower, org.freedesktop.fwupd

        ID                                     Branch           Op          Remote           Download
 1. [✓] org.freedesktop.fwupd.Locale           stable           i           flathub          3.2 kB / 1.8 MB
 2. [✓] org.freedesktop.fwupd                  stable           i           flathub          2.4 MB / 3.1 MB

Installation complete.
$ flatpak run org.freedesktop.fwupd
ldconfig: Can't create temporary cache file /run/ld-so-cache-dir/0ae87624ea5c37aef5b1af8b02de5d49a0d1a8543eeb68e20af02d14ac848e3c~: Permission denied
error: ldconfig failed, exit status 256
$ sudo flatpak run org.freedesktop.fwupd
error: "flatpak run" is not intended to be ran with sudo
```

**Postscript**

So I decided to try the repository version . .

```
$ sudo apt install fwupd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Recommended packages:
  bolt fwupd-signed
The following NEW packages will be installed
  fwupd
0 to upgrade, 1 to newly install, 0 to remove and 6 not to upgrade.
Need to get 0 B/2,089 kB of archives.
After this operation, 5,784 kB of additional disk space will be used.
Selecting previously unselected package fwupd.
(Reading database ... 444208 files and directories currently installed.)
Preparing to unpack .../fwupd_1.3.9-4ubuntu0.1_amd64.deb ...
Unpacking fwupd (1.3.9-4ubuntu0.1) ...
Setting up fwupd (1.3.9-4ubuntu0.1) ...
fwupd-offline-update.service is a disabled or a static unit not running, not starting it.
fwupd-refresh.service is a disabled or a static unit not running, not starting it.
fwupd.service is a disabled or a static unit not running, not starting it.
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for dbus (1.12.16-2ubuntu2.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
$ fwupdmgr
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.2' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.2' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.3' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.7' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.9' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.3.3' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.1' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.3.2' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.8' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.3.7' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.6' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.1' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.3.4' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.5' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.4' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.3.1' not found (required by fwupdmgr)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.3' not found (required by /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.1)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.4' not found (required by /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.1)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.2' not found (required by /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.1)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.5' not found (required by /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.1)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.2.9' not found (required by /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.1)
fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.3.6' not found (required by /usr/lib/x86_64-linux-gnu/libfwupdplugin.so.1)
```

hughsie, 2020-07-10:
> fwupdmgr: /usr/local/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.2' not found (required by fwupdmgr)

Did you also install the source version to /usr/local/? I think fwupdmgr is being sourced from /usr/local/ for some reason.

LinuxOnTheDesktop, 2020-07-10:
Not that I recall. How should I check and/or correct, please?

superm1, 2020-07-10:
If you have stuff in `/usr/local`, you've manually installed at some point.  Remove it to use your packaged stuff.

superm1, 2020-07-10:
And btw after you've cleared out your stuff in `/usr/local`, you probably need to run `ldconfig`

closed 2020-07-10 by superm1

LinuxOnTheDesktop, 2021-05-09:
I tried again to install the Flatpak version (in order to try to . . fix [another problem](https://github.com/fwupd/fwupd/issues/3231)). The Flatpak still will not run. Would you please advise? Here is what happened.

```
$ flatpak run org.freedesktop.fwupd
error: While opening repository /home/<myUserName>/.local/share/flatpak/repo: /home/<myUserName>/.local/share/flatpak/repo: opendir(/home/<myUserName>/.local/share/flatpak/repo): Permission denied
<myUserName>@x1 ~ $ sudo flatpak run org.freedesktop.fwupd
error: "flatpak run" is not intended to be ran with sudo
<myUserName>@x1 ~ $ flatpak run org.freedesktop.fwupd
error: While opening repository /home/<myUserName>/.local/share/flatpak/repo: /home/<myUserName>/.local/share/flatpak/repo: opendir(/home/<myUserName>/.local/share/flatpak/repo): Permission denied
<myUserName>@x1 ~ $ chown -R <myUserName>:<myUserName> /home/<myUserName>/.local/share/flatpak/repo
chown: cannot read directory '/home/<myUserName>/.local/share/flatpak/repo': Permission denied
<myUserName>@x1 ~ $ sudo chown -R <myUserName>:<myUserName> /home/<myUserName>/.local/share/flatpak/repo
<myUserName>@x1 ~ $ flatpak run org.freedesktop.fwupd
error: mkdir(/home/<myUserName>/.var/app/org.freedesktop.fwupd/data): Permission denied
<myUserName>@x1 ~ $ sudo chown -R <myUserName>:<myUserName> /home/<myUserName>/.var/app/org.freedesktop.fwupd
<myUserName>@x1 ~ $ flatpak run org.freedesktop.fwupd
This program may only work correctly as root
Command not found

Use fwupdtool --help for help
```

LinuxOnTheDesktop, 2021-05-09:
And now . . this, which is . . sub-optimal. Note that I had installed the Flatpak, just now, via [Mint's (horribly broken?) Software Manager](https://github.com/linuxmint/mintinstall/issues/249?_pjax=%23js-repo-pjax-container).

```
$ flatpak remove org.freedesktop.fwupd

        ID                                   Branch         Op
 1. [-] org.freedesktop.fwupd                stable         r
 2. [-] org.freedesktop.fwupd.Locale         stable         r

Uninstall complete.
<myUserName>@x1 ~ $ flatpak install org.freedesktop.fwupd
Looking for matches…
Found similar ref(s) for ‘org.freedesktop.fwupd’ in remote ‘flathub’ (system).
Use this remote? [Y/n]: 
Info: org.freedesktop.Platform.Icontheme.Paper is end-of-life, with reason: The Icontheme extensions are not used anymore.

org.freedesktop.fwupd permissions:
    network      devices      file access [1]      system dbus access [2]

    [1] /boot, /sys, xdg-download
    [2] org.freedesktop.UDisks2, org.freedesktop.UPower, org.freedesktop.fwupd

        ID                                     Branch           Op          Remote           Download
 1.     org.freedesktop.fwupd.Locale           stable           i           flathub          < 344.6 kB (partial)
 2.     org.freedesktop.fwupd                  stable           i           flathub            < 4.9 MB

Proceed with these changes to the system installation? [Y/n]:

Info: org.freedesktop.Platform.Icontheme.Paper is end-of-life, with reason: The Icontheme extensions are not used anymore.

org.freedesktop.fwupd permissions:
    network      devices      file access [1]      system dbus access [2]

    [1] /boot, /sys, xdg-download
    [2] org.freedesktop.UDisks2, org.freedesktop.UPower, org.freedesktop.fwupd

        ID                                     Branch           Op          Remote           Download
 1. [✓] org.freedesktop.fwupd.Locale           stable           i           flathub          3.3 kB / 344.6 kB
 2. [✓] org.freedesktop.fwupd                  stable           i           flathub          4.1 MB / 4.9 MB

Installation complete.
$ flatpak run org.freedesktop.fwupd
This program may only work correctly as root
Command not found

Use fwupdtool --help for help
$ sudo flatpak run org.freedesktop.fwupd
error: "flatpak run" is not intended to be ran with sudo

```

LinuxOnTheDesktop, 2021-05-09:
Cf. [this report of woe](https://forums.linuxmint.com/viewtopic.php?t=297476), by someone else, and where no remedy was forthcoming, on the Linux Mint forum.

hughsie, 2021-05-10:
> error: "flatpak run" is not intended to be ran with sudo

I think you want `sudo -i flatpak run ...`

LinuxOnTheDesktop, 2021-05-10:
I lack the time to test this. For, I removed the Flatpak. But: presuming you are right, then you need to have _instructions_ that say that. Others should not have to waste as much time as I did.

hughsie, 2021-05-10:
> then you need to have instructions that say that

Where did you read the flatpak instructions from, out of interest?

LinuxOnTheDesktop, 2021-05-10:
I read no instructions. Do you provide any?