fwupd/fwupd #2263: fwupd-refresh.timer failes repeatedly

LinuxOnTheDesktop, 2020-07-12:
**Describe the bug**

fwupd-refresh.timer fails repeatedly.

Relevant systemd output:
```
  UNIT                LOAD   ACTIVE SUB    DESCRIPTION                     
● fwupd-refresh.timer loaded failed failed Refresh fwupd metadata regularly
```

Syslog: `fwupd-refresh.timer: Failed to queue unit startup job: Unit tmp.mount is masked.`

Resetting the service works but the problem soon reoccurs.

**Steps to Reproduce**

Unknown. Possibly just: install fwup from Mint 20 Cinnamon's repos. I would test on a Mint 20 VM but . . VirtualBox (on Windows) has stopped working.

**Expected behavior**

No such failure.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.9
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.9
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): apt

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20KHCTO1WW
│
├─Thunderbolt Controller:
│     Device ID:           eded407096031a81370bc25ae01650697a7c72b6
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     43.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               2da42a33-cd30-5ef5-a8fb-2c800a4b760f
│                          96411853-9f56-53d2-b05c-a54fcad5ee62
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─LENSE20512GMSP34MEAT2TA:
│     Device ID:           04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:             NVM Express Solid State Drive
│     Current version:     2.8.8341
│     Vendor:              Lenovo (NVME:0x17AA)
│     GUIDs:               963522d7-bc73-51fa-a70c-cf615f42bcf6
│                          b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
│                          ef93a17b-97f6-550e-a5bc-32c22ff6cdf8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           1c53551e7da69d896138fac1ae131c83ad46d923
│     Current version:     0.1.49
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                a4b51dca-8f97-4310-8821-3330f83c9135
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     73.4.17568.4452
│ │   Vendor:              ST Microelectronics (TPM:STM)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        84df3581-f896-54d2-bd1a-372602f04c32
│ │                        bfaed10a-bbc1-525b-a329-35da2f63e918
│ │                        70b7b833-7e1a-550a-a291-b94a12d0f319
│ │                        06f005e9-cb62-5d1a-82d9-13c534c53c48
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       Vendor:            ST Microelectronics (TPM:STM)
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─UEFI Device Firmware:
│     Device ID:           29749712e93b58730ecd1b8537a5503ff5b7fb27
│     Current version:     184.77.3664
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                9c9d9769-32fa-4841-b550-ea998e754e99
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9698faabddf0d7b18925cfbbda95f8b0d0dacc53
│     Current version:     0.1.20
│     Minimum Version:     0.1.20
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           9e329270a7a68d289c82fe77d32d02208ddf0890
│     Current version:     0.73.4
│     Minimum Version:     0.73.4
│     Vendor:              DMI:LENOVO
│     GUID:                cea87551-1701-43fb-afbc-6e8ce9728345
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UHD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     07
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               fed2efa4-7045-55a1-b4fc-b29283d59fe5
│                          8b72a10c-1279-5f8e-a28a-34fb11a58240
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR24.10_B0036
      Bootloader Version:  BOT03.01_B0008
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
```

**Additional questions**
- Operating system and version: Mint 20 Cinnamon
- Have you tried rebooting? Yes.
- Is this a regression? Unknown. (Previously I was using the Snap and then Flatpak version of fwupd.)

superm1, 2020-07-13:
Something on your system (maybe you) has masked `tmp.mount`:
https://www.mail-archive.com/systemd-devel@lists.freedesktop.org/msg43023.html

LinuxOnTheDesktop, 2020-07-13:
Thanks. However, I do not think I have done that. For, I can find no relevant symblink in my systemd directory (/etc/systemd/system) and there is no mention of /tmp in this list:

```
$ sudo systemctl list-unit-files --state=masked
UNIT FILE                           STATE  VENDOR PRESET
alsa-utils.service                  masked enabled      
bootlogd.service                    masked enabled      
cgroup-lite.service                 masked enabled      
cryptdisks-early.service            masked enabled      
cryptdisks.service                  masked enabled      
hwclock.service                     masked enabled      
lvm2-monitor.service                masked enabled      
ModemManager.service                masked enabled      
NetworkManager-wait-online.service  masked enabled      
pppd-dns.service                    masked enabled      
pulseaudio-enable-autospawn.service masked enabled      
rc.service                          masked enabled      
rcS.service                         masked enabled      
samba-ad-dc.service                 masked enabled      
saned.service                       masked enabled      
smartd.service                      masked enabled      
sudo.service                        masked enabled      
systemd-rfkill.service              masked enabled      
ureadahead.service                  masked enabled      
x11-common.service                  masked enabled      
systemd-rfkill.socket               masked enabled      
alsa-utils.service                  masked enabled      
bootlogd.service                    masked enabled      
cgroup-lite.service                 masked enabled      
cryptdisks-early.service            masked enabled      
cryptdisks.service                  masked enabled      
hwclock.service                     masked enabled      
lvm2-monitor.service                masked enabled      
ModemManager.service                masked enabled      
NetworkManager-wait-online.service  masked enabled      
pppd-dns.service                    masked enabled      
pulseaudio-enable-autospawn.service masked enabled      
rc.service                          masked enabled      
rcS.service                         masked enabled      
samba-ad-dc.service                 masked enabled      
saned.service                       masked enabled      
smartd.service                      masked enabled      
sudo.service                        masked enabled      
systemd-rfkill.service              masked enabled      
ureadahead.service                  masked enabled      
x11-common.service                  masked enabled      
systemd-rfkill.socket               masked enabled      
fstrim.timer                        masked enabled    
```

Still, I do have the following in my fstab:

`tmpfs	/tmp	tmpfs	defaults,noatime,size=11%,mode=1777,nosuid,nodev	0	0`

superm1, 2020-07-13:
Follow the directions in that link.

" "masked" means there is a symlink with the name of *.mount unit, 
in one of the unit search paths (listed in "man systemd.unit"),
pointing to /dev/null.
  You have to: 1) remove the symlinks; 2) figure what creates those
symlinks on your system.
"

superm1, 2020-07-13:
And if you can't find the symlinks, then I think you should raise this to systemd team.

johannbg, 2020-07-21:
This is neither an fwupd issue nor systemd. The reporter basically needs to temporary disable private temp in the unit file for fwupd ( set PrivateTmp=false ) and figure out what's causing this. It could be masked in *any* of the search path as has been mentioned or could be bugs like these [1] or packaging issue in the distro/snaps/flatpack etc. it could even be an image issue if he's running of an usb stick or iso image or something ( if mint is not actually installed ) in any case a downstream issue so this issue could just be closed "NOTABUG". 

1. https://bugs.debian.org/cgi-bin/bugreport.cgi?att=0;bug=948739;msg=26

LinuxOnTheDesktop, 2020-07-21:
Thanks, johannbg. I have managed to rid one of my computers of the problem - by, I think, completely removing fwupd and all associated service and timer files. But the problem persists on another computer. (Both computers run installed versions of Mint on which fwupd is installed via the apt repositories. On both computers it used to be installed as a - problematic - flatpak and, before that, as a snap.) If I continue to have trouble with the other computer then I will do as you suggest. Why though do I need to set that PrivateTmp value to false? What will so doing do, please?

johannbg, 2020-07-22:
When you set an existing PrivateTmp value in type units to false you are disabling namespace mountpoint for the unit.
If people/distribution have decided to manage /tmp in fstab by themselves instead of letting systemd just handle it for them by adding an entry like "tmpfs   /tmp         tmpfs   rw,nodev,nosuid,size=1G          0  0" to /etc/fstab and then mask the tmp.mount type unit as an result of deciding to manage /tmp by themselves, they should also include an tmp.conf in the /etc/tmpfiles.d/ directory which excludes namespace mountpoints from removal so they dont break PrivateTmp=yes in type unit files. 
Basically you add your own configuration ( man tmpfiles.d for details how to do that ) or just copy whatever is in /usr/lib/tmpfiles.d/tmp.conf and add that +
x /tmp/systemd-private-*
x /var/tmp/systemd-private-*
X /tmp/systemd-private-*/tmp
X /var/tmp/systemd-private-*/tmp
( these are two small x and two large X man tmpfiles.d for details what they do ) into /etc/tmpfiles.d/tmp.conf

LinuxOnTheDesktop, 2020-07-22:
Thanks again, johannbg.

I am inclined to see the situation not as me messing with systemd but rather as systemd usurping something that previously was left up to the user.

For a quiet, life, though, I will just remove my `tmpfs /tmp` line from fstab.

LinuxOnTheDesktop, 2022-03-23:
I find that I still have the problem despite implementing the workaround that johannbg gave. Still, I have a fairly old version of fwupd - namely, 1.7.5.3. (That is the latest version in my distro's repos. I was using the flatpak version of a while but, in search of simplicity, I removed it.) I think I will try masking the timer (via `sudo systemctl mask --now fwupd-refresh.timer`).

superm1, 2022-04-25:
fairly certain the reincarnation of this bug opened a month ago is actually https://github.com/fwupd/fwupd/issues/3037.  I have a PR under review for that.  Let's wait until that lands somewhere you can try and if it doesn't fix it for you re-open at that time with new logs and such.

closed 2022-04-25 by superm1

LinuxOnTheDesktop, 2022-11-27:
A note for anyone else who has employed @johannbg's (now obsolete, I take it) workaround.

It seems that having a /etc/tmpfiles.d/tmp.conf file that comprises the lines that johannbg gave - or even, surprisingly, a _blank_ version of that file - _will prevent rebooting from ever deleting **any** /tmp files_. Eventually - when it had begun to cause problems - I noticed this happening on my systems. The solution is to delete the conf file.