fwupd/fwupd #2265: Intel BootGuard reported as Invalid on platform with BootGuard enabled

dnoliver, 2020-07-16:
**Describe the bug**
A clear and concise description of what the bug is.

The `fwupdmgr security --force` report says that Intel BootGuard is invalid, in a CompuLab Fitlet2
The OEM says that Intel BootGuard is correctly enabled in the device. 

After inspecting the test code in here: https://github.com/fwupd/fwupd/commit/6269a839ebf770ecbef8700656426c0fe338f7bb, I can see that the reason why the report is showing invalid can be:
1.	Verified Boot is not enabled or,
2.	ACM protection is not required or,
3.	Policy is not “immediately shutdown” or,
4.	The vendor did not set the FPF OTP fuse

Is there any way to have a more granular report on why `fwupdmgr` is saying that Boot Guard is disabled?

```shell
[test@compulab-fitlet2-1181004-04904 ~]$ rpm -qa fwupd
fwupd-1.5.0-0.603.20200706git.fc32.x86_64

[test@compulab-fitlet2-1181004-04904 ~]$ fwupdmgr --version
client version: 1.4.1-247-gfb0a938f
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.1-247-gfb0a938f

[test@compulab-fitlet2-1181004-04904 ~]$ sudo fwupdmgr security --force
Host Security ID: HSI:1+U (v1.5.0)

HSI-1
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ TXE manufacturing mode:        Locked
✔ TXE override strap:            Locked
✔ TXE v0:3.1.70.2334:            Valid
✔ UEFI dbx:                      Found

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid
✘ Intel BootGuard:               Invalid

HSI-3
✔ Suspend-to-ram:                Disabled
✘ Intel CET:                     Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✔ Firmware updates:              Supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ UEFI secure boot:              Enabled
✔ fwupd plugins:                 Untainted

Host Security ID attributes uploaded successfully, thanks!
```

**Steps to Reproduce**
Steps to reproduce the behavior.

1. Install Fedora 32 in a CompuLab Fitlet2
2. Install the fwupd package from copr
3. Run the test

**Expected behavior**
A clear and concise description of what you expected to happen.

I am not sure if it is false positive or device misconfiguration, so not sure what I should expect

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

```shell
[test@compulab-fitlet2-1181004-04904 ~]$ fwupdmgr --version
client version: 1.4.1-247-gfb0a938f
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.1-247-gfb0a938f
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

```shell
dnf copr enable rhughes/fwupd
dnf update
dnf update fwupd
```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

```shell
[test@compulab-fitlet2-1181004-04904 ~]$ fwupdmgr get-devices --show-all-devices
fitlet2
│
├─Celeron N3350/Pentium N4200/Atom E3900 Series Integrated Graphics Controller:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     0b
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               7ee0c805-ca7f-51ea-8bbd-e20e623b9323
│                          8257b546-4d2d-5c32-adc5-fc54b8fdfe53
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─Intel(R) Atom™ Processor E3950 @ 1.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x38
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─System Firmware:
│     Device ID:           11d8ed06243abd8103ab13e360847338b4149b1b
│     Current version:     4638
│     Minimum Version:     4638
│     Vendor:              Compulab (DMI:Compulab)
│     GUIDs:               49ff2877-8596-4460-9a97-e70d90609e25
│                          230c8b18-8d9b-53ec-838b-6cfc0383493a
│                          cc319b71-9b87-5791-afbd-744feb22923c
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Summary:             TPM 2.0 Device
│ │   Current version:     301.9.0.0
│ │   Vendor:              Intel (TPM:INTC)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        34801700-3a50-5b05-820c-fe14580e4c2d
│ │                        8e1cbc5d-5a11-5149-bfea-b6065d5296ba
│ │                        03f304f4-223e-54f4-b2c1-c3cf3b5817c6
│ │                        52d7b679-db28-5bf7-bd87-41d77aeec600
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─TS128GMTS600:
│     Device ID:           c46944fe84f2b057692258cc87a812cc86364881
│     Summary:             ATA Drive
│     Current version:     P1225CH1
│     Vendor:              Transcend (ATA:0x8564)
│     GUIDs:               6001e5d6-33a9-5b6c-9884-b461cdc2cf32
│                          920180a6-f62c-564a-b6d2-1fb6d7decd3e
│                          e25079dc-149b-5eb7-aee9-c7ebe27e4fd3
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           f3051add1ff6b15831703b30f1782eff9b2173ee
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
                           279ed287-3607-549e-bacc-f873bb9838c4
      Device Flags:        • Updatable
                           • Supported on remote server
```

**Additional questions**
- Operating system and version: Fedora Server 32
- Have you tried rebooting? Yes
- Is this a regression? Unknown

superm1, 2020-07-16:
Added a commit to that PR that should allow `--verbose` to show you why.

hughsie, 2020-07-16:
@superm1 I'm tempted to split these out from one HSI attribute to 4 attributes -- agree? Potentially they are different HSI severities and spitting them allows us to set the HSI level as required. For instance:

>    Verified Boot is not enabled or,

HSI:2

>    ACM protection is not required or,

HSI:2

>    Policy is not “immediately shutdown” or,

HSI:3?

>    The vendor did not set the FPF OTP fuse

HSI:2?

superm1, 2020-07-16:
I personally don't think the individual values matter; it's a false sense of security.  Sorta like if any part of the comprehensive chain are broken you're as weak as the weakest link.

hughsie, 2020-07-18:
>  Sorta like if any part of the comprehensive chain are broken you're as weak as the weakest link

Right, but my logic with splitting them up was to make it very clear where the failure was; if you're a small OEM it's not particularly easy to parse the failure. It also allows us server side to find out what platforms are not getting which bits right, for instance the FPF fuse failure is completely expected for a pre-production model.

superm1, 2020-07-18:
By that logic maybe it's worth actually capturing more smbios data too. You can tell if it's an old board or newer board if it's properly populated.

hughsie, 2020-07-20:
> to make it very clear where the failure was

I'd also like each HSI test (or subtest) to have a URL that the reporter can follow with more details about the failure -- end users might not be so interested in the failure details, but an ODM or ISV will. I do think something like `fwupd_security_attr_set_parent_id` makes sense to collapse them down in the case of all failures or all passes.

> By that logic maybe it's worth actually capturing more smbios data too

What can we look at to see "this is a pre-production laptop"? If we had some data I think a lot of the low-HSI laptops should be specially excluded on the LVFS using this metric.

superm1, 2020-07-20:
>What can we look at to see "this is a pre-production laptop"? If we had some data I think a lot of the low-HSI laptops should be specially excluded on the LVFS using this metric.

It's up to the OEM to properly populate it.  For Dell it's `/sys/class/dmi/id/board_version`

dnoliver, 2020-07-20:
I have compiled this version, and I can see now in the debug log the following line:

```shell
14:57:42:0990 FuPluginPciMei       Verified boot invalid
```

On other topic, I believe that it will be not possible for an Apollo Lake platform to be compliant with HSI 2 as it is today.
According to "Apollo Lake Platform - Intel®Trusted Execution Engine (Intel®TXE) Firmware Bring-Up Guide", there are only 3 Boot Guard profiles available in that TXE:

* Boot Guard Profile 0 - Legacy: in this profile Boot Guard boot block verification and measurement protection is off.
* Boot Guard Profile 1 - V: Strict Verification Enforcement. Prevents unverified bios components from running.
* Boot Guard Profile 2 - VM: Strict Verification and Measurement enforcement. Image Creation/Flashing Procedure. Prevents unverified Bios components from running.

From this presentation https://github.com/flothrone/bootguard/blob/master/Intel%20BG%20part2.pdf, slide 24, Boot Guard Profile 1 and 2 will be equivalent to VE and VM (I think), and this tool is requiring FVE and/or FVME (with immediately shutdown), which I think it is not supported by the TXE in this platform.

So, +1 now for this comment https://github.com/fwupd/fwupd/issues/2265#issuecomment-659656104 :)

superm1, 2020-07-20:
Yes, I agree with that approach too now.

closed 2020-07-20 by hughsie