fwupd/fwupd #2267: Failed to update metadata for lvfs: no signature method in results

GAZ082, 2020-07-19:
**Describe the bug**
`Failed to update metadata for lvfs: no signature method in results`

**Steps to Reproduce**
`sudo fwupdmgr get-updates`

**Expected behavior**
`Update drivers?`

**fwupd version information**
Please provide the version of the daemon and client.
```
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4
```
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`AUR https://www.archlinux.org/packages/community/x86_64/gnome-firmware/`

**fwupd device information**
```
P55-US3L
│
├─Intel(R) Core™ i5 CPU         760  @ 2.80GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x4
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─KINGSTON SA400S37240G:
│     Device ID:           f219ac034743df43be49a235e41f128dd47d855f
│     Summary:             ATA Drive
│     Current version:     SBFK71B1
│     Vendor:              Kingston (ATA:0x2646)
│     GUIDs:               4f924f77-2b5d-51f2-8bdb-f1a28eb885ff
│                          2a76fab0-59b3-53a9-bdc6-7ba2cc24ee37
│                          5bfd3e07-4ee5-5934-85e6-7aa10b2eca42
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Lexa PRO [Radeon 540/540X/550/550X / RX 540X/550/550X] (Lexa PRO [Radeon RX 550]):
      Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Current version:     c7
      Vendor:              Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
      GUIDs:               86aee031-c0ce-5ac8-9570-3fec122d4893
                           4daaf54e-4446-5b25-8979-7fc017051118
      Device Flags:        • Internal device
                           • Cryptographic hash verification is available
```

**Additional questions**
- Operating system and version:
Archlinux
Linux xiridio 5.7.9-arch1-1 #1 SMP PREEMPT Thu, 16 Jul 2020 19:34:49 +0000 x86_64 GNU/Linux
- Have you tried rebooting?
Yes.
- Is this a regression?
Dont know, first time using.

pjentsch0, 2020-07-19:
I am also getting this error.

Downloaded through `https://aur.archlinux.org/packages/gnome-firmware-git/`
```
client version:	1.4.4
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.4
```

```
OS: Arch Linux x86_64
Kernel: 5.7.9-zen1-1-zen
```

hughsie, 2020-07-19:
> no signature method in results

Did you compile libjcat without gpg or pkcs#7 support perhaps?

GAZ082, 2020-07-19:
@hughsie Will check with the package maintainer.

GAZ082, 2020-07-19:
@hughsie check the libjcat package build file and I guess it does not include PGP? Or perhaps it's in default and not needed  to include?

```
# Maintainer: Bruno Pagani

pkgname=libjcat
pkgver=0.1.3
pkgrel=1
pkgdesc="Library for reading and writing Jcat files"
arch=(x86_64)
url="https://github.com/hughsie/libjcat"
license=(LGPL)
depends=(gobject-introspection-runtime json-glib gnutls gpgme)
makedepends=(meson gobject-introspection gtk-doc vala help2man)
source=("https://people.freedesktop.org/~hughsient/releases/${pkgname}-${pkgver}.tar.xz"{,.asc})
sha256sums=('8b8585cc34f1b2e87d21c6eab8a1081cf6aa45294339bb5fca8a8437a17d2b73' 'SKIP')
validpgpkeys=(163EB50119225DB3DF8F49EA17ACBA8DFA970E17) # Richard Hughes <richard@hughsie.com>

build() {
    arch-meson ${pkgname}-${pkgver} build -D gtkdoc=true
    meson compile -C build
}

check() {
    meson test -C build --print-errorlogs
}

package() {
    DESTDIR="${pkgdir}" meson install -C build
    rm -r "${pkgdir}"/usr/{lib,share}/installed-tests/
}
```

superm1, 2020-07-20:
CC @ArchangeGabriel

n1rna, 2020-07-20:
Same problem here

kflak, 2020-07-20:
Same here. Arch Linux 5.6.19-rt11-1-rt, Lenovo Thinkpad T480s

ArchangeGabriel, 2020-07-20:
Did this issue started to appear y-day for all of you? Seems concomitant with the libjcat 0.1.3 update… But I’ve checked the build log again, nothing suspicious in it.

ArchangeGabriel, 2020-07-20:
@superm1 @hughsie Any instructions for running jcat-tool by hand on the downloaded DB (where?) to try to reproduce at a lower level?

hughsie, 2020-07-20:
> Seems concomitant with the libjcat 0.1.3 update

Can you try re-compiling fwupd against the new libjcat? IIRC, there was a subtle behaviour change which causes fwupd to fail if you mix jcat versions.

ArchangeGabriel, 2020-07-20:
> > Seems concomitant with the libjcat 0.1.3 update
> 
> Can you try re-compiling fwupd against the new libjcat? IIRC, there was a subtle behaviour change which causes fwupd to fail if you mix jcat versions.

Indeed fixing the issue. Pushing, should be available to all soon.

superm1, 2020-07-21:
Closing issue as it's a distro thing caused by mixing and matching libjcat

closed 2020-07-21 by superm1