fwupd/fwupd #2269: Incorrect efi title

johannbg, 2020-07-21:
Currently the efi title ( Linux-Firmware-Updater ) get's created with "-" as opposed to just regular space as it is written so end users end with an inconsistant efi boot menu that looks like this 
Linux Boot Manager ( sd-boot )
Linux-Firmware-Updater ( fw-update )
as opposed to consistent boot menu like this
Linux Boot Manager
Linux Firmware Updater 
( just a minor cosmetic issue )

There is also a question if fwupd should not claim and reside under it's own namespace in the EFI directory ( /EFI/lfu/fwupdx64.efi as opposed to /EFI/fedora/fwupdx64.efi ) which makes it distribution agnostic.

superm1, 2020-07-21:
Sounds like you're using Lenovo hardware.  Because of `$reasons` I don't understand, Lenovo can't handle "Linux Firmware Updater" and fwupd has to use "Linux-Firmware-Updater" hardcoded.  @hughsie might have more background of this.

Code for it is here:
https://github.com/fwupd/fwupd/blob/6a6029f132f684d1bf1159a667f4e3c746b9a797/plugins/uefi/fu-plugin-uefi.c#L518
https://github.com/fwupd/fwupd/blob/a11eab61e1e1cdcf8a349083dd3da9931515f235/plugins/uefi/fu-uefi-device.c#L648

> There is also a question if fwupd should not claim and reside under it's own namespace in the EFI directory ( /EFI/lfu/fwupdx64.efi as opposed to /EFI/fedora/fwupdx64.efi ) which makes it distribution agnostic.

Two reasons.
1. EFI binary needs to be signed by the distro to work with secure boot.
2. For shim chainloading to work it needs to be distributed in the same directory as shim (IIRC).

johannbg, 2020-07-21:
Yup I'm on Lenovo but it still has not problem with sd-boot labelling it as "Linux Boot Manager" so it could be model specific. Nr 1 should not be a problem nr.2 I believe is grub specific but I can test that with sd-boot and see if it works or not and where it breaks if it breaks.

hughsie, 2020-07-21:
> so it could be model specific

It's vendor firmware version specific. tl;dr: In some early Lenovo firmwares there is a list of allowed labels that are not "cleaned up" if they have the same path. `Linux-Firmware-Updater` is on that list, but `Linux Firmware Updater` isn't. Obviously this is wildly out of specification and in newer BIOS versions they stopped doing this (or perhaps they just added `Linux Firmware Updater` to the list...) but without knowing what models and versions are affected it's just much safer to use the old name for all Lenovo hardware.

johannbg, 2020-07-21:
Ok closing this issue since it's NOTABUG but a FEATURE. Will open separated bugs if I come across any issues running with fwupd and sd-boot.

closed 2020-07-21 by johannbg