fwupd/fwupd #2276: Thinkpad X1 Carbon Gen7: Update fails when trying to install multiple updates at once

mendel5, 2020-07-25:
The problem described in this issue happened to me almost a month ago. I'm sorry if it is lacking some details but maybe it is still helpful.

I have not used fwupdmgr on my laptop (Thinkpad X1 Carbon Gen7) for about 3 months. When I did it again there were four updates available. I remember one update was for the Thunderbolt controller and another one for some type of UEFI device. I also remember that two of the four updates specifically used UEFI update capsules.

I entered the following commands:
```
fwupdmgr refresh
fwupdmgr get-devices
fwupdmgr get-updates
fwupdmgr update
```

When fwupdmgr asked me whether to install all updates I pressed "Y" for all of the four updates. Then I restarted the computer to have the updates installed. When I ran the commands listed above again, two of the four updates have been successfully installed. The successful updates were the ones that did not use the UEFI update capsules. But the two updates using the UEFI update capsule were still shown as available so I concluded that they hadn't been installed.

The solution was to install both updates capsules on their own. First one of the capsule updates, then reboot, then the other one, then reboot, then it was done. Unfortunately I don't have any kind of output from fwupdmgr anymore.

I was expecting all four updates, two using UEFI update capsules and two without, to be installed with one single reboot. That was not possible so I first installed the two updates without capsules, rebooted, installed the first capsule, rebooted and then the second capsule. That was successful. Is it possible to install four updates, two of which are UEFI capsules, with just a single reboot?

hughsie, 2020-07-26:
> The solution was to install both updates capsules on their own

Do you remember the version you were coming from, i.e. the "old version"? `fwupdmgr get-history` might help.

mendel5, 2020-07-27:
> Do you remember the version you were coming from, i.e. the "old version"? fwupdmgr get-history might help.

Sure, here it is:

```
$ fwupdmgr --version
client version:	1.4.4
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.4.4
```

```
$ fwupdmgr get-history
20QD00M4GE
│
├─Thunderbolt Controller:
│ │   Device ID:           470906fa3412b62a24358dda0865908d5969c99b
│ │   Previous version:    43.00
│ │   GUID:                c7920601-0cda-507f-851e-92229eb1d370
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Reported to remote server
│ │                        • Device stages updates
│ │ 
│ └─ThinkPad X1 Carbon 7th / X1 Yoga 4th Thunderbolt Controller:
│       New version:       44.00
│       Remote ID:         lvfs
│       Summary:           Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th Thunderbolt Firmware
│       License:           Proprietary
│       Size:              258,0 kB
│       Created:           2020-01-21
│       Urgency:           High
│       Vendor:            Lenovo Ltd.
│       Description:       Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th Thunderbolt Firmware
│     
├─WDC PC SN730 SDBQNTY-512G-1001:
│ │   Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│ │   Previous version:    11110101
│ │   Update State:        success
│ │   Last modified:       2020-04-29 13:51
│ │   GUID:                a39943dd-3afb-54f8-b110-c5a21f071200
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Supported on remote server
│ │                        • Needs a reboot after installation
│ │                        • Reported to remote server
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─PCSN730 Device Update:
│       New version:       11130101
│       Remote ID:         lvfs
│       Summary:           WD PCSN730 Firmware Update for Lenovo PC
│       License:           Proprietary
│       Size:              2,6 MB
│       Created:           2019-12-30
│       Urgency:           High
│       Vendor:            Western Digital
│       Description:       Do NOT turn off your computer or remove the AC adapter while update is in progress.
│       
│       The computer shall be restarted after updating firmware completely. The device may not properly function until you shut down or reboot PC.
│       
│       Supported devices and firmware version : WDC PCSN730 NVMe SSD Firmware for Lenovo PC
│       
│       Supported Product Scope : Lenovo ThinkPad, ThinkCentre, ThinkStation, IdeaCentre
│     
[... 45 lines cut ...]
  │ 
  └─ThinkPad X1 Carbon 7th / X1 Yoga 4th (W-BIOS for Machine types: 20QF, 20QG, 20QD, 20QE) System Update:
        New version:       0.1.33
        Remote ID:         lvfs
        Summary:           Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th System Firmware
        License:           Proprietary
        Size:              25,0 MB
        Created:           2016-07-08
        Urgency:           High
        Vendor:            Lenovo Ltd.
        Description:       Lenovo ThinkPad X1 Carbon 7th / X1 Yoga 4th System Firmware Version 1.33
        
        The computer will be restarted automatically after updating BIOS completely . Do NOT turn off your computer or remove the AC adaptor while update is in progress.
        
        New functions or enhancements
        
         • Updated Network Boot Device and Boot conditions. (Note) Above update will show "Self-Healing BIOS  backup progressing ... xx %" massage on screen during BIOS update process.
         • Updated the Arrow key behavior in ThinkPad Setup with Graphical Setup UI.
        
        Problem fixes
        
         • Fixed an issue where Intel TXT Feature cannot be Enabled in ThinkPad Setup when Device Guard is Enabled.
```

If you need more info just let me know.