fwupd/fwupd #2277: Understanding the Device ID, GUIDs and Serial number

mendel5, 2020-07-25:
I am not sure if I understand the meaning of and difference between the device ID, GUIDs and serial number correctly. Maybe someone can explain this to me.

In my questions I will use two different computers and some examples. Let's start with some infos.
```
sudo fwupdmgr --version

client version:	1.4.4
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.4.4

Operating system: Linux Mint 20
Install method: snap

The above info is applicable for both computers.
```

Info about computer 1:
```
H97-D3H
│
├─Crucial CT275MX300SSD1:
│     Device ID:           f219ac034743df43be49a235e41f128dd47d855f
│     Summary:             ATA Drive
│     Current version:     M0CR070
│     Vendor:              Micron (ATA:0x1344)
│     Serial Number:       [redacted]
│     GUIDs:               2ba20be0-33a6-5e9a-b14a-3eb292a479e3 ← IDE\Crucial_CT275MX300SSD1__________________M0CR070
│                          2774a44c-b02a-5b6d-9b32-c6a749dc104c ← IDE\0Crucial_CT275MX300SSD1__________________
│                          a36974c8-92c2-5472-9e94-628fc111758f ← Crucial_CT275MX300SSD1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Intel(R) Xeon(R) CPU E3-1231 v3 @ 3.40GHz:
      Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:     0x27
      Vendor:              GenuineIntel
      GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
      Device Flags:        • Internal device
```

Info about computer 2:
```
B250-HD3P
│
├─Crucial CT275MX300SSD1:
│     Device ID:           612faa85309fec62ddfdce9f4635cdfef838d74e
│     Summary:             ATA Drive
│     Current version:     M0CR070
│     Vendor:              Micron (ATA:0x1344)
│     Serial Number:       [redacted]
│     GUIDs:               2ba20be0-33a6-5e9a-b14a-3eb292a479e3 ← IDE\Crucial_CT275MX300SSD1__________________M0CR070
│                          2774a44c-b02a-5b6d-9b32-c6a749dc104c ← IDE\0Crucial_CT275MX300SSD1__________________
│                          a36974c8-92c2-5472-9e94-628fc111758f ← Crucial_CT275MX300SSD1
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Intel(R) Core™ i7-7700K CPU @ 4.20GHz:
      Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
      Current version:     0xca
      Vendor:              GenuineIntel
      GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
      Device Flags:        • Internal device
 
```

Let's start with the simplest subject, the **serial number**. According to Wikipedia:
> A serial number is a unique identifier assigned incrementally or sequentially to an item, to uniquely identify it.
> https://en.wikipedia.org/wiki/Serial_number

In the info of computer 1 and 2 I have redacted the serial numbers of the CT275MX300SSD1. But I can assure you they are different. Since I expected them to be different everything is fine here. I don't have any questions about this. My conclusion is:
- **If you have two devices of the same type/model from the same manufacturer (here: CT275MX300SSD1) they should have different serial numbers.**

Now to the next subject, the **device ID**. If you compare the device IDs of CT275MX300SSD1 in computer 1 and 2 you will see that they are different. Ok, fine with me. But if you compare the device IDs of both Intel processors in computer 1 and computer 2 you will see that they are the same. But the CPUs are very different from each other, unlike the SSDs. One processor is a Xeon series from 2014 and the other one is a Core series from 2016. My question is:
- **Why do the CT275MX300SSD1 have different device IDs even though they are of the same type/model, but two completely different Intel CPUs have the same device ID?**

The last subject is the **GUID**. As I have explained in this comment https://github.com/fwupd/fwupd/issues/2205#issuecomment-663894261 I first had to figure out that GUIDs are created using `uuid.uuid5` (version 5) which uses SHA-1. At first I thought the GUIDs were created using version 4 which would have been generated randomly. More info can be found at https://en.wikipedia.org/wiki/Universally_unique_identifier#Versions . My question is:
- **Why do the CT275MX300SSD1 have three GUIDs each but the Intel CPUs only have one GUID each?**

My last question is a mix of the device ID and GUID and how fwupd is able to identify devices of the same type on different computers:
- Let's assume that Intel was to provide a firmware update to all of their Xeon CPUs which could be installed through fwupd. But if this update is accidentally applied to an Intel Core series CPU it would brick / permanently destroy the Intel Core CPU. Therefore we have to make very sure that the firmware only gets applied to Intel Xeon CPUs and not to Intel Core CPUs. Now if we compare the Intel(R) Xeon(R) CPU E3-1231 v3 from computer 1 with the Intel(R) Core™ i7-7700K from computer 2, we can see that they have both the same device ID and (!) the same GUID. **How would fwupd be able to differentiate between the two CPUs and not destroy the Core i7-7700K through installing the update which is only meant to be installed on Intel Xeon CPUs?**

hughsie, 2020-07-25:
This isn't an issue. Please use the fwupd mailing list.

closed 2020-07-25 by hughsie

mendel5, 2020-07-27:
I'm just trying to understand how fwupd works. Why can we not talk about this here on Github? I thought that's what the question label is for.

superm1, 2020-07-28:
So I feel like https://fwupd.github.io/tutorial.html is missing some information.  Maybe you would consider to submit a pull request to better document what you viewed as missing from documentation?

>If you have two devices of the same type/model from the same manufacturer (here: CT275MX300SSD1) they should have different serial numbers.

Correct.

>Why do the CT275MX300SSD1 have different device IDs even though they are of the same type/model, but two completely different Intel CPUs have the same device ID?

Device ID is meant to disambiguate different devices connected to the system both in a physical or logical topology.  
You can look at these for more about what they mean:
https://fwupd.github.io/fwupd-FuDevice.html#fu-device-set-logical-id
https://fwupd.github.io/fwupd-FuDevice.html#fu-device-set-physical-id

Between two machines they might be the same, might be different.  But between two devices on same machine they will always be different.

>Why do the CT275MX300SSD1 have three GUIDs each but the Intel CPUs only have one GUID each?

GUIDs are ways to represent that view of the hardware.  Some are more specific than others.  For example if I have firmware that should only ever patch a single SSD in the world this one makes sense:

`
2ba20be0-33a6-5e9a-b14a-3eb292a479e3 ← IDE\Crucial_CT275MX300SSD1__________________M0CR070
`

And if I want to patch all of that model on the IDE bus:

`
2774a44c-b02a-5b6d-9b32-c6a749dc104c ← IDE\0Crucial_CT275MX300SSD1__________________
`
Makes sense

And if I don't care about the bus:

`
a36974c8-92c2-5472-9e94-628fc111758f ← Crucial_CT275MX300SSD1
`

makes sense

superm1, 2020-07-28:
>How would fwupd be able to differentiate between the two CPUs and not destroy the Core i7-7700K through installing the update which is only meant to be installed on Intel Xeon CPUs?

CPUs don't contain an updatable flag.  They're information only in fwupd.

mendel5, 2020-07-29:
Thanks for the detailed answer!