fwupd/fwupd #2278: Replace all instances of "github.com/hughsie" with "github.com/fwupd"

mendel5, 2020-07-26:
As far as I can tell the fwupd and lvfs-website repos used to be part of your @hughsie personal account but have been moved to the organization @fwupd . We should adjust all links that still point to the old repo and make them point to the new repo. Here are the instances that still include links to the old repository:

```
$ grep -rin "github.com/hughsie"
plugins/synaptics-mst/fu-plugin-synaptics-mst.c:24:/* see https://github.com/hughsie/fwupd/issues/1121 for more details */
subprojects/libxmlb.wrap:3:url = https://github.com/hughsie/libxmlb.git
subprojects/libjcat.wrap:3:url = https://github.com/hughsie/libjcat.git
subprojects/gusb.wrap:3:url = https://github.com/hughsie/libgusb.git
.github/pull_request_template.md:2:- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
```

In the instances listed above we should replace `github.com/hughsie` with `github.com/fwupd`.

superm1, 2020-07-28:
Some of those have moved over but not all of them.  Would you mind submitting a PR?

hughsie, 2020-07-28:
> we should replace github.com/hughsie with github.com/fwupd.

libxmlb, libjcat and libgusb are not fwupd specific and are shared with multiple projects. I think they live in the right place in my own personal silo. The two remaining cases are fine to move, but PR please.

mendel5, 2020-07-28:
> PR please

Sure, I will create a PR for the two following files:
```
plugins/synaptics-mst/fu-plugin-synaptics-mst.c:24:/* see https://github.com/hughsie/fwupd/issues/1121 for more details */
.github/pull_request_template.md:2:- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
```

mendel5, 2020-07-28:
Please have a look at https://github.com/fwupd/fwupd/pull/2288 .

superm1, 2020-07-28:
Merged it, thanks!

closed 2020-07-28 by superm1