fwupd/fwupd #2279: fwupd[7368]: segfault at ffffffffffffffc8 ip 00007fab5177de90 sp 00007ffd4c6404b0 error 5 in libxmlb.so.1.0.0[7fab51771000+1d000]

paulmenzel, 2020-07-27:
Looking through the logs of Ubuntu 18.04.4 LTS (Bionic Beaver) with

    ii  libxmlb1:amd64                  0.1.8-1~ubuntu18.04. amd64                Binary XML library

and

    ii  fwupd                           1.2.10-1ubuntu2~ubun amd64                Firmware update daemon

the Linux logs contained the crash below.

    [Fr Jul 10 08:21:15 2020] fwupd[7368]: segfault at ffffffffffffffc8 ip 00007fab5177de90 sp 00007ffd4c6404b0 error 5 in libxmlb.so.1.0.0[7fab51771000+1d000]
    [Fr Jul 10 08:21:15 2020] Code: 00 00 0f 1f 84 00 00 00 00 00 48 89 f7 e9 58 80 00 00 0f 1f 84 00 00 00 00 00 55 53 48 89 f3 48 83 ec 08 48 63 05 00 02 21 00 <f6> 44 07 18 01 74 4b 48 89 f7 e8 11 80 00 00 8d 70 ff 48 89 df e8

Searching the Web there are logs with similar crashes.

hughsie, 2020-07-27:
> fwupd[7368]: segfault at ffffffffffffffc8

Are there any backtraces for this crash? That's not nearly enough information to begin debugging I'm afraid.

superm1, 2020-07-27:
I glanced through errors.ubuntu.com tracker, and the closest thing is this:

```
#0  0xb7523f8c in xb_machine_stack_pop (self=0x0, stack=0xb72cfc3c) at ../src/xb-machine.c:969
        priv = <optimized out>
#1  0xb7525940 in xb_machine_func_string_cb (self=0x0, stack=0xb72cfc3c, result=0xb71d7009 <g_hash_table_lookup+9>, user_data=0xb732fcc8, exec_data=0xb71c1607 <g_bytes_unref+7>, error=0x8006e68c) at ../src/xb-machine.c:1702
        op1 = <optimized out>
#2  0xb71c1637 in g_bytes_unref (bytes=0x8136eac0) at ../../../../glib/gbytes.c:336
        bytes = 0x8136eac0
#3  0x80045dcc in fu_smbios_item_free (item=0x81340a20) at ../src/fu-smbios.c:456
No locals.
#4  0xb71ba5ce in g_ptr_array_foreach (array=0x8134f860, func=0x80045db0 <fu_smbios_item_free>, user_data=0x0) at ../../../../glib/garray.c:1522
        i = 0
        _g_boolean_var_ = <optimized out>
#5  0xb71ba687 in ptr_array_free (array=0x8134f860, flags=flags@entry=FREE_SEGMENT) at ../../../../glib/garray.c:1107
        rarray = 0x8134f860
        segment = <optimized out>
#6  0xb71ba70e in g_ptr_array_unref (array=0x8134f860) at ../../../../glib/garray.c:1051
        rarray = 0x8134f860
        __func__ = "g_ptr_array_unref"
#7  0x80046085 in fu_smbios_finalize (object=0x8134f848) at ../src/fu-smbios.c:466
        self = <optimized out>
#8  0xb72e3a9a in g_object_unref (_object=0x8134f848) at ../../../../gobject/gobject.c:3340
        weak_locations = <optimized out>
        object = 0x8134f848
        old_ref = <optimized out>
        _object = <optimized out>
        has_toggle_ref = <optimized out>
        weak_locations = <optimized out>
        weak_ref_location = <optimized out>
        has_toggle_ref = <optimized out>
        object = <optimized out>
        old_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        object = 0x8134f848
        old_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        has_toggle_ref = <optimized out>
        weak_locations = <optimized out>
        weak_ref_location = <optimized out>
        has_toggle_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
#9  0x80040b41 in fu_plugin_finalize (object=0x81365500) at ../src/fu-plugin.c:2016
        self = <optimized out>
        priv = <optimized out>
        func = 0x0
#10 0xb72e3a9a in g_object_unref (_object=0x81365500) at ../../../../gobject/gobject.c:3340
        weak_locations = <optimized out>
        object = 0x81365500
        old_ref = <optimized out>
        _object = <optimized out>
        has_toggle_ref = <optimized out>
        weak_locations = <optimized out>
        weak_ref_location = <optimized out>
        has_toggle_ref = <optimized out>
        object = <optimized out>
        old_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        object = 0x81365500
        old_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        has_toggle_ref = <optimized out>
        weak_locations = <optimized out>
        weak_ref_location = <optimized out>
[... 73 lines cut ...]
        _g_boolean_var_ = <optimized out>
        object = 0x8134d848
        old_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
        _g_boolean_var_ = <optimized out>
        has_toggle_ref = <optimized out>
        weak_locations = <optimized out>
        weak_ref_location = <optimized out>
        has_toggle_ref = <optimized out>
        _g_boolean_var_ = <optimized out>
#18 0x8001abaf in fu_main_private_free (priv=0x8134b0e0) at ../src/fu-main.c:1486
No locals.
#19 glib_autoptr_cleanup_FuMainPrivate (_ptr=<synthetic pointer>) at ../src/fu-main.c:1500
No locals.
#20 main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:1517
        immediate_exit = 0
        timed_exit = 0
        options = {{long_name = 0x8004d26e "timed-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0xbfae1550, description = 0x8004d279 "Exit after a small delay", arg_description = 0x0}, {long_name = 0x8004d292 "immediate-exit", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0xbfae154c, description = 0x8004d5d8 "Exit after the engine has loaded", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
        priv = 0x8134b0e0
        error = 0x0
        argv0_file = <optimized out>
        context = <optimized out>
```

Which is also on fwupd 1.2.10-1ubuntu2~ubuntu18.04.3, and libxmlbf 0.1.8-1~ubuntu18.04.1.  There are only two reports of it though.

superm1, 2020-07-27:
In libxmlb master this has changed recently: https://github.com/hughsie/libxmlb/commit/52f64aa8bbf785244a7f3a2b3881e265c01eff82

hughsie, 2020-07-27:
> xb_machine_stack_pop (self=0x0

Err, is Ubuntu using git master libxmlb? That's not a good idea as the ABI still isn't finished yet for the soname break...

hughsie, 2020-07-27:
> ../src/fu-smbios.c

... and that looks like an old fwupd too. The crash is pretty wacky, but I remember debugging something like that a while ago.

paulmenzel, 2020-07-27:
From `/var/log/apport.log.1`:

    ERROR: apport (pid 7020) Fri Jul 10 08:21:18 2020: host pid 7368 crashed in a separate mount namespace, ignoring

This might be caused by [Ubuntu bug 1870060 in apport (Ubuntu Focal) "systemd ProtectSystem/mount namespace makes apport fail (impact most of our default system services)"][1]

[1]: https://bugs.launchpad.net/ubuntu/+source/apport/+bug/1870060

superm1, 2020-07-27:
> > xb_machine_stack_pop (self=0x0
> 
> Err, is Ubuntu using git master libxmlb? That's not a good idea as the ABI still isn't finished yet for the soname break...

No, it looks like somehow my comment got struck out.  It's libxmlb 0.1.8.  I was just saying it's changed in master, so if there was a bug it might have been resolved in master or changed.

superm1, 2020-07-27:
@paulmenzel so can you reproduce this readily?  If so, can you please capture an actual traceback?

paulmenzel, 2020-07-27:
> @paulmenzel so can you reproduce this readily? If so, can you please capture an actual traceback?

No, I can’t reproduce.

superm1, 2020-07-27:
> > ../src/fu-smbios.c
> 
> ... and that looks like an old fwupd too. The crash is pretty wacky, but I remember debugging something like that a while ago.

Yeah I remember you doing one, and I touched one in the same file: https://github.com/fwupd/fwupd/commit/3f9a1c182a4b05d641680e9505c24e10d75cfe3e

>No, I can’t reproduce.

I think without more details we can't do anything.  Please capture a bug trace, or at least send it up to errors.ubuntu.com if it comes up again.

>This might be caused by Ubuntu bug 1870060 in apport (Ubuntu Focal) "systemd ProtectSystem/mount namespace makes apport fail (impact most of our default system services)"

That's 20.04 specific it appears.

closed 2020-07-27 by superm1