fwupd/fwupd #2280: Add the ability to reject specific firmware updates

Yawning, 2020-07-27:
**Is your feature request related to a problem? Please describe.**

Certain firmware updates cause problems for end users (eg: https://github.com/fwupd/firmware-lenovo-thinkpad/issues/54 https://github.com/fwupd/firmware-lenovo-thinkpad/issues/63), and are repeatedly offered as update candidates when they should not be (or are already) installed.

**Describe the solution you'd like**

It should be possible for the end user to reject specific firmware updates, without having to reject all updates by device GUID.

eg: Explicitly reject `Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware 1.0.04`, but accept `Lenovo ThinkPad T14s Gen 1 / ThinkPad X13 Gen 1 Embedded Controller Firmware 1.0.06`.

**Describe alternatives you've considered**

The current solution of rejecting all updates for the troublesome device is poor as "having to remember to check the vendor's website for newer, hopefully fixed updates" is not great UX.

hughsie, 2020-07-27:
@Yawning do you want to restrict to a specific firmware, or a specific download? For instance, Lenovo could issue an update 1.2.3 with checksum `deadbeef` that doesn't work. They might pull that and reissue 1.2.3 with checksum `beefbeef`.

Would you want to block the install of 1.2.3 or just the one with checksum `deadbeef`?

Yawning, 2020-07-27:
> Would you want to block the install of 1.2.3 or just the one with checksum `deadbeef`?

Probably the latter (eg: reject downloads by SHA256 digest), since a fixed firmware release would resolve the sort of issues that make me want this capability in the first place.

I think most of the cases I can think of for the former (reject all downloads with a specific version) would be covered by the existing device GUID based option, but maybe I'm not being imaginative enough.

hughsie, 2020-07-28:
> eg: reject downloads by SHA256 digest

So the problem here is that we don't actually expose the SHA256 hashes of the payload or archive to the user in the `fwupdmgr get-releases` output. @superm1 got any ideas?

superm1, 2020-07-28:
I don't think I understand the problem.  There is a firmware issue with applying the update and you wan the ability to blacklist that individual update?  Basically https://github.com/fwupd/fwupd/issues/900 ?

Yawning, 2020-07-28:
> I don't think I understand the problem. There is a firmware issue with applying the update and you wan the ability to blacklist that individual update? Basically #900 ?

Essentially yes.

hughsie, 2020-07-28:
I think #900 is "update was installed, but broken, so I downgraded" rather than "update didn't install"

superm1, 2020-07-28:
But the outcome is the same, "I want the ability to prevent the specific update release from installing again in the future"

hughsie, 2020-07-28:
Ahh, so we can use the history db to get the correct hash regardless. i.e. "I tried, it failed, don't try again"

superm1, 2020-07-28:
I think we need to discuss how the actual flow for the user would work though.  Should it be next time they call `fwupdmgr upgrade` there is a prompt for ones you tried but they failed and you let them prevent them from running again?

Or is it just a separate `fwupdmgr block %ARG` where if no ARG was provided you show recent things from the history database ?

hughsie, 2020-07-28:
> Or is it just a separate fwupdmgr block

That's what I imagined, i.e. we show the user recent failures from the history DB as a list.

superm1, 2020-07-28:
OK sounds good to me.

But to solve both this and #900 at the same time I would say just show recent releases from history DB.  Because if you upgraded/downgraded those are both "success" but you're still trying to block the release from trying again in the future due to some other reason.

hughsie, 2020-07-28:
> OK sounds good to me.

Do you think we should allow the user to select a hash from the history database, and then update the `BlockedFirmware` key in `daemon.conf` or should we store the config in the database like we do the approved list? I don't think we've got any history of writing to files in /etc so far.

superm1, 2020-07-28:
I think database like approved list.

closed 2020-08-10 by hughsie