fwupd/fwupd #2287: What do these abbreviations mean? (TBT, DMI, HIDRAW, NVME)

mendel5, 2020-07-28:
See question below, first some info:

```
$ fwupdmgr --version

client version:	1.4.4
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.4.4
```

```
$ fwupdmgr get-devices

20QD00M4GE
│
├─Thunderbolt Controller:
│     Device ID:           470906fa3412b62a24358dda0865908d5969c99b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370
│                          96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           c085a3cbf4dc1d60d110b7616816e2ffd279b64e
│     Current version:     0.1.17
│     Minimum Version:     0.1.17
│     Vendor:              DMI:LENOVO
│     GUID:                f72e048b-65bd-4e71-9071-1ac7045223e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel Management Engine:
│     Device ID:           0ec6a2d51c2c55a649247d119ffb8e229168d6c9
│     Current version:     192.49.1534
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                c3e4be53-e714-4ea1-bb9c-7fe13a98b556
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           f09b748a98b31bb5c7c64d83f24ddbc80dff7c5f
│     Current version:     0.1.33
│     Minimum Version:     0.0.1
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                4bbc40fa-f81e-4206-bc70-a1f7b744d964
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─Touchpad:
│     Device ID:           91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:     1.3.3013412
│     Bootloader Version:  8.0.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               10010313-27c0-5c7f-bd99-955028fd3a01
│                          71463ccd-7553-5d7d-9fac-7ed8073a7853
│                          981824ee-4283-595b-a2e4-46eae92dba40
│                          0957b08b-b580-5c3a-8073-98b9bb15bc7f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Supported on remote server
│   
├─UEFI Device Firmware:
│     Device ID:           b596f8dec94a959248534b35641fe2341e27ed06
│     Current version:     16785559
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                9045ae09-34ab-46d6-826d-b02f859dd20b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           f61a138e8240c473a88d703babf7be5bc0e098e6
│     Current version:     4784136
│     Minimum Version:     4784136
│     Vendor:              DMI:LENOVO
│     GUID:                db246b7d-a882-4e0e-984f-23ac1b0270db
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           97d441c3228fe8615f883b166a8809389803f9e2
│     Current version:     26403
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                68d3b08f-6ed2-43ca-86f0-f1ad6e41b390
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:             NVM Express Solid State Drive
      Current version:     11130101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                           06b4e2aa-91af-508b-b06e-65e3b3189e97
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

I have some questions about the output of `fwupdmgr get-devices`:
- In `Lenovo (TBT:0x0109)` does `TBT` refer to Thunderbolt, e.g. https://aur.archlinux.org/packages/tbt/ ?
- In `DMI:LENOVO` and `LENOVO (DMI:LENOVO)` does `DMI` refer to the Desktop Management Interface, e.g. https://en.wikipedia.org/wiki/Desktop_Management_Interface ?
- In `Synaptics (HIDRAW:0x06CB)` does `HIDRAW` refer to the Human Interface Devices Raw Access, e.g. https://www.kernel.org/doc/Documentation/hid/hidraw.txt ?
- In `Sandisk Corp (NVME:0x15B7)` does `NVME` refer to the NVMe storage command line interface utility, e.g. https://www.mankier.com/1/nvme ?

superm1, 2020-07-28:
Those are referring to vendor identifiers in subsystems.  The vendor identifier is used to restrict which components the vendor is allowed to upload _to_ LVFS for distribution.  For example in above you wouldn't want Sandisk to be able to update your UEFI device firmware, only the storage device.

mendel5, 2020-07-28:
So in the case of `Synaptics (HIDRAW:0x06CB)` the hidraw driver/ package was used by fwupdmgr to figure out the manufacturer?

superm1, 2020-07-28:
The kernel's hidraw interface is what is used to communicate with the device in that case.  You can dig into the actual plugin code to see more of how that interaction actually works.  There is no extra userland software needed other than gudev.

closed 2020-07-29 by superm1

mendel5, 2020-07-29:
Ok, makes sense.