fwupd/fwupd #2292: TPM PCR0 differs from reconstruction

bulletmark, 2020-07-30:
I see this error with a link that points me to  https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction. That page says there is a fwupd bug causing this but fixed in version 1.4.5. Looking at https://github.com/fwupd/fwupd/releases there is no 1.4.5? [Nor is there any 1.5.* versions there either]. If there was such an official version then no doubt my Arch Linux package would get promptly updated to it. My Arch Linux version is 1.4.4-2.

As an aside, I have found some errors from fwupd are presented to the user in extremely cryptic technical terms which probably mean something to a fwupd developer but are just gobbledygook to a normal user. This error message is a good example of that.

joakim-tjernlund, 2020-07-31:
1.4.5 has been release now

superm1, 2020-07-31:
That message is no longer present in 1.4.5 as well.  It was intended in earlier releases to help us catch bugs.  And it did!

closed 2020-07-31 by superm1