fwupd/fwupd #2294: Dell PowerEdge Server: Construction of PCR0 using TPM event log is failing

prbinu, 2020-07-31:
**Describe the bug**
Background: https://github.com/tpm2-software/tpm2-tools/issues/1975#issuecomment-666719973

The construction of PCR0 using TPM event log in a tool called fwupdtpmevelog.

**Steps to Reproduce**

```
$ sudo fwupdmgr security --force
Host Security ID: HSI:0+! (v1.5.0)

HSI-1
✔ AMT manufacturing mode:        Locked
✔ AMT override:                  Locked
✔ Intel DCI debugger:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✘ UEFI dbx:                      Not found: https://github.com/fwupd/fwupd/wiki/Missingdbx

HSI-2
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✘ IOMMU:                         Not found
✘ Intel DCI debugger:            Unlocked
✘ TPM PCR0 reconstruction:       Not found

HSI-3
✔ Intel BootGuard error policy:  Valid
✔ Suspend-to-ram:                Disabled
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```

```bash
$ sudo  tsseventextend -sim -if /sys/kernel/security/tpm0/binary_bios_measurements 
eventextend: failed, rc 0000009a
TPM_RC_INSUFFICIENT - the TPM was unable to unmarshal a value because there were not enough octets in the input buffer Handle number unspecified
````

**Expected behavior**

The construction of PCR0 using TPM event log should match actual digest in  TPM PCR0

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the external fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --filter=~internal
PowerEdge R740
```

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [ ] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**
No

**Verbose daemon logs**
First enable daemon verbose logs collection.
```shell
fwupdmgr modify-config "VerboseDomains" "*"
```

Then try to reproduce the issue.  Even if it doesn't reproduce, please attach the
daemon verbose logs collected from the system journal.
```shell
journalctl -b -u fwupd.service
```

**Additional questions**
- Operating system and version: Ubuntu 20-04
- Have you tried unplugging the dock or any peripherals from your machine? No
- Have you tried to power cycle the dock from the AC adapter? No
- Is this a regression?

prbinu, 2020-07-31:
Also noticed "Firmware Bug" on that host.
```
$ dmesg | grep -i  tpm
[    0.000000] efi:  ACPI=0x6fffe000  ACPI 2.0=0x6fffe014  SMBIOS=0x69359000  SMBIOS 3.0=0x69357000  MEMATTR=0x6520c020  TPMEventLog=0x5020b020
[    0.000000] [Firmware Bug]: TPM Final Events table missing or invalid
[    0.009453] ACPI: SSDT 0x000000006FFFC000 0005F8 (v02 DELL   Tpm2Tabl 00001000 INTL 20180508)
[    0.009455] ACPI: TPM2 0x000000006FFFB000 000038 (v04 DELL   PE_SC3   00000002      01000013)
[    4.056503] tpm_tis MSFT0101:00: 2.0 TPM (device-id 0xFE, rev-id 4)
 ```

hughsie, 2020-07-31:
@superm1

superm1, 2020-07-31:
@charles-rose can you look at this one?

prbinu, 2020-08-12:
Just checking to see if there any update on this issue

superm1, 2020-08-13:
@prbinu Nothing right now.  I'm going to transfer it to the project that has various firmware issues.  If we conclude it's a fwupd issue we'll transfer it back.