fwupd/fwupd #2296: why fwupdmgr update do nothing?

Bonteruel, 2020-08-01:
I was running
$ fwupdmgr get-devices
and it show me available updates for my SSD and the HDD, but when I try to apply them with 
$ fwupdmgr update
the program do nothing.
It shows 2 available firmware updates but don't download them.
If you want to know, the output of fwupdmgr get-devices is this:
HP SSD S700 500GB
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 41f37fa4-1547-5b0d-9590-361fa72eacef <- IDE\HP_SSD_S700_500GB_______________________S0704A1
  Guid:                 7757a94e-3025-5be0-acc2-cdd5d242e92b <- IDE\0HP_SSD_S700_500GB_______________________
  Guid:                 cdf8337b-1809-5c0c-865b-ef7b6a2c9f82 <- HP SSD S700 500GB
  Serial:               HBSA39471204280
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              S0704A1
  Icon:                 drive-harddisk
  Created:              2020-08-01

TOSHIBA HDWL110
  DeviceId:             0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
  Guid:                 2dda3b56-e90b-5d08-be60-ee95f801a9ef <- IDE\TOSHIBA_HDWL110_________________________JU000A
  Guid:                 1f02e129-be3b-5e98-a295-b16bb982c058 <- IDE\0TOSHIBA_HDWL110_________________________
  Guid:                 8bc39900-5acd-5141-b937-ee081bfdf46d <- TOSHIBA HDWL110
  Serial:               Z9F1PPV2T
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              JU000A
  Icon:                 drive-harddisk
  Created:              2020-08-01
Don't know if is important, but both devices are replaces; the SSD replace the original HDD, and the HDD is in a caddy drive, replacing the DVD drive

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version
```

exploide, 2020-08-02:
For an in-depth analysis, I think you need to wait for Richard or Mario to take a look.

But I noticed you are using an horribly outdated version of fwupd. I think SSD/HDD support was achieved rather recently. Maybe you can try with a newer version of fwupd.

(PS: wrapping your output in three backticks makes it much easier to read here)

Bonteruel, 2020-08-02:
Hi. Before you suggested, I installed the snap version. I didn't before because it take a long time for my conection, but finally installed. 
This is the output:
fwupd.fwupdmgr update
• S700 500GB has no available firmware updates
• TOSHIBA HDWL110 has no available firmware updates
• Touchpad has no available firmware updates
No updatable devices
But I don't understand why appears the three devices with
fwupd.fwupdmgr get-devices
I thought it means there is available updates
Anyway, aparently with the fresh snap version is solved, but not completely understood how the commands report updates. If you simply tell me that I have the firmware updated(even if I didn't install anything) then I take your answer. Thank you for your reply.

exploide, 2020-08-02:
Ah, I see. So a brief description of the most important commands:

- `fwupdmgr get-devices` Lists all devices on your system fwupd is aware of. This does not mean it currently has any update available nor does it mean updating is supported at all for that device.
- `fwupdmgr refresh` updates the metadata to get aware of new firmware files on the LVFS server.
- `fwupdmgr get-updates` shows available updates for your devices.
- `fwupdmgr update` actually installs the available updates.

I think `fwupdmgr --help` is also helpful.

So in summary, if you refreshed the matadata but `get-updates` shows nothing, even with an up to date fwupd version, then your firmware already is fully updated. `get-devices` only tells you what devices fwupd is aware of on your system.

exploide, 2020-08-06:
Seems the question is answered. I'm closing. Feel free to respond if anything is left from your side.

closed 2020-08-06 by exploide