fwupd/fwupd #2298: fu-device-list:  fu_device_list_replace should save custom flags

bleungatchromium, 2020-08-03:
**Describe the bug**
A clear and concise description of what the bug is.
My plugin sets some custom flags. On attach or detach, when my usb device is rebooted, and fwupd replaces on the new device enumeration, the new instance of the device doesn't bring over the custom flags.

**Steps to Reproduce**
Steps to reproduce the behavior.
Reproduced it consistently using my WIP series here #2289

**Expected behavior**
A clear and concise description of what you expected to happen.
On device replacement, custom flags that were set on the old device should be copied to the new one.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```
client version: 1.5.0
compile-time dependency versions
gusb: 0.3.5

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

bleungatchromium, 2020-08-03:
I posted this to my series that addresses this: https://github.com/fwupd/fwupd/pull/2289/commits/e60beb5361badb121f043926fe58be85a477b79e

hughsie, 2020-08-03:
That looks entirely sane; if you split out the commit into a new PR we can merge it as-is. Out of interest, what custom flags are you using? Custom flags used by more than one plugin usually get converted to, well, _ordinary_ flags :)

bleungatchromium, 2020-08-03:
> That looks entirely sane; if you split out the commit into a new PR we can merge it as-is. Out of interest, what custom flags are you using? Custom flags used by more than one plugin usually get converted to, well, _ordinary_ flags :)

I'm using flags to indicate whether my rw region was written this time around or my ro region. This is a tiny bit of state I need preserved between ANOTHER_WRITE_REQUIRED loops.

PR #2301 (2020-08-03): Save custom flags on device replace
New instances of a device may be interested in custom flags. Copy them
over on fu_device_list_replace.

fixes #2298

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [X] Code fix
- [ ] Feature
- [ ] Documentation
commit 8b9f9ab4 (2020-08-03): Save custom flags on device replace

New instances of a device may be interested in custom flags. Copy them
over on fu_device_list_replace.

fixes #2298
commit b7402c9d (2020-08-18): Save custom flags on device replace

New instances of a device may be interested in custom flags. Copy them
over on fu_device_list_replace.

fixes #2298
closed 2020-08-03 by hughsie