fwupd/fwupd #2299: TPM PCR0 differs from reconstruction after BIOS update

ppwaskie, 2020-08-03:
I'm running a Dell XPS 9300 laptop with Ubuntu 20.04 and Windows 10.  I just updated my BIOS to Dell's 1.1 image, and on the subsequent boot successfully updated it.  However, after booting Linux, I now see TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction in my Ubuntu Software update app.

See attached for the output from fwupdtpmevlog.  Not sure how to resolve this.  I'm also running 1.3.11 of fwupd.

[foo.txt](https://github.com/fwupd/fwupd/files/5014896/foo.txt)

hughsie, 2020-08-03:
@superm1 1.3.11 should include 57b8b7fe9b5a358b9f686f053b4fb5e5d1a6bd3e -- any ideas?

ppwaskie, 2020-08-03:
Specifically, I'm running 1.3.11-1~focal1 that is packaged by Ubuntu/Canonical.

superm1, 2020-08-10:
> in my Ubuntu Software update app

Can you show a screenshot where you're seeing it?  It's surprising that it would be in the app.  It should have been moved into the journal if it still exists.
https://github.com/fwupd/fwupd/commit/1b73a132461ca15d4aadf4c0cfd2b5e95ac14d02#diff-18e6895f0172a2d6ef4200a07a6d8f07

>@superm1 1.3.11 should include 57b8b7f -- any ideas?

It's probably a lower level problem - in firmware at this point.