fwupd/fwupd #2302: Thinkpad T470s: Thunderbolt Controller not supported anymore with 1.4.5?

dkadioglu, 2020-08-05:
**Describe the bug**
After the update from 1.4.4 to 1.4.5 the Thunderbolt Controller of the Lenovo Thinkpad T470s seems to be not supported anymore. At least, fwupdmgr get-updates/get-devices don't list it anymore.

**Steps to Reproduce**
Run fwupdmgr get-updates on T470s with fwupd 1.4.4:
• Thunderbolt Controller has the latest available firmware version
• Embedded Controller has the latest available firmware version
• Intel Management Engine has the latest available firmware version
• SAMSUNG MZVLW1T0HMLH-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
• Unifying Receiver has the latest available firmware version

Run fwupdmgr get-updates on T470s with fwupd 1.4.5:
• Embedded Controller has the latest available firmware version
• Intel Management Engine has the latest available firmware version
• SAMSUNG MZVLW1T0HMLH-000L7 has the latest available firmware version
• System Firmware has the latest available firmware version
• Unifying Receiver has the latest available firmware version

**Expected behavior**
The Thunderbolt Controller should still be listed with 1.4.5.

**fwupd version information**
client version: 1.4.4
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.4

client version: 1.4.5
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.4.5

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
pacman
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.
with 1.4.4:
```20HGS00V00
│
├─Thunderbolt Controller:
│     Device ID:           9cf44cb16f407badb950570b6c96d46f00f310b8
│     Zusammenfassung:     Unmatched performance for high-speed I/O
│     Current version:     20.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               96c3f95f-611a-56f8-be95-0f912cb84657 ← TBT-01091609
│                          d5852bbd-0777-54b3-8f67-552be27d023f ← TBT-01091609-controller0-0
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           03a6930fc4c833443a7502dde379b98b961c8c57
│     Current version:     0.1.21
│     Minimum Version:     0.0.1
│     Vendor:              DMI:LENOVO
│     GUID:                e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─HD Graphics 620:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
[... 179 lines cut ...]
│     GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
│                          5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
│                          ddf995da-1b32-5a8a-bc1b-8d5af4b38b51 ← TPM\VEN_IFX&MOD_SLB9670
│                          6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
│                          301555de-680d-5ddc-b995-7553fc9138f1 ← TPM\VEN_IFX&MOD_SLB9670&VER_2.0
│     Device Flags:        • Internal device
│   
└─Unifying Receiver:
      Device ID:           b8f11848dfcbf441df375b9f11b0e42d085f688b
      Zusammenfassung:     A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0014
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
                           • Supported on remote server
```

**Additional questions**
- Operating system and version: Arch Linux (fully updated)
- Have you tried rebooting? yes
- Is this a regression? yes

hughsie, 2020-08-05:
It will appear if you plug in a Thunderbolt device into the port. ThunderBolt is a very odd controller where if you have nothing plugged in then it actually powers down to the point of disappearing from the PCI bus completely. We used to use a workaround where we could use a GPIO to force power the controller to wake it up enough to query it, but it was unreliable when flashing some hardware and so we removed the feature.

superm1, 2020-08-10:
Closed as expected behavior now.

closed 2020-08-10 by superm1