fwupd/fwupd #2308: fwupdmgr security check on Intel Bay Trail systems

dnoliver, 2020-08-10:
**Describe the question**
A clear and concise description of what you are wondering about.

Currently, if you run the security check on a system using Intel Bay Trail, there is no information about the TXE version.
The reason could be that this systems do not provide version information in the `mei0/fw_ver` file:

```shell
cat /sys/devices/pci0000:00/0000:00:1a.0/mei/mei0/fw_ver
0:0.0.0.0
0:0.0.0.0
0:0.0.0.0
```

For checking the version on this systems, you need to use the "Legacy" version of the CSME Detection Tool https://downloadcenter.intel.com/download/29057/CSME_Detection_Tool_Legacy_Linux.tar.gz

The result is that, you get no version information with the security test. 

If you run the `intel_csme_detection_tool_legacy` tool, you get:

```shell
Intel(R) CSME Detection Tool For Legacy Systems
Copyright(C) 2017-2019, Intel Corporation, All rights reserved.

Application Version: 1.0.16.0
Scan date: 2020-08-10 16:22:32 GMT

*** Host Computer Information ***
Name: ------------------------
Manufacturer: ----------------
Model: ------------------------
Processor Name: Intel(R) Atom(TM) CPU  E3845  @ 1.91GHz
OS Version: Fedora 31 Thirty One (5.5.17-200.fc31.x86_64)

*** Intel(R) ME Information ***
Engine: Intel(R) Trusted Execution Engine
Version: 1.0.2.1060

*** Risk Assessment ***
Based on the analysis performed by this tool: The system is not supported
Firmware versions of Intel(R) ME 3.x thru 5.x,
Intel(R) TXE 1.x thru 2.x and Intel(R) Server Platform Services 1.x thru 2.x are no longer supported,
thus were not assessed for the vulnerabilities/CVEs listed in these Security Advisories.
There is no new release planned for these versions.

For more information refer to the
Intel(R) CSME Detection Tool For Legacy Systems Guide or the
Intel Security Advisory Intel-SA-00086 at the following link:
https://www.intel.com/sa-00086-support
```

My question is, should you get a similar output on this tool as well?

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.
```shell
$ fwupdmgr --version
```

fwupdmgr version is the latest built master

hughsie, 2020-08-11:
I'm assuming you get `ME family not supported` when running `sudo FWUPD_MEI_VERBOSE=1 fwupdtool --verbose security`, right? And if so, what's the output of `sudo fwupdtool --verbose get-devices --show-all-devices`? Thanks.

dnoliver, 2020-08-11:
For the first assumption, I get this output:

```shell
20:58:35:0761 FuEngine             disabling plugin because: failed to coldplug using amt: ME refused connection
```

The output of `get-devices` is:

```shell
[test@advantech-1 fwupd]$ sudo ./build/src/fwupdtool --verbose get-devices --show-all-devices
21:01:06:0284 FuDebug              Verbose debugging enabled (on console 1)
21:01:06:0326 FuConfig             loading config values from /usr/local/etc/fwupd/daemon.conf
21:01:06:0328 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
21:01:06:0332 FuRemoteList         loading /usr/local/share/fwupd/metainfo
21:01:06:0333 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
21:01:06:0333 XbSilo               file: 6d4d1e64-0863-72e6-6254-bbd148933bbb, current:6d4d1e64-0863-72e6-6254-bbd148933bbb, cached: (null)
21:01:06:0334 XbSilo               loading silo with file contents
21:01:06:0334 Fwupd                looking for os-release at /etc/os-release
21:01:06:0335 FuRemoteList         loading remotes from /usr/local/etc/fwupd/remotes.d/lvfs.conf
21:01:06:0335 FuRemoteList         loading remotes from /usr/local/etc/fwupd/remotes.d/lvfs-testing.conf
21:01:06:0337 FuRemoteList         loading remotes from /usr/local/etc/fwupd/remotes.d/vendor.conf
21:01:06:0337 FuRemoteList         loading remotes from /usr/local/etc/fwupd/remotes.d/vendor-directory.conf
21:01:06:0338 FuRemoteList         loading remotes from /usr/local/etc/fwupd/remotes.d/fwupd-tests.conf
21:01:06:0338 FuRemoteList         loading remotes from /usr/local/etc/fwupd/remotes.d/dell-esrt.conf
21:01:06:0339 FuRemoteList         ignoring unfound remote fwupd
21:01:06:0339 FuRemoteList         ordering lvfs-testing=lvfs+1
21:01:06:0339 FuRemoteList         ignoring unfound remote fwupd
21:01:06:0339 FuRemoteList         ignoring unfound remote fwupd
21:01:06:0339 FuRemoteList         ignoring unfound remote fwupd
21:01:06:0340 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
21:01:06:0340 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
21:01:06:0340 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:01:06:0340 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd/LVFS-CA.pem
21:01:06:0340 Jcat                 reading /usr/local/etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
21:01:06:0341 Jcat                 loaded 1 certificates
21:01:06:0341 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
21:01:06:0341 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
21:01:06:0341 Jcat                 trying to load certificate from /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem
21:01:06:0341 Jcat                 reading /usr/local/etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
21:01:06:0342 Jcat                 loaded 1 certificates
21:01:06:0342 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2459 bytes
21:01:06:0342 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
21:01:06:0376 FuEngine             client certificate exists and working
21:01:06:0376 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
21:01:06:0377 FuHistory            got schema version of 6
21:01:06:0377 FuHwids              smbios property Manufacturer=Advantech
21:01:06:0377 FuHwids              smbios property EnclosureKind=3
21:01:06:0377 FuHwids              smbios property Family=To be filled by O.E.M.
21:01:06:0377 FuHwids              smbios property ProductName=UNO-1372G-E3AE
21:01:06:0377 FuHwids              smbios property ProductSku=To be filled by O.E.M.
21:01:06:0378 FuHwids              smbios property BiosVendor=American Megatrends Inc.
21:01:06:0378 FuHwids              smbios property BiosVersion=5.6.5
21:01:06:0378 FuHwids              smbios property BiosMajorRelease=05
21:01:06:0378 FuHwids              smbios property BiosMinorRelease=06
21:01:06:0378 FuHwids              smbios property BaseboardManufacturer=Advantech
21:01:06:0378 FuHwids              smbios property BaseboardProduct=UNO-1372G-E3AE
21:01:06:0389 FuQuirks             no /var/local/lib/fwupd/quirks.d, skipping
21:01:06:0389 XbSilo               attempting to load /var/local/cache/fwupd/quirks.xmlb
21:01:06:0389 XbSilo               file: e1743a81-376a-3092-f3c5-fb73d7756228, current:e1743a81-376a-3092-f3c5-fb73d7756228, cached: (null)
21:01:06:0389 XbSilo               loading silo with file contents
21:01:06:0392 FuEngine             remote lvfs-testing not enabled, so skipping
21:01:06:0393 FuEngine             remote fwupd-tests not enabled, so skipping
21:01:06:0393 FuEngine             no /var/local/lib/fwupd/remotes.d/lvfs/metadata.xml.gz, so skipping
21:01:06:0393 FuEngine             remote vendor not enabled, so skipping
21:01:06:0393 FuEngine             remote vendor-directory not enabled, so skipping
21:01:06:0393 XbSilo               attempting to load /var/local/cache/fwupd/metadata.xmlb
21:01:06:0393 XbSilo               0.03ms      parse blob
21:01:06:0393 XbSilo               0.16ms      loaded file
21:01:06:0393 XbSilo               file: ad08ca8d-7d25-729e-cf51-8df413a2040a, current:ad08ca8d-7d25-729e-cf51-8df413a2040a, cached: (null)
[... cut ...]
21:01:07:0569 FuPlugin             performing destroy() on linux_lockdown
21:01:07:0570 FuPlugin             performing destroy() on uefi_dbx
21:01:07:0570 FuPlugin             performing destroy() on dell
21:01:07:0571 FuPlugin             performing destroy() on synaptics_mst
21:01:07:0571 FuPlugin             performing destroy() on redfish
21:01:07:0572 FuPlugin             performing destroy() on logind
21:01:07:0572 FuPlugin             performing destroy() on uefi
21:01:07:0573 FuPlugin             performing destroy() on linux_tainted
```

superm1, 2020-09-09:
@dnoliver presumably with the recent changes in master for bay trail this one can be closed now, right?

dnoliver, 2020-09-09:
yea I think so, 

This is what I see now for HSI-1. It is limited information,  but should improve when those kernel modules are available :)

```shell
test@advantech-2 fwupd]$ sudo ./build/src/fwupdtool security --force
Loading…                 [***************************************]
Host Security ID: HSI:0+! (v1.5.0)

HSI-1
✔ TPM v2.0:                      Found: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Tpm.Version20
✘ SPI write:                     Not found: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.Bioswe

HSI-2
✔ TPM PCR0 reconstruction:       Valid: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Tpm.ReconstructionPcr0
✘ IOMMU:                         Not found: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Iommu

HSI-3
✘ Intel CET Enabled:             Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.IntelCet.Enabled
✘ Pre-boot DMA protection:       Invalid: https://fwupd.github.io/hsi.html#org.fwupd.hsi.AcpiDmar
✘ Suspend-to-idle:               Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.SuspendToIdle
✘ Suspend-to-ram:                Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.SuspendToRam

HSI-4
✘ Encrypted RAM:                 Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.EncryptedRam
✘ Intel SMAP:                    Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.IntelSmap

Runtime Suffix -U
✘ Firmware updates:              Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Fwupd.Updates

Runtime Suffix -A
✘ Firmware attestation:          Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Fwupd.Attestation

Runtime Suffix -!
✔ Linux kernel:                  Untainted: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.Tainted
✔ Linux swap:                    Encrypted: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.Swap
✔ fwupd plugins:                 Untainted: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Fwupd.Plugins
✘ Linux kernel lockdown:         Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.Lockdown
✘ UEFI secure boot:              Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Uefi.SecureBoot

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```

superm1, 2020-09-09:
Great, thanks!  Yeah once the modules land I guess we should revisit.

closed 2020-09-09 by superm1