fwupd/fwupd #2319: dbxtool fails to list for me

superm1, 2020-08-13:
Got a chance to try what was in master.

```
$ ./build/plugins/uefi-dbx/dbxtool  -l -v
(dbxtool:33629): GLib-GIO-DEBUG: 13:49:09.612: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(dbxtool:33629): FuPluginUefiDbx-DEBUG: 13:49:09.614: EFI_SIGNATURE_LIST X509
(dbxtool:33629): FuPluginUefiDbx-DEBUG: 13:49:09.614: EFI_SIGNATURE_LIST SHA256
(dbxtool:33629): FuPluginUefiDbx-DEBUG: 13:49:09.614: EFI_SIGNATURE_LIST X509
(dbxtool:33629): FuPluginUefiDbx-DEBUG: 13:49:09.614: EFI_SIGNATURE_LIST X509
(dbxtool:33629): FuPluginUefiDbx-DEBUG: 13:49:09.614: EFI_SIGNATURE_LIST SHA256
Failed to load system dbx: more than one EFI_SIGNATURE_LIST items
```

More verbose why there are so many items, mokutil can parse them harder.
```
$ mokutil --dbx
[key 1]
SHA1 Fingerprint: c0:13:86:a9:07:49:64:04:f2:76:c3:c1:85:3a:bf:4a:52:74:af:88
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            61:0c:6a:19:00:00:00:00:00:04
        Signature Algorithm: sha256WithRSAEncryption
        Issuer: C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Root Certificate Authority 2010
        Validity
            Not Before: Jul  6 20:40:23 2010 GMT
            Not After : Jul  6 20:50:23 2025 GMT
        Subject: C=US, ST=Washington, L=Redmond, O=Microsoft Corporation, CN=Microsoft Windows PCA 2010
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:c0:79:bb:3a:b1:f0:0f:84:b8:ad:64:2a:75:16:
                    73:d0:bb:07:f6:3e:0d:9d:14:e4:b1:9f:c1:c8:94:
                    b0:38:7c:1f:d0:33:55:f5:ba:23:66:f5:2e:28:48:
                    53:c7:16:83:ba:f5:51:ac:7e:ac:e0:26:7f:0f:74:
                    fc:59:95:dc:c9:c6:a2:f7:52:70:5a:2c:1d:94:ab:
                    19:bf:af:95:7d:af:66:a1:6f:9b:62:6e:6d:4b:bc:
                    2f:35:6c:de:a4:6a:63:5a:5f:fb:f3:0d:4d:61:cc:
                    0a:7e:31:eb:6c:0a:d0:4d:97:0f:fd:7f:38:46:e6:
                    8a:c7:73:69:76:55:69:96:4c:e4:d8:f0:34:eb:ba:
                    b1:1f:ce:29:7e:c4:4f:9d:13:15:ab:13:1b:72:58:
                    62:56:6c:8a:81:a3:64:77:98:46:65:29:9d:83:14:
                    a5:4c:08:a0:83:d7:23:1f:f3:5f:df:6f:2c:cf:da:
                    16:d8:0e:72:04:28:d8:6b:3e:f8:13:b1:7c:a2:17:
                    79:4f:7e:dc:3a:e4:9d:70:27:6b:bf:db:fc:1e:c7:
                    07:d8:c0:be:0b:93:1e:28:e0:73:6d:d2:54:e9:28:
                    4c:bf:6b:5d:9f:ff:5d:33:12:37:95:25:61:34:6a:
                    42:cb:7c:9d:3a:bb:88:59:e1:a3:42:6d:3a:50:5b:
                    48:d1
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            1.3.6.1.4.1.311.21.1:
                ...
            X509v3 Subject Key Identifier:
                D1:4F:A9:8A:07:08:CE:F4:24:18:98:E5:00:FF:F3:D6:79:1D:37:BC
            1.3.6.1.4.1.311.20.2:
                .
.S.u.b.C.A
            X509v3 Key Usage:
                Digital Signature, Certificate Sign, CRL Sign
            X509v3 Basic Constraints: critical
                CA:TRUE
            X509v3 Authority Key Identifier:
                keyid:D5:F6:56:CB:8F:E8:A2:5C:62:68:D1:3D:94:90:5B:D7:CE:9A:18:C4

            X509v3 CRL Distribution Points:

                Full Name:
[... 384 lines cut ...]
  cc8eec6eb9212cbf897a5ace7e8abeece1079f1a6def0a789591cb1547f1f084
  cf13a243c1cd2e3c8ceb7e70100387cecbfb830525bbf9d0b70c79adf3e84128
  d89a11d16c488dd4fbbc541d4b07faf8670d660994488fe54b1fbff2704e4288
  d9668ab52785086786c134b5e4bddbf72452813b6973229ab92aa1a54d201bf5
  da3560fd0c32b54c83d4f2ff869003d2089369acf2c89608f8afa7436bfa4655
  df02aab48387a9e1d4c65228089cb6abe196c8f4b396c7e4bbc395de136977f6
  df91ac85a94fcd0cfb8155bd7cbefaac14b8c5ee7397fe2cc85984459e2ea14e
  e051b788ecbaeda53046c70e6af6058f95222c046157b8c4c1b9c2cfc65f46e5
  e051b788ecbaeda53046c70e6af6058f95222c046157b8c4c1b9c2cfc65f46e5
  e36dfc719d2114c2e39aea88849e2845ab326f6f7fe74e0e539b7e54d81f3631
  e39891f48bbcc593b8ed86ce82ce666fc1145b9fcbfd2b07bad0a89bf4c7bfbf
  e6856f137f79992dc94fa2f43297ec32d2d9a76f7be66114c6a13efc3bcdf5c8
  eaff8c85c208ba4d5b6b8046f5d6081747d779bada7768e649d047ff9b1f660c
  ee83a566496109a74f6ac6e410df00bb29a290e0021516ae3b8a23288e7e2e72
  eed7e0eff2ed559e2a79ee361f9962af3b1e999131e30bb7fd07546fae0a7267
  f1b4f6513b0d544a688d13adc291efa8c59f420ca5dcb23e0b5a06fa7e0d083d
  f2a16d35b554694187a70d40ca682959f4f35c2ce0eab8fd64f7ac2ab9f5c24a
  f31fd461c5e99510403fc97c1da2d8a9cbe270597d32badf8fd66b77495f8d94
  f48e6dd8718e953b60a24f2cbea60a9521deae67db25425b7d3ace3c517dd9b7
  c805603c4fa038776e42f263c604b49d96840322e1922d5606a9b0bbb5bffe6f
  1f16078cce009df62edb9e7170e66caae670bce71b8f92d38280c56aa372031d
  37a480374daf6202ce790c318a2bb8aa3797311261160a8e30558b7dea78c7a6
  408b8b3df5abb043521a493525023175ab1261b1de21064d6bf247ce142153b9
  540801dd345dc1c33ef431b35bf4c0e68bd319b577b9abe1a9cff1cbc39f548f
```

hughsie, 2020-08-13:
What does the original `dbxtool` print out of interest? Thanks.

superm1, 2020-08-13:
```
$ ./dbxtool -l -v
Getting next EFI_SIGNATURE_DATA
Getting next ESL buffer
   1: {zero} {x509_cert} 3082066b30820453a003020102020a610c6a19000000000004300d06092a864886f70d01010b0500308188310b3009060355040613025553311330110603550408130a57617368696e67746f6e3110300e060355040713075265646d6f6e64311e301c060355040a13154d6963726f736f667420436f72706f726174696f6e31323030060355040313294d6963726f736f667420526f6f7420436572746966696361746520417574686f726974792032303130301e170d3130303730363230343032335a170d3235303730363230353032335a3079310b3009060355040613025553311330110603550408130a57617368696e67746f6e3110300e060355040713075265646d6f6e64311e301c060355040a13154d6963726f736f667420436f72706f726174696f6e312330210603550403131a4d6963726f736f66742057696e646f777320504341203230313030820122300d06092a864886f70d01010105000382010f003082010a0282010100c079bb3ab1f00f84b8ad642a751673d0bb07f63e0d9d14e4b19fc1c894b0387c1fd03355f5ba2366f52e284853c71683baf551ac7eace0267f0f74fc5995dcc9c6a2f752705a2c1d94ab19bfaf957daf66a16f9b626e6d4bbc2f356cdea46a635a5ffbf30d4d61cc0a7e31eb6c0ad04d970ffd7f3846e68ac77369765569964ce4d8f034ebbab11fce297ec44f9d1315ab131b725862566c8a81a36477984665299d8314a54c08a083d7231ff35fdf6f2ccfda16d80e720428d86b3ef813b17ca217794f7edc3ae49d70276bbfdbfc1ec707d8c0be0b931e28e0736dd254e9284cbf6b5d9fff5d331237952561346a42cb7c9d3abb8859e1a3426d3a505b48d10203010001a38201e3308201df301006092b06010401823715010403020100301d0603551d0e04160414d14fa98a0708cef4241898e500fff3d6791d37bc301906092b0601040182371402040c1e0a00530075006200430041300b0603551d0f040403020186300f0603551d130101ff040530030101ff301f0603551d23041830168014d5f656cb8fe8a25c6268d13d94905bd7ce9a18c430560603551d1f044f304d304ba049a0478645687474703a2f2f63726c2e6d6963726f736f66742e636f6d2f706b692f63726c2f70726f64756374732f4d6963526f6f4365724175745f323031302d30362d32332e63726c305a06082b06010505070101044e304c304a06082b06010505073002863e687474703a2f2f7777772e6d6963726f736f66742e636f6d2f706b692f63657274732f4d6963526f6f4365724175745f323031302d30362d32332e63727430819d0603551d2004819530819230818f06092b0601040182372e03308181303d06082b060105050702011631687474703a2f2f7777772e6d6963726f736f66742e636f6d2f504b492f646f63732f4350532f64656661756c742e68746d304006082b0601050507020230341e32201d004c006500670061006c005f0050006f006c006900630079005f00530074006100740065006d0065006e0074002e201d300d06092a864886f70d01010b050003820201002e41a686b5066ff08085fa3bca17e9c9fae439c29470c36494c3d856a6908efee49af46df56f8e538d5aa8f3aedb466cbe7f1d54561b3c1d71c45115547ebfeea5954233fd0d902424e3f9dc96cafcb8acbff4c23956b8bbed73b317dd7e8650238b5624cabba61d9a872f2785e7a1b60a9c0d1b8cf3006241ba48748782fd50c9f48729c303aa2bdf1a2979e812249a86edd02ed34081f5075f3306545d40b5f7b162fd4d48f76e4147521cbb1bc2573aa8995693d4c6de26a8607586bbec62a6f01d0445df3ea784d15b4423632536776fae5bdc22d514236a417fd042a6dbef257b04e3d296376206aff81b0f8eb3399abb89f53506e5a45bc38c9e375f53d1a337fda44fe81b0e6b76e4b88fb0c2eafd75f72c41b79ca3e11e05fe9792cb7f59036da84e8d4e8017d4d572f656e4489fa323ba06a0c08ed1884f9320f2705fd86b72a32049fc770c5dc5c7e1020f3842100edb02ae9a371d5080291ea4a7d9c69a2555fd40caad6410e831f91254791af20ed8d6ab1e33fe02e7266d61498ff125c28b7499dff9931a901ceedd94330e4250db7f50f89f6282eca682167c66bcec99b0c1585da8b09a611491d1992f49e43e8199d6e6efcae3fd3eeeec098603070d1b0d7cebf45ac995cf87120a5decc50292cd059972ca7df12a101838e431a328b4e64cc552a39c6ac77dc07109040d70de023f87ee56a1ecebb54c85
Getting next EFI_SIGNATURE_DATA
Getting next EFI_SIGNATURE_LIST
   2: {microsoft} {sha256} 80b4d96931bf0d02fd91a61e19d14f1da452e66db2408ca8604d411f92659f0a
Getting next esd element
   3: {microsoft} {sha256} f52f83a3fa9cfbd6920f722824dbe4034534d25b8507246b3b957dac6e1bce7a
Getting next esd element
   4: {microsoft} {sha256} c5d9d8a186e2c82d09afaa2a6f7f2e73870d3e64f72c4e08ef67796a840f0fbd
Getting next esd element
   5: {microsoft} {sha256} 363384d14d1f2e0b7815626484c459ad57a318ef4396266048d058c5a19bbf76
Getting next esd element
   6: {microsoft} {sha256} 1aec84b84b6c65a51220a9be7181965230210d62d6d33c48999c6b295a2b0a06
Getting next esd element
   7: {microsoft} {sha256} e6ca68e94146629af03f69c2f86e6bef62f930b37c6fbcc878b78df98c0334e5
Getting next esd element
   8: {microsoft} {sha256} c3a99a460da464a057c3586d83cef5f4ae08b7103979ed8932742df0ed530c66
Getting next esd element
   9: {microsoft} {sha256} 58fb941aef95a25943b3fb5f2510a0df3fe44c58c95e0ab80487297568ab9771
Getting next esd element
  10: {microsoft} {sha256} 5391c3a2fb112102a6aa1edc25ae77e19f5d6f09cd09eeb2509922bfcd5992ea
Getting next esd element
  11: {microsoft} {sha256} d626157e1d6a718bc124ab8da27cbb65072ca03a7b6b257dbdcbbd60f65ef3d1
Getting next esd element
  12: {microsoft} {sha256} d063ec28f67eba53f1642dbf7dff33c6a32add869f6013fe162e2c32f1cbe56d
Getting next esd element
  13: {microsoft} {sha256} 29c6eb52b43c3aa18b2cd8ed6ea8607cef3cfae1bafe1165755cf2e614844a44
Getting next esd element
  14: {microsoft} {sha256} 90fbe70e69d633408d3e170c6832dbb2d209e0272527dfb63d49d29572a6f44c
Getting next esd element
  15: {microsoft} {sha256} 075eea060589548ba060b2feed10da3c20c7fe9b17cd026b94e8a683b8115238
Getting next esd element
  16: {microsoft} {sha256} 07e6c6a858646fb1efc67903fe28b116011f2367fe92e6be2b36999eff39d09e
Getting next esd element
  17: {microsoft} {sha256} 09df5f4e511208ec78b96d12d08125fdb603868de39f6f72927852599b659c26
Getting next esd element
  18: {microsoft} {sha256} 0bbb4392daac7ab89b30a4ac657531b97bfaab04f90b0dafe5f9b6eb90a06374
Getting next esd element
[... cut ...]
 238: {microsoft} {sha256} 1f16078cce009df62edb9e7170e66caae670bce71b8f92d38280c56aa372031d
Getting next esd element
 239: {microsoft} {sha256} 37a480374daf6202ce790c318a2bb8aa3797311261160a8e30558b7dea78c7a6
Getting next esd element
 240: {microsoft} {sha256} 408b8b3df5abb043521a493525023175ab1261b1de21064d6bf247ce142153b9
Getting next esd element
 241: {microsoft} {sha256} 540801dd345dc1c33ef431b35bf4c0e68bd319b577b9abe1a9cff1cbc39f548f
Getting next EFI_SIGNATURE_DATA
Getting next EFI_SIGNATURE_LIST
```

closed 2020-08-13 by hughsie