fwupd/fwupd #2322: OUI Quirk Issue

Darkovian, 2020-08-14:
Hello, I've been trying to get fwupd working today after putting it off awhile and ran into this issue. The link mentioned submitting an issue with the information if I wasn't comfortable doing it the other way. Here is the information fwupd is giving me:

[DeviceInstanceId=OUI\030302]
Vendor = SK hynix
VendorId = ATA:0x1C5C

Thanks, hope this helps!

superm1, 2020-08-14:
Would you be able to submit a PR for adding this?   The file to patch is https://github.com/fwupd/fwupd/blob/master/plugins/ata/ata.quirk

Darkovian, 2020-08-14:
I believe I managed to submit a PR for the addition. Hopefully I got it right, not totally used to all of git just yet.

Link: [https://github.com/fwupd/fwupd/pull/2323](https://github.com/fwupd/fwupd/pull/2323)

superm1, 2020-08-14:
closed by https://github.com/fwupd/fwupd/pull/2323

closed 2020-08-14 by superm1