fwupd/fwupd #2328: PCI BCR Security checks in Intel Bay Trail have incorrect offset

dnoliver, 2020-08-18:
**Describe the bug**
A clear and concise description of what the bug is.

The current BCR offset defined in https://github.com/fwupd/fwupd/blob/master/plugins/pci-bcr/fu-plugin-pci-bcr.c#L17 is not correct for Bay Trail. 
The correct offset for Bay Trail is `0x54h`
Having and incorrect offset makes the 3 checks for the BIOS Lock functionality to be not correct.

The current output of `lspci` in the test device is:

```shell
lspci -xxx -nn -s0:1f.0
00:1f.0 ISA bridge [0601]: Intel Corporation Atom Processor Z36xxx/Z37xxx Series Power Control Unit [8086:0f1c] (rev 11)
00: 86 80 1c 0f 07 00 10 02 11 00 01 06 00 00 80 00
10: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
20: 00 00 00 00 00 00 00 00 00 00 00 00 86 80 1c 0f
30: 00 00 00 00 e0 00 00 00 00 00 00 00 00 00 00 00
40: 03 04 00 00 02 30 d0 fe 03 05 00 00 02 c0 d0 fe
50: 02 80 d0 fe 02 10 d0 fe 02 00 f0 fe 02 50 d0 fe
60: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
70: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
80: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
90: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
d0: 00 00 00 00 00 00 00 00 cf ff 00 00 00 00 00 00
e0: 09 00 0c 10 00 00 00 00 00 00 00 00 00 00 00 00
f0: 01 c0 d1 fe 00 00 00 00 1a 0f 13 01 03 03 00 00
```

Running the `fwupdtool security --force` (with the 0xdc) on this platform yields:

```shell
HSI-1
✔ SPI write:                     Disabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.Bioswe
✔ TPM v2.0:                      Found: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Tpm.Version20
✔ UEFI dbx:                      Found: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Uefi.Dbx
✘ SPI BIOS region:               Unlocked: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.SmmBwp
✘ SPI lock:                      Disabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.Ble
```

By changing the offset to the correct one defined in the platform spec:

```diff
diff --git a/plugins/pci-bcr/fu-plugin-pci-bcr.c b/plugins/pci-bcr/fu-plugin-pci-bcr.c
index e2fc1e9f..138423e3 100644
--- a/plugins/pci-bcr/fu-plugin-pci-bcr.c
+++ b/plugins/pci-bcr/fu-plugin-pci-bcr.c
@@ -14,7 +14,7 @@ struct FuPluginData {
        guint8                   bcr;
 };
 
-#define BCR                    0xdc
+#define BCR                    0x54
 #define BCR_WPD                        (1 << 0)
 #define BCR_BLE                        (1 << 1)
 #define BCR_SMM_BWP            (1 << 5)
```

The test execution on that same platform yields (with the 0x54 offset):

```shell
HSI-1
✔ SPI write:                     Disabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.Bioswe
✔ TPM v2.0:                      Found: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Tpm.Version20
✔ UEFI dbx:                      Found: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Uefi.Dbx
✘ SPI BIOS region:               Unlocked: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.SmmBwp
✔ SPI lock:                      Enabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.Ble
```

**Steps to Reproduce**
Steps to reproduce the behavior.

1. In a device using Bay Trail, run the latest master version of `fwupdtool security --force`

**Expected behavior**
A clear and concise description of what you expected to happen.

The tool should look at the correct offset, and report 

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
```

Latest Master

```
# ./build/src/fwupdtool --version
client version: 1.4.1-336-gfd0ee515
compile-time dependency versions
        gusb:   0.3.5
        efivar: 37
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```

I may have a broken environment, this say:

```shell
./build/src/fwupdmgr get-devices --show-all-devices
Unable to connect to service
```

**Additional questions**
- Operating system and version: Fedora Server 32
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2020-08-18:
This is an Atom I guess. We're moving the BCR tests into the kernel, but this is going to take some time. In the kernel the chipset quirks are already set up, so looking months into the future we can expect this to be fixed without kludges. For the more immediate fix I'm not sure what to do. I'm guessing you have a match against PCI device `00:1F.0` i.e. the LPC ISA bridge.

Maybe the way is to identify the CPU as Atom and change the BCR register in the code. @superm1 got any clever ideas?

superm1, 2020-08-18:
>Maybe the way is to identify the CPU as Atom and change the BCR register in the code. @superm1 got any clever ideas?

Do we know that all atoms export at that offset?  Or it's just bay trail?  My primary thought is we should probably move this to a quirk file and let the attribute load from that.

dnoliver, 2020-08-18:
To clarify:

I have 2 Intel Atom based devices.
One with Apollo Lake, the other with Bay Trail

The Apollo Lake: 4x Intel(R) Atom(TM) Processor E3950 @ 1.60GHz
For this one, the current `0xdc` offset is correct

The Bay Trail: 4x Intel(R) Atom(TM) CPU E3845 @ 1.91GHz
For this one, the `0x54h` is the correct offset

hughsie, 2020-08-19:
>  One with Apollo Lake, the other with Bay Trail

Do you know how can we identify "Bay Trail" from the CPUID?

dnoliver, 2020-08-19:
cpuid cli identifies it as Bay Trail

```shell
CPU 0:
   vendor_id = "GenuineIntel"
   version information (1/eax):
      processor type  = primary processor (0)
      family          = 0x6 (6)
      model           = 0x7 (7)
      stepping id     = 0x9 (9)
      extended family = 0x0 (0)
      extended model  = 0x3 (3)
      (family synth)  = 0x6 (6)
      (model synth)   = 0x37 (55)
      (simple synth)  = Intel Atom (unknown type) (Bay Trail D0) [Silvermont], 22nm

$ cpuid | grep 'Bay Trail'
      (simple synth)  = Intel Atom (unknown type) (Bay Trail D0) [Silvermont], 22nm
   (synth) = Intel Atom E3800 (Bay Trail-I D0) [Silvermont], 22nm
      (simple synth)  = Intel Atom (unknown type) (Bay Trail D0) [Silvermont], 22nm
   (synth) = Intel Atom E3800 (Bay Trail-I D0) [Silvermont], 22nm
      (simple synth)  = Intel Atom (unknown type) (Bay Trail D0) [Silvermont], 22nm
   (synth) = Intel Atom E3800 (Bay Trail-I D0) [Silvermont], 22nm
      (simple synth)  = Intel Atom (unknown type) (Bay Trail D0) [Silvermont], 22nm
   (synth) = Intel Atom E3800 (Bay Trail-I D0) [Silvermont], 22nm
```

And here is a list of Bay Trail products https://ark.intel.com/content/www/us/en/ark/products/codename/55844/bay-trail.html

hughsie, 2020-08-19:
Do you have the cpuid output from the Apollo Lake system too please.

dnoliver, 2020-08-19:
```shell
CPU 0:
   vendor_id = "GenuineIntel"
   version information (1/eax):
      processor type  = primary processor (0)
      family          = 0x6 (6)
      model           = 0xc (12)
      stepping id     = 0x9 (9)
      extended family = 0x0 (0)
      extended model  = 0x5 (5)
      (family synth)  = 0x6 (6)
      (model synth)   = 0x5c (92)
      (simple synth)  = Intel Atom (unknown type) (Apollo Lake B0/B1/D0) [Goldmont], 14nm
```

dnoliver, 2020-09-02:
Unfortunately, I made a mistake :(

The offset for Bay Trail I reported is not the correct one.
The correct memory location that reports the state of the PCI BCR is `0xfed010fc`

With BIOS Lock enabled, the reported state is:

BCR_WPD = 0
BCR_BLE = 1 
BCR_SMM_BWP = 1

```shell
[root@bay-trail fwupd]# busybox devmem 0xfed010fc
0x00000022
```

With BIOS Lock disabled, I get the following report:

BCR_WPD = 0
BCR_BLE = 0 
BCR_SMM_BWP = 0

```shell
[root@bay-trail fwupd]# busybox devmem 0xfed010fc
0x00000000
```

`lspci -xxx` output does not change if you enable or disable BIOS Lock on this device. The only way of checking that value I found was with `busybox devmem`

On the other hand, after some debugging, I noticed that the the quirk is not applied in my Bay Trail device. 
I never hit this line on a debugging session https://github.com/fwupd/fwupd/blob/master/plugins/pci-bcr/fu-plugin-pci-bcr.c#L40.

hughsie, 2020-09-03:
> Unfortunately, I made a mistake

No worries dude, we're all imperfect :)

> I noticed that the the quirk is not applied in my Bay Trail device.

Can you attach the `sudo fwupdmgr get-devices --show-all-devices` output of the latest git please? That should tell us if we got the instance ID wrong.

dnoliver, 2020-09-03:
```shell
[test@advantech-2 fwupd]$ sudo /usr/bin/fwupdmgr get-devices --show-all-devices
UNO-1372G-E3AE
│
├─        Intel(R) Atom™ CPU  E3845  @ 1.91GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x0000090c
│     Vendor:              Intel
│     GUIDs:               b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                          30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                          466ccad0-1a32-567f-8623-13721a6a0167 ← CPUID\PRO_0&FAM_06&MOD_37
│                          3d7640b5-47ba-5f4e-be46-fed711c84c77 ← CPUID\PRO_0&FAM_06&MOD_37&STP_9
│     Device Flags:        • Internal device
│   
├─Atom Processor Z36xxx/Z37xxx Series Graphics & Display:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     11
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               ec0fad98-3d0c-5b17-b1dd-ec2beeace2ef ← PCI\VEN_8086&DEV_0F31&REV_11
│                          ce801c1e-fe8e-52e8-ae6d-439478884437 ← PCI\VEN_8086&DEV_0F31
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─QUECTEL Mobile Broadband Module:
│     Device ID:           8009709f7b15ea18e2c69680834a18c79a7ec260
│     Summary:             Mobile broadband device
│     Current version:     EC25AFFAR07A08M4G
│     Vendor:              QUALCOMM INCORPORATED (USB:0x2c7c)
│     GUIDs:               397b2c13-36fb-582e-b26c-d4640690223d ← USB\VID_2C7C&PID_0125&REV_0318&CARRIER_DEFAULT
│                          1a2996cb-f86e-5583-a464-e1b96e1c6ae9 ← USB\VID_2C7C&PID_0125&REV_0318
│                          587bf468-6859-5522-93a7-6cce552a0aa3 ← USB\VID_2C7C&PID_0125
│                          22ae45db-f68e-5c55-9c02-4557dca238ec ← USB\VID_2C7C
│     Device Flags:        • Updatable
│   
├─RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller (RTL8111/8168 PCI Express Gigabit Ethernet controller):
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Current version:     06
│     Vendor:              Realtek Semiconductor Co., Ltd. (PCI:0x10EC)
│     GUIDs:               0c704ab8-4b8b-5ac9-99ef-115002a66f8d ← PCI\VEN_10EC&DEV_8168&REV_06
│                          f2e84a73-581d-5f30-8b04-7bdc404ce0bb ← PCI\VEN_10EC&DEV_8168
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
├─SQF-S25M8-64G-SAE:
│     Device ID:           df24e4ff1af516b7c30ec99405c9315e8cdfdab7
│     Summary:             ATA Drive
│     Current version:     SAFM02A3
│     Vendor:              Phison (ATA:0x1987)
│     Serial Number:       FEBC0799060300004844
│     GUIDs:               e29ae95c-7d6b-5c14-bcd1-c8fc3680ab03 ← IDE\SQF-S25M8-64G-SAE_______________________SAFM02A3
│                          04acff22-1c04-5d54-842c-db9635c1eaef ← IDE\0SQF-S25M8-64G-SAE_______________________
│                          6ee7cc9a-b017-562f-b239-8ed4c7b62e85 ← SQF-S25M8-64G-SAE
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:           1d70a31dd3ff0941761550a38436fc990c6c3ff0
│ │   Current version:     15
│ │   Minimum Version:     15
│ │   Vendor:              Advantech (DMI:American Megatrends Inc.)
│ │   GUIDs:               7039436b-6acf-433b-86a1-368ec2ef7e1f
│ │                        230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │                        733edd7c-75ee-5db0-8a0a-3138c35f755c ← UEFI\RES_{7039436B-6ACF-433B-86A1-368EC2EF7E1F}
│ │   Device Flags:        • Internal device
│ │                        • Updatable
│ │                        • Requires AC power
│ │                        • Needs a reboot after installation
│ │                        • Cryptographic hash verification is available
│ │                        • Device is usable for the duration of the update
│ │ 
│ └─UEFI dbx:
│       Device ID:         362301da643102b9f38477387e2193e57abaa590
│       Summary:           UEFI Revocation Database
│       Current version:   78
│       Minimum Version:   78
│       Vendor:            UEFI:Linux Foundation
│       Install Duration:  1 second
│       GUIDs:             c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
│                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:      • Internal device
│                          • Updatable
│                          • Needs a reboot after installation
│     
└─TPM:
  │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
  │   Current version:     5.62.12.13824
  │   Vendor:              Infineon (TPM:IFX)
  │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6 ← system-tpm
  │                        5eebb112-75ad-5536-b173-a11eb3399402 ← TPM\VEN_IFX&DEV_0000
  │                        1a10e308-1ceb-5b61-bbd0-07461469ab8f ← TPM\VEN_IFX&MOD_SLB9665
  │                        6d81ab63-db2e-50ac-934f-6be9accf5e02 ← TPM\VEN_IFX&DEV_0000&VER_2.0
  │                        5cec2f57-3b14-5f7d-9436-882896e3923d ← TPM\VEN_IFX&MOD_SLB9665&VER_2.0
  │   Device Flags:        • Internal device
  │ 
  └─Event Log:
        Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
        GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
        Device Flags:      • Internal device
```

hughsie, 2020-09-04:
Does the wip/hughsie/atom branch get things working?

dnoliver, 2020-09-04:
I am now getting the quirk applied, so that is solved.
But I am not getting the correct values yet.

[image]

So, the BcrAdd is 252 -> FCh. Is it truncated?

And the values read does not match the output of devmem

```shell
[root@advantech-2 fwupd]# busybox devmem 0xfed010fc
0x00000022
```

I tried forcing the offset to be `0xfed010fc`

```shell
goffset offset = 0xfed010fc;
if (!fu_udev_device_pread (device, /*priv->bcr_addr*/ offset, &priv->bcr, error)) {
	g_prefix_error (error, "could not read BCR");
	return FALSE;
}
```

but I get this error (failed with Success):

```shell
18:17:47:0711 FuEngine             failed to add udev device /sys/devices/pci0000:00/0000:00:1f.0: failed to add device using on pci_bcr: could not read BCRfailed to read from port 0xfed010fc: Success
```

hughsie, 2020-09-04:
Ohh, 0xfed010fc is a physical memory location?! Then the port read isn't going to work of course, my bad. Out of interest, does the port read of 0x54h work correctly now the quirk entry is fixed?

hughsie, 2020-09-04:
But is bioswp obtained correctly?

dnoliver, 2020-09-04:
If bioswp == BCR_SMM_BWP, then no.
The value read at the 0x54 handle is 0x02, which means BCR_SMM_BWP = 0. 
And the value reported in 0xfed010fc is 0x22, which means BCR_SMM_BWP = 1.
So, with the 0x54 handle, I get the following result:

```shell
Process ./build/src/fwupdtool created; pid = 13425
Listening on port 1234
Remote debugging from host ::1, port 39008
Loading…                 [***************************************]
Host Security ID: HSI:0+! (v1.5.0)

HSI-1
✔ SPI lock:                      Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.Ble
✔ SPI write:                     Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.Bioswe
✔ TPM v2.0:                      Found: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Tpm.Version20
✘ SPI BIOS region:               Unlocked: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.SmmBwp
```

hughsie, 2020-09-07:
I think, perhaps, we should stop trying to do this in userspace. Some background: fwupd has two ways to get the SMM_BWP -- the userspace way of getting BCR from an address which works in 99% of cases, or the kernel exporting a node which reads the BCR from the right address. @dgutson is working on the kernel driver right now, but it's not yet upstream. I know for Atom hardware it's a little more involved than just reading out a register for the general case. Maybe for Atom we just set the BCR to 0x0 and skip these checks?

dnoliver, 2020-09-09:
Agree with you,

This tests works great on Apollo Lake (at least in the device I tested them). We discovered that we had BIOS Lock disabled thanks to this tests!

But on Bay Trail, it makes sense to skip. There are alternative ways of checking for BIOS Lock functionality on it as well (not as comfortable as this one, though)

closed 2020-09-09 by hughsie