fwupd/fwupd #2333: Presentation: Easiest way to infect hardware? (Introducing the Linux Vendor Firmware Service)

mendel5, 2020-08-23:
At https://people.freedesktop.org/~hughsient/temp/LVFS.pdf there is a presentation called "Introducing the Linux Vendor Firmware Service".

The title of slide number 5 is "Easiest way to infect hardware?" and this diagram is shown:
[image]

What does the light blue "???" stand for?

hughsie, 2020-08-23:
You need to watch the presentation: https://www.youtube.com/watch?v=u7A7VqQCpIE&t=38s

A github issue really isn't the best way to ask these kinds of questions.

closed 2020-08-23 by hughsie

mendel5, 2020-08-23:
Thanks for the link, I just watched the presentation.

The answer seems to be:
> The easiest way to infect hardware is through unpatched vulnerabilities. Often there is a patch/ firmware update available for a specific kind of hardware but many people don't install these firmware updates.