fwupd/fwupd #2335: Intel Boot Guard status invalid in Intel Apollo Lake

dnoliver, 2020-08-24:
**Describe the bug**
A clear and concise description of what the bug is.

The following is the output of the Host Security ID check in an Apollo Lake device

```shell
[root@fitlet-1 fwupd]# ./build/src/fwupdtool security --force
Loading…                 [***************************************]
Host Security ID: HSI:1+! (v1.5.0)

HSI-1
✔ Intel DCI debugger:            Disabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.IntelDci.Enabled
✔ SPI BIOS region:               Locked: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.SmmBwp
✔ SPI lock:                      Enabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.Ble
✔ SPI write:                     Disabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Spi.Bioswe
✔ TPM v2.0:                      Found: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Tpm.Version20
✔ TXE manufacturing mode:        Locked: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Mei.ManufacturingMode
✔ TXE override:                  Locked: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Mei.OverrideStrap
✔ TXE v0:3.1.70.2334:            Valid: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Mei.Version

HSI-2
✔ IOMMU:                         Enabled: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Iommu
✔ Intel BootGuard:               Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.IntelBootguard
✔ Intel BootGuard OTP fuse:      Valid: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.IntelBootguard
✔ Intel DCI debugger:            Locked: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.IntelDci.Locked
✔ TPM PCR0 reconstruction:       Valid: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.Tpm.ReconstructionPcr0
✘ Intel BootGuard ACM protected: Invalid: https://fwupd.github.io/hsi.htmlorg.fwupd.hsi.IntelBootguard.Acm
✘ Intel BootGuard verified boot: Invalid: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.IntelBootguard
```

As noted in the output, `Intel BootGuard ACM protected` and `Intel BootGuard verified boot` tests are failing.
But running the TXEInfo.efi tool in that device says that Measured and Verified Boot is enabled. Also, the Boot guard policy is listed as VM (which AFAIK stands for Verified and Measured).
There is no clear information about ACM protected (maybe the Security Version Number = 3 is related to that?)

```
Intel(R) TXEInfo Version: 3.1.76.2360
Copyright(C) 2005 - 2020, Intel Corporation. All rights reserved.

FW Status Register1: 0x80000245
FW Status Register2: 0x39850000
FW Status Register3: 0x30B50608
FW Status Register4: 0x00080004
FW Status Register5: 0x00000000
FW Status Register6: 0x40000000

  CurrentState:                               Normal
  ManufacturingMode:                          Disabled
  FlashPartition:                             Valid
  OperationalState:                           CM0 with UMA
  InitComplete:                               Complete
  BUPLoadState:                               Success
  ErrorCode:                                  No Error
  ModeOfOperation:                            Normal
  SPI Flash Log:                              Present
  Phase:                                      BringUp
  TXE File System Corrupted:                  No
  PhaseStatus:                                UNKNOWN
  FPF and TXE Config Status:                  Committed
FW Capabilities value is 0x31109440
Feature enablement is 0x31109440
Platform type is 0x73FF0321
Intel(R) TXE code versions:
                 
    Table Type 255 ( 0x FF ) found, size of   0 (0x 00 ) bytes 
BIOS Version                                 FLT2.46.38
Vendor ID                                    8086
PCH Version                                  B
FW Version                                   3.1.70.2334
Security Version (SVN)                       3
Number of IFWI Modules                       5
[... 222 lines cut ...]
│                          e25079dc-149b-5eb7-aee9-c7ebe27e4fd3 ← TS128GMTS600
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:           f3051add1ff6b15831703b30f1782eff9b2173ee
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
                           • Supported on remote server
```

**Additional questions**
- Operating system and version: Fedora Server 32
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2020-08-26:
> but the structure may be different for TXE

Urgh, I did not know this. Is there any documentation on the hfsts registers for TXE? In coreboot perhaps? Thanks.

LeonidFirdmanCompulab, 2020-09-06:
Hello,

The TXE Host Firmware Status Registers are described in the "Apollo Lake Platform: Intel Architecture Firmware Specification—Volume 2 of 2 BIOS Specification" document (document number: 559811).

The link for the document: http://www.intel.com/cd/edesign/library/asmo-na/eng/559811.htm
Intel customer's or employee's access is needed.

hughsie, 2020-09-06:
> Intel customer's or employee's access is needed

I'm afraid I'm neither :(

closed 2020-09-08 by hughsie

dnoliver, 2020-09-08:
Confirmed that tests are skipped in Apollo Lake by building and running latest master version

Thank you!

```shell
[root@fitlet-1 fwupd]# ./build/src/fwupdtool security --force
Loading…                 [***************************************]
Host Security ID: HSI:2 (v1.5.0)

HSI-1
✔ Intel DCI debugger:            Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.IntelDci.Enabled
✔ SPI BIOS region:               Locked: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.SmmBwp
✔ SPI lock:                      Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.Ble
✔ SPI write:                     Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Spi.Bioswe
✔ TPM v2.0:                      Found: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Tpm.Version20
✔ TXE manufacturing mode:        Locked: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Mei.ManufacturingMode
✔ TXE override:                  Locked: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Mei.OverrideStrap
✔ TXE v0:3.1.70.2334:            Valid: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Mei.Version

HSI-2
✔ IOMMU:                         Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Iommu
✔ Intel DCI debugger:            Locked: https://fwupd.github.io/hsi.html#org.fwupd.hsi.IntelDci.Locked
✔ TPM PCR0 reconstruction:       Valid: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Tpm.ReconstructionPcr0

HSI-3
✔ Suspend-to-ram:                Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.SuspendToRam
✘ Intel CET Enabled:             Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.IntelCet.Enabled
✘ Pre-boot DMA protection:       Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.AcpiDmar
✘ Suspend-to-idle:               Disabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.SuspendToIdle

HSI-4
✔ Intel SMAP:                    Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.IntelSmap
✘ Encrypted RAM:                 Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.EncryptedRam

Runtime Suffix -U
✘ Firmware updates:              Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Fwupd.Updates

Runtime Suffix -A
✘ Firmware attestation:          Not supported: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Fwupd.Attestation

Runtime Suffix -!
✔ Linux kernel:                  Untainted: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.Tainted
✔ Linux kernel lockdown:         Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.Lockdown
✔ Linux swap:                    Encrypted: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Kernel.Swap
✔ UEFI secure boot:              Enabled: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Uefi.SecureBoot
✔ fwupd plugins:                 Untainted: https://fwupd.github.io/hsi.html#org.fwupd.hsi.Fwupd.Plugins
```

xaionaro, 2022-08-04:
Could somebody advise where I can get the `TXEInfo.efi` binary?

Sean-StarLabs, 2022-08-04:
Intel Resource and Documentation Center if you have an NDA

xaionaro, 2022-08-04:
> if you have an NDA

I have. But how to find the tool? I tried to enter "TXEInfo.efi", but do not see it. Would you mind to give a link (to the intel's page)?

LeonidFirdman, 2022-08-06:
> > if you have an NDA
> 
> I have. But how to find the tool? I tried to enter "TXEInfo.efi", but do not see it. Would you mind to give a link (to the intel's page of the tool)?

Search for the "Apollo Lake Intel Trusted Execution Engine Firmware" in the Intel Database.
https://www.intel.com/content/www/us/en/search.html?ws=text#q=Apollo%20Lake%20Intel%20Trusted%20Execution%20Engine%20Firmware&sort=relevancy

xaionaro, 2022-08-08:
@LeonidFirdman : Thank you! Found.