fwupd/fwupd #2342: DFU Plugin: Can not support the DFU FW package larger than 256MB

zhj8407, 2020-08-28:
**Describe the bug**
The DFU plugin can not support the package larger than 256MB.
In the function dfu_target_download_element_dfu(), it uses a guint16 to store the number of chunks. Assume the transfer_size reported by DFU capable device is 4096. The maximum size of firmware package supported is 65536 * 4096 = 256MB. Some devices like Poly Studio, have larger firmware package. They are not supported by this plugin.

**Steps to Reproduce**
1. Plug in the Poly Studio
2. Run "dfu-tool write [firmware.dfu]"

**Expected behavior**
The whole firmware package could be sent to device

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.3.11
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.3.11
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
apt install fwupd

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices

─Poly Richard DFU:
│     Device ID:           cf294bf55b333004beb7c41f952c1838c23e1f4a
│     Current version:     9.6
│     Vendor:              Poly (USB:0x095D)
│     GUIDs:               3e77de51-8ee7-57d8-b4d2-23a57800b177
│                          d79fbe4d-18a7-5c2d-8da9-215dfa6cac00
│     Device Flags:        • Updatable
│                          • Is in bootloader mode

```

**Additional questions**
- Operating system and version: Ubuntu 20.04
- Have you tried rebooting? Yes
- Is this a regression? No

hughsie, 2020-08-29:
All fixed with master now?

zhj8407, 2020-08-29:
Yes. With the commit https://github.com/fwupd/fwupd/commit/783bc5c8a8d3f47206588c48f0f58113ef611b95. Thanks.

closed 2020-08-29 by hughsie