fwupd/fwupd #2345: List of all fwupd device flags: add "require_dc" and "not_updatable"

mendel5, 2020-08-30:
There is a list of all fwupd device flags at https://github.com/fwupd/fwupd/blob/master/libfwupd/fwupd-enums.h#L86 :
```
/**
 * FwupdDeviceFlags:
 * @FWUPD_DEVICE_FLAG_NONE:			No flags set
 * @FWUPD_DEVICE_FLAG_INTERNAL:			Device cannot be removed easily
 * @FWUPD_DEVICE_FLAG_UPDATABLE:		Device is updatable in this or any other mode
 * @FWUPD_DEVICE_FLAG_ONLY_OFFLINE:		Update can only be done from offline mode
 * @FWUPD_DEVICE_FLAG_REQUIRE_AC:		Requires AC power
 * @FWUPD_DEVICE_FLAG_LOCKED:			Is locked and can be unlocked
 * @FWUPD_DEVICE_FLAG_SUPPORTED:		Is found in current metadata
 * @FWUPD_DEVICE_FLAG_NEEDS_BOOTLOADER:		Requires a bootloader mode to be manually enabled by the user
 * @FWUPD_DEVICE_FLAG_REGISTERED:		Has been registered with other plugins
 * @FWUPD_DEVICE_FLAG_NEEDS_REBOOT:		Requires a reboot to apply firmware or to reload hardware
 * @FWUPD_DEVICE_FLAG_REPORTED:			Has been reported to a metadata server
 * @FWUPD_DEVICE_FLAG_NOTIFIED:			User has been notified
 * @FWUPD_DEVICE_FLAG_USE_RUNTIME_VERSION:	Always use the runtime version rather than the bootloader
 * @FWUPD_DEVICE_FLAG_INSTALL_PARENT_FIRST:	Install composite firmware on the parent before the child
 * @FWUPD_DEVICE_FLAG_IS_BOOTLOADER:		Is currently in bootloader mode
 * @FWUPD_DEVICE_FLAG_WAIT_FOR_REPLUG:		The hardware is waiting to be replugged
 * @FWUPD_DEVICE_FLAG_IGNORE_VALIDATION:	Ignore validation safety checks when flashing this device
 * @FWUPD_DEVICE_FLAG_TRUSTED:			Extra metadata can be exposed about this device
 * @FWUPD_DEVICE_FLAG_NEEDS_SHUTDOWN:		Requires system shutdown to apply firmware
 * @FWUPD_DEVICE_FLAG_ANOTHER_WRITE_REQUIRED:	Requires the update to be retried with a new plugin
 * @FWUPD_DEVICE_FLAG_NO_AUTO_INSTANCE_IDS:	Do not add instance IDs from the device baseclass
 * @FWUPD_DEVICE_FLAG_NEEDS_ACTIVATION:		Device update needs to be separately activated
 * @FWUPD_DEVICE_FLAG_HISTORICAL		Device is for historical data only
 * @FWUPD_DEVICE_FLAG_ENSURE_SEMVER:		Ensure the version is a valid semantic version, e.g. numbers separated with dots
 * @FWUPD_DEVICE_FLAG_ONLY_SUPPORTED:		Only devices supported in the metadata will be opened
 * @FWUPD_DEVICE_FLAG_WILL_DISAPPEAR:		Device will disappear after update and can't be verified
 * @FWUPD_DEVICE_FLAG_CAN_VERIFY:		Device checksums can be compared against metadata
 * @FWUPD_DEVICE_FLAG_CAN_VERIFY_IMAGE:		Image can be dumped from device for verification
 * @FWUPD_DEVICE_FLAG_DUAL_IMAGE:		Device update architecture uses A/B partitions for updates
 * @FWUPD_DEVICE_FLAG_SELF_RECOVERY:		In flashing mode device will only accept intended payloads
 * @FWUPD_DEVICE_FLAG_USABLE_DURING_UPDATE:	Device remains usable while fwupd flashes or schedules the update
 * @FWUPD_DEVICE_FLAG_VERSION_CHECK_REQUIRED:	All firmware updates for this device require a firmware version check
 * @FWUPD_DEVICE_FLAG_INSTALL_ALL_RELEASES:	Install each intermediate release rather than jumping direct to newest
 * @FWUPD_DEVICE_FLAG_MD_SET_NAME:		Set the device name from the metadata <name> if available
 * @FWUPD_DEVICE_FLAG_MD_SET_NAME_CATEGORY:	Set the device name from the metadata <category> if available
 * @FWUPD_DEVICE_FLAG_MD_SET_VERFMT:		Set the device version format from the metadata if available
 * @FWUPD_DEVICE_FLAG_ADD_COUNTERPART_GUIDS:	Add counterpart GUIDs from an alternate mode like bootloader
 * @FWUPD_DEVICE_FLAG_NO_GUID_MATCHING:		Force an explicit ID match when adding devices to the device list
 * @FWUPD_DEVICE_FLAG_UPDATABLE_HIDDEN:		Device is updatable but should not be called by the client
 * @FWUPD_DEVICE_FLAG_SKIPS_RESTART:		Device relies upon activation or power cycle to load firmware
 *
 * The device flags.
 **/
```

1. I was wondering why there is a `require_ac` device flag at https://github.com/fwupd/fwupd/blob/master/libfwupd/fwupd-enums.h#L91 but no `require_dc` device flag. Would it be possible to add the fwupd device flag `require_dc` to keep the consistency with the `require_ac` device flag?

2. When I asked a question about firmware updates to CPUs (see https://github.com/fwupd/fwupd/issues/2277 ), Mario answered: `CPUs don't contain an updatable flag. They're information only in fwupd.`
So currently one only knows that CPUs are not updatable because they are **missing** information (namely the `updatable` flag). It would be more obvious if there was a device flag that explicitly states that some devices are currently not updatable. Such a fwupd device flag could be named `not_updatable` or `information_only` or something similar. This flag would make it more obvious that some devices like CPUs cannot be updated.

superm1, 2020-08-30:
Flags are added when they first have a use. The require ac flag was added to the specific case of flashing system firmware while on a consistent power source. I don't believe we have a use case for a DC flag. Can you please advise what you think it makes sense for?

Regarding the information only aspect, I think this is a client detail. We can adjust fu-util-common.c to look for the lack of an updatable flag to output something like what you suggest. No need to add a new flag just to show that inverse.

mendel5, 2020-08-31:
> Can you please advise what you think it makes sense for?

I noticed that some devices have a `require_ac` flag but other devices have no electric current flag at all. So I was just wondering why there is something _missing_. I thought that if a device doesn't have a `require_ac` flag then it probably uses direct current and could therefore have a `require_dc` flag.

> No need to add a new flag just to show that inverse.

You are right, I think both cases can be solved by inversion.

1. If a device lacks a `require_ac` flag then show that it requires direct current (dc)

2. If a device lacks an `updatable` flag then show that is not updatable / information only.

Would that be ok?

superm1, 2020-08-31:
> I noticed that some devices have a require_ac flag but other devices have no electric current flag at all. So I was just wondering why there is something missing. I thought that if a device doesn't [...]

Well actually the inversion of `require_ac` isn't always true though.  Require AC is a safety flag.  I'm not aware of any devices that need to run updates on NOT AC.

mendel5, 2020-09-01:
> I'm not aware of any devices that need to run updates on NOT AC.

I think I found a device where that seems to be the case. The output of `$ fwupdmgr get-devices` for my Lenovo Thinkpad can be found at https://github.com/fwupd/fwupd/issues/2287 .

The Touchpad listed below has the `updatable` flag but no `requires_ac` flag:
```
│   
├─Touchpad:
│     Device ID:           91961e6ddc950eb21e71db6f33ec09a133933a95
│     Current version:     1.3.3013412
│     Bootloader Version:  8.0.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               10010313-27c0-5c7f-bd99-955028fd3a01
│                          71463ccd-7553-5d7d-9fac-7ed8073a7853
│                          981824ee-4283-595b-a2e4-46eae92dba40
│                          0957b08b-b580-5c3a-8073-98b9bb15bc7f
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Supported on remote server
│      
```

hughsie, 2020-09-01:
> The Touchpad listed below has the updatable flag but no require_ac flag:

It's indeed update-able on battery power.

mendel5, 2020-09-02:
> update-able on battery power

Wait, the `require_ac` flag means that a device needs to be plugged into a power source? Is it like a security mechanism to prevent running out of battery power during the update mechanism and then accidentally bricking the device?

hughsie, 2020-09-02:
> means that a device needs to be plugged into a power source

Exactly that.

mendel5, 2020-09-02:
> Exactly that.

Wow, I understood the `@FWUPD_DEVICE_FLAG_REQUIRE_AC` in a completely different way than what it actually means.

Let's have a look at this device from https://github.com/fwupd/fwupd/issues/2287 :
```
│
├─Thunderbolt Controller:
│     Device ID:           470906fa3412b62a24358dda0865908d5969c99b
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               c7920601-0cda-507f-851e-92229eb1d370
│                          96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
```
To me this clearly means: **The Thunderbolt controller is a device that runs on alternating current.**

This surprised me very much, as I thought all devices inside a computer are running on direct current (see e.g. https://electronics.stackexchange.com/questions/5949/is-it-possible-for-a-computer-to-use-ac-power ).

That's the reason why I thought a `require_dc` flag seems like a logical thing to have...

mendel5, 2020-09-03:
@hughsie Is this https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN2HTF.firmware the same Thunderbolt Controller as the one mentioned in the comment above?

mendel5, 2020-09-09:
@superm1 Please don't close this issue...

I think it would be beneficial if the `require_ac` flag gets partially changed to something that describes that the device needs to be plugged into a power source.

@hughsie Can you tell me if the Thunderbolt Controller linked above is the same one as mentioned in the comment?

superm1, 2020-09-09:
"Requires AC power"

What more do you think?

hughsie, 2020-09-10:
> Requires AC power

Maybe this a localisation issue; in the UK it's more common to say "on _mains_ power". The other thing we could clarify is _what_ needs AC power, e.g. "System requires AC power" or "Computer needs AC power"

mendel5, 2020-09-12:
Thanks for your answer!

How about `System requires external power source`?

hughsie, 2020-09-14:
> System requires external power source

To me, that makes it sound like you need to add something extra.

mendel5, 2020-09-14:
> that makes it sound like you need to add something extra

Yes, you need to plug in the power connector :smile:

mendel5, 2020-09-19:
Sorry for the delay. You can find my pull request at https://github.com/fwupd/fwupd/pull/2403 .

superm1, 2020-09-20:
#2403 was merged, closing this one

closed 2020-09-20 by superm1

mendel5, 2020-09-22:
@superm1 Thanks for accepting my pull request. Can you replace the `invalid` label on this issue with e.g. the `enhancement` or `question` label?