fwupd/fwupd #2346: Progress bar improperly display when using urxvt.

mkrasnitski, 2020-08-30:
**Describe the bug**
When fwupdmgr downloads anything in urxvt, the progress bar it uses doesn't display properly. 
[image]

I suspect this has to do with the way lines are overwritten. When erasing lines, a number of backspace characters is output equal to the length of the line, and somehow urxvt doesn't like this. However, it does handle commands like `printf "Test\b\b"` fine, which may imply the issue lies elsewhere. 
https://github.com/fwupd/fwupd/blob/395f936b72c6a933fdee4b8c2565c2d847fa9dd9/src/fu-progressbar.c#L98-L106

**Steps to Reproduce**
Run `fwupdmgr refresh --force` while using urxvt.

**Expected behavior**
The progress bar should work as expected. For example, the kitty terminal handles it fine:
[image]

**fwupd version information**
Please provide the version of the daemon and client.
```shell
 ❯ fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.5
	efivar:	37
daemon version:	1.4.5

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
 ❯ fwupdmgr get-devices --show-all-devices
20R1S0M100
│
├─Thunderbolt Controller:
│     Device ID:           74201269a6e0ceefcee7217c842491d41483b4f2
│     Summary:             Unmatched performance for high-speed I/O
│     Current version:     44.00
│     Vendor:              Lenovo (TBT:0x0109)
│     GUIDs:               74cdddba-f7b6-5a31-a3dc-64186b5a5491
│                          18bb00e3-5d7a-55d3-831e-0e000c184b0e
│                          c7920601-0cda-507f-851e-92229eb1d370
│                          96c6512b-0138-51da-b02f-f44f2c4c7744
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Device stages updates
│   
├─Embedded Controller:
│     Device ID:           d4db94d2324ae6986698e7489b234c0377fd8465
│     Current version:     0.1.9
│     Minimum Version:     0.1.9
│     Vendor:              DMI:LENOVO
│     Update State:        success
│     GUID:                ef793001-2e3e-42fa-ad94-c8d75c62909e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Intel(R) Core™ i5-10210U CPU @ 1.60GHz:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0xd6
│     Vendor:              GenuineIntel
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─System Firmware:
│     Device ID:           09647b198775f1868cc695ec2685477e0964c323
│     Current version:     0.1.16
│     Vendor:              LENOVO (DMI:LENOVO)
│     Update State:        success
│     GUID:                765f81e8-cc44-4d09-8fc4-9a47b167166e
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─TPM:
│ │   Device ID:           c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Current version:     73.64.17568.6659
│ │   Vendor:              ST Microelectronics (TPM:STM)
│ │   GUIDs:               ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                        84df3581-f896-54d2-bd1a-372602f04c32
│ │                        bfaed10a-bbc1-525b-a329-35da2f63e918
│ │                        70b7b833-7e1a-550a-a291-b94a12d0f319
│ │                        06f005e9-cb62-5d1a-82d9-13c534c53c48
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:      • Internal device
│     
├─Touchpad:
│     Device ID:           2bd025a21741529fac39d6f088b9debfee7d4c9a
│     Current version:     1.3.3013337
│     Bootloader Version:  8.0.0
│     Vendor:              Synaptics (HIDRAW:0x06CB)
│     GUIDs:               4150912a-f249-5f7e-bfa7-2d49e9c9de8c
│                          9c27eba7-52a4-5e6f-b28c-b5d9a3991e35
│                          19326b41-711b-5b64-89a0-6c86f3139bf1
│                          54ed5b90-c4f7-5551-a4cc-e6ac9957a4a8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Supported on remote server
│   
├─UEFI Device Firmware:
│     Device ID:           fe0a5afcd3063283e8535294cd48ad186d78707a
│     Current version:     3758818485
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                aa6f3567-b1f3-46d1-bf8a-a21f17f286aa
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           6c506442b8cac6d37736e5ae723b86b22271612c
│     Current version:     16777225
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                bb6c1645-0b81-4a03-a9b9-412059ce56e5
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           317fd9ef3f069f67047ab1163f8c040c3853cc38
│     Current version:     4784192
│     Minimum Version:     4784192
│     Vendor:              DMI:LENOVO
│     GUID:                1181d850-c3be-4b1d-985a-b28dca6ba6d8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UHD Graphics:
│     Device ID:           bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:     02
│     Vendor:              Intel Corporation (PCI:0x8086)
│     GUIDs:               9779d679-0f31-5dd5-b3c2-f6ff275a9ca9
│                          5589d313-88e5-54d3-8f4c-b2ee3882d4bc
│     Device Flags:        • Internal device
│                          • Cryptographic hash verification is available
│   
└─WDC PC SN730 SDBQNTY-256G-1001:
      Device ID:           3743975ad7f64f8d6575a9ae49fb3a8856fe186f
      Summary:             NVM Express Solid State Drive
      Current version:     11130101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                           1836f81c-3a3b-52b2-bb89-e5dc480ca9ec
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Additional questions**
- Operating system and version: Arch Linux 5.8.5

superm1, 2020-08-31:
This is presumably urxvt with zsh looking at that screenshot, right?

Maybe this is what's going on:
https://askubuntu.com/questions/54145/how-to-fix-strange-backspace-behaviour-with-urxvt-zsh

Basically try setting `TERM=xterm-256color` and see if that helps.

mkrasnitski, 2020-08-31:
Setting `TERM` to any variant of `xterm` doesn't fix the problem.

closed 2020-10-08 by stale[bot]

bubbleguuum, 2021-02-14:
Confirming this issue. This is annoying for those of us using urxvt.

hughsie, 2021-02-15:
None of us use or test on urxvt, sorry. If you want to fix this bug that's great, but we're not planning to work on it, sorry.

bubbleguuum, 2021-02-15:
Fair enough, So I clone the project to dig into this issue.

It turns out it is caused by single character …  found in strings in `fu_progressbar_status_to_string `

If you remove it (or replace it with three dots characters), it fixes the problem.

After displaying values of `self->to_erase` in `fu_progressbar_erase_line()`  with and without the single tree-dot character, it turns out that this character cause a +2 increase of `self->to_erase`. So we have 2 `g_print("\b")` calls for this single tree-dot character. On urxvt this results in 2 characters being removed (the single dot character and the previous one) while on other terminals it only removes the single dot character. Not sure why this difference of handling between terminals, it but must be something deep and obscure in the terminal code in the way they handle \b. It seems urxvt is doing the right thing.

This can be trivially fixed by not using single 3-dot characters, but instead 3 separate dot characters, in all of fu-progressbar.c

bubbleguuum, 2021-02-15:
A better fix, tested working with `xterm` and `urxvt` is the proper computation of `to_erase` in function `fu_progressbar_refresh()`:

`self->to_erase = g_utf8_strlen(str->str, -1);`

hughsie, 2021-02-15:
Space before bracket then pull request please! 🙂

bubbleguuum, 2021-02-15:
Done