fwupd/fwupd #2356: TPM PCR0 differs from reconstruction, Thinkpad Carbon X1 Gen8

amorsen, 2020-09-04:
**Describe the bug**
fwupd tells me to report a bug, so here we go...

**Steps to Reproduce**
[root@odysseus ~]# LANG=C fwupdtool get-updates
Loading?                 [-                                      ]09:11:54:0733 FuPluginTpmEventlog  TPM PCR0 differs from reconstruction, please see https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction
Loading?                 [ -                                     ]09:11:54:0898 FuPluginDfu          04f2:b6cb is missing download capability
Loading?                 [***************************************]
? Thunderbolt Controller has no available firmware updates
? Embedded Controller has the latest available firmware version
? Integrated Camera has no available firmware updates
? Intel Management Engine has the latest available firmware version
? Prometheus has the latest available firmware version
? Prometheus IOTA Config has the latest available firmware version
? SAMSUNG MZVLB512HBJQ-000L7 has no available firmware updates
? System Firmware has the latest available firmware version
? Touchpad has the latest available firmware version
? UEFI Device Firmware has no available firmware updates
? UEFI Device Firmware has no available firmware updates

**fwupd version information**

[root@odysseus ~]# rpm -q fwupd
fwupd-1.4.5-3.fc32.x86_64
[root@odysseus ~]# fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5

**fwupd device information**

Will attach along with fwupdtpmevlog.

**Additional questions**
- Operating system and version:

Fedora 32

- Have you tried rebooting?
Yes

- Is this a regression?

I do not know

[fwupdmgr.txt](https://github.com/fwupd/fwupd/files/5173976/fwupdmgr.txt)
[fwupdtpmevlog.txt](https://github.com/fwupd/fwupd/files/5173977/fwupdtpmevlog.txt)

hughsie, 2020-09-04:
@superm1 is this expected?

    PCR:                     BIOS (0)
    Type:                    0x3
    Description:             EV_NO_ACTION
    ChecksumSha1:            0000000000000000000000000000000000000000
    ChecksumSha256:          0000000000000000000000000000000000000000000000000000000000000000
    BlobStr:                 StartupLocality..

I wonder if we ignore that the PCR0 would reconstruct okay?

superm1, 2020-09-04:
I'm pretty sure that's part of ACM start-up, but someone who is familiar with that BIOS would need to confirm it.