fwupd/fwupd #2360: fwupdtool refresh : failed to verify data, expected abe4258d1dbbc02ad732dc1feefeb266ae872e6c

joakim-tjernlund, 2020-09-08:
**Describe the bug**
fwupdtool refresh
Loading…                 [-                                      ]ERROR:esys:src/tss2-esys/esys_context.c:69:Esys_Initialize() Initialize default tcti. ErrorCode (0x000a000a) 
Loading…                 [***************************************]
checksum failure: failed to verify data, expected abe4258d1dbbc02ad732dc1feefeb266ae872e6c

**Steps to Reproduce**
See above 

**fwupd version information**
master from today

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
emerge sys-apps/fwupd

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

20L7004MMX
│
├─Thunderbolt host controller:
│     Device ID:          ab7d5105abea00e669b468df7882afc0b34e133e
│     Summary:            Unmatched performance for high-speed I/O
│     Current version:    20.00
│     Vendor:             Lenovo (TBT:0x0109)
│     GUIDs:              3849839e-d8b2-5941-9e69-5acc397d3ed3 ← THUNDERBOLT\VEN_0109&DEV_1708&REV_00
│                         7e3190a2-d5a5-5a9e-b582-494b47f7288c ← THUNDERBOLT\VEN_0109&DEV_1708
│                         061f2847-061b-5cfb-9c39-3dbfe2b49d69 ← TBT-01091708-native
│                         65a3d897-4a18-5d27-83d8-eb8a231b7aa4 ← TBT-01091708-native-controller0-0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Requires AC power
│                         • Supported on remote server
│                         • Device stages updates
│   
├─USB3.1 Hub:
│     Device ID:          d68b5d74453efa336214e42d39ec96008a3fed64
│     Summary:            USB 3.x Hub
│     Current version:    10.74
│     Vendor:             VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:   15 seconds
│     GUIDs:              e2c96871-8be3-530e-a796-5652cee64eeb
│                         351195ca-bd3e-5a03-911c-0be6686d91dd ← USB\VID_17EF&PID_3070&REV_0A74
│                         2c08e9d9-1f25-5564-b4db-49867dd2c593 ← USB\VID_17EF&PID_3070
│                         816fc838-78d7-5deb-85ef-a95935318198 ← USB\VID_17EF&PID_3070&HUB_0010
│                         7705d2da-41a0-57db-9fb3-c831b76638ad ← USB\VID_17EF&PID_3070&DEV_VL820Q7
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─USB3.1 Hub:
│     Device ID:          da77984c82b59c6fc69516431f467fd9a8d39a7f
│     Summary:            USB 3.x Hub
│     Current version:    10.73
│     Vendor:             VIA Labs, Inc. (USB:0x17EF)
│     Install Duration:   15 seconds
│     GUIDs:              e2c96871-8be3-530e-a796-5652cee64eeb
│                         f54dad5a-2e3d-5139-a478-2be7c915a33c ← USB\VID_17EF&PID_3070&REV_0A73
│                         2c08e9d9-1f25-5564-b4db-49867dd2c593 ← USB\VID_17EF&PID_3070
│                         6d51c3aa-707f-5d41-812d-5cb5c8a4a25d ← USB\VID_17EF&PID_3070&HUB_0004
│                         b8540e4f-9b8c-5fdf-afff-b9352bd88b10 ← USB\VID_17EF&PID_3070&DEV_VL820Q8
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
├─Embedded Controller:
│     Device ID:          323bda90831520867f8c02544efb9711dfca9fb6
│     Current version:    0.1.20
[... 113 lines cut ...]
│                         9af0af2d-99ff-551b-9134-e94809cc8016 ← USB\VID_17EF&PID_3071&DEV_VL820Q7
│     Device Flags:       • Updatable
│                         • Cryptographic hash verification is available
│                         • Device stages updates
│                         • Device can recover flash failures
│   
└─USB2.0 Hub:
      Device ID:          983c3cffc6fd36d32b00b62928d30721eaeb93db
      Summary:            USB 2.x Hub
      Current version:    10.73
      Vendor:             VIA Labs, Inc. (USB:0x17EF)
      Install Duration:   15 seconds
      GUIDs:              fa8ac445-2c73-5bd7-99ab-6cfd94d1e40c ← USB\VID_17EF&PID_3071&REV_0A73
                          e2c96871-8be3-530e-a796-5652cee64eeb ← USB\VID_17EF&PID_3071
                          6f3f32bd-ac98-523e-9128-ef67941fb614 ← USB\VID_17EF&PID_3071&HUB_20
                          193e6679-e01f-5afc-a4b9-bcf31e14f09f ← USB\VID_17EF&PID_3071&DEV_VL820Q8
      Device Flags:       • Updatable
                          • Cryptographic hash verification is available
                          • Device stages updates
                          • Device can recover flash failures

**Additional questions**
- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

hughsie, 2020-09-08:
Is lvfs-testing enabled for you?

joakim-tjernlund, 2020-09-08:
> Is lvfs-testing enabled for you?

Yes

hughsie, 2020-09-08:
I've regenerated lvfs-testing now manually, which fixes things for me. I've started some root cause debugging.

joakim-tjernlund, 2020-09-08:
> I've regenerated lvfs-testing now manually, which fixes things for me. I've started some root cause debugging.

Thanks.
Is the "Initialize default tcti. ErrorCode (0x000a000a)" error part of the problem ?
If not, why am I seeing that error?

superm1, 2020-09-08:
>Is the "Initialize default tcti. ErrorCode (0x000a000a)" error part of the problem ?
If not, why am I seeing that error?

Completely unrelated to this issue.  It seems that your TPM or your TPM stack are not working is that mesage.

joakim-tjernlund, 2020-09-08:
> > Is the "Initialize default tcti. ErrorCode (0x000a000a)" error part of the problem ?
> > If not, why am I seeing that error?
> 
> Completely unrelated to this issue. It seems that your TPM or your TPM stack are not working is that mesage.

I don't use TPM, fwupd does pull in tpm2-tss though

superm1, 2020-09-08:
>I don't use TPM

So that's why it couldn't initialize the TCTI then, seems without a TPM to communicate with the TCTI wouldn't start up.  Assuming I decoded it right, it decodes as "TSS2_TCTI_RC_IO_ERROR"
https://github.com/tpm2-software/tpm2-tss/blob/e9085c78d68b85803a2274a75369bda15aa0ebc8/include/tss2/tss2_common.h#L153

joakim-tjernlund, 2020-09-08:
> So that's why it couldn't initialize the TCTI then, seems without a TPM to communicate with the TCTI wouldn't start up. Assuming I decoded it right, it decodes as "TSS2_TCTI_RC_IO_ERROR" [...]

Yes, does fwupd need to throw an error msg here? The error msg makes me this something is very wrong

superm1, 2020-09-08:
Right now we suppress warnings from the TCTI, but not errrors.
https://github.com/fwupd/fwupd/blob/0d4e2ca4c54afe6da33a9663a4741236d0ff12fb/plugins/uefi/fu-uefi-pcrs.c#L120

If you're not using a TPM, you can just compile without TSS2 support.  It's not a "required" dependency.
https://github.com/fwupd/fwupd/blob/3e82beeddac31292c50229e59e2404865edee5ad/meson.build#L229

joakim-tjernlund, 2020-09-08:
> Right now we suppress warnings from the TCTI, but not errrors.
>
> https://github.com/fwupd/fwupd/blob/0d4e2ca4c54afe6da33a9663a4741236d0ff12fb/plugins/uefi/fu-uefi-pcrs.c#L120
>
> If you're not using a [...]

But then I could not use the Gentoo standard pkg ...

superm1, 2020-09-08:
Perhaps it makes sense in Gentoo to propose a USE flag specifically to `tpm`?

joakim-tjernlund, 2020-09-08:
> Perhaps it makes sense in Gentoo to propose a USE flag specifically to `tpm`?

maybe, but what about other, binary, dists that has no such USE flag feature?
Or is it possible that I am missiing the /dev/tpm* I/Fs ? should I have them even thoug I don't care about TPM?

superm1, 2020-09-08:
>maybe, but what about other, binary, dists that has no such USE flag feature?

I don't know anywhere else that `/dev/tpm*` isn't exporting unless someone chooses to not compile kernel with TPM support.

>Or is it possible that I am missiing the /dev/tpm* I/Fs ? should I have them even thoug I don't care about TPM?

You should be having `/dev/tpm*` nodes unless your device/container really doesn't have a physical or firmware TPM.

joakim-tjernlund, 2020-09-08:
```
dmesg | grep -i tpm
[    0.000000] efi:  TPMFinalLog=0x6ff92000  SMBIOS=0x6f02b000  SMBIOS 3.0=0x6f028000  ACPI=0x6fffe000  ACPI 2.0=0x6fffe014  MPS=0x6f448000  ESRT=0x6eeb5000  MEMATTR=0x69a09018 
[    0.014062] ACPI: SSDT 0x000000006FFE8000 0003DB (v02 LENOVO Tpm2Tabl 00001000 INTL 20160527)
[    0.014065] ACPI: TPM2 0x000000006FFE7000 000034 (v03 LENOVO TP-N22   00001420 PTEC 00000002)
se-jocke3-lx ~ # zcat /proc/config.gz | grep TPM
CONFIG_TCG_TPM=m
CONFIG_HW_RANDOM_TPM=y
# CONFIG_TCG_VTPM_PROXY is not set
```
What does that look like for you ?

superm1, 2020-09-08:
Here's what I've got in my kernel:

```
CONFIG_TCG_TPM=y
CONFIG_TCG_TIS_CORE=y
CONFIG_TCG_TIS=y
CONFIG_TCG_TIS_SPI=m
# CONFIG_TCG_TIS_SPI_CR50 is not set
CONFIG_TCG_TIS_I2C_ATMEL=m
CONFIG_TCG_TIS_I2C_INFINEON=m
CONFIG_TCG_TIS_I2C_NUVOTON=m
CONFIG_TCG_NSC=m
CONFIG_TCG_ATMEL=m
CONFIG_TCG_INFINEON=m
CONFIG_TCG_XEN=m
CONFIG_TCG_CRB=y
CONFIG_TCG_VTPM_PROXY=m
CONFIG_TCG_TIS_ST33ZP24=m
CONFIG_TCG_TIS_ST33ZP24_I2C=m
CONFIG_TCG_TIS_ST33ZP24_SPI=m
```

joakim-tjernlund, 2020-09-08:
Just saw this, there is an tpm USE flag that isn't set:
```
Installed versions:  9999*l^t{tbz2}[1](15:21:17 08/09/20)(elogind introspection man nvme thunderbolt uefi -agent -amt -consolekit -dell -gtk-doc -minimal -redfish -synaptics -systemd -test -tpm 

Seems like fwupd picks up tpm because I have it installed?

joakim-tjernlund, 2020-09-08:
UEFI pulls in tpm2-tss so I cannot avoid that,
and I have  $(meson_use tpm plugin_tpm)
meaning that USE=tpm must be set for fwupd ebuild to select tpm_plugin

superm1, 2020-09-08:
So there is tpm functionality in the UEFI plugin, but also in the tpm plugin.  It's not a requirement for UEFI plugin, but optional. 
 Looking over the ebuild turning off tpm use flag just doesn't enable tpm plugin.

joakim-tjernlund, 2020-09-08:
hmm, not sure what you mean. Should I rm dep on tpm2-tss in the ebuild for UEFI?
That would still pull in tpm2-tss into fwup if tpm2-tss happens to be installed.

BTW, noted this meson warning:
```
WARNING: Project specifies a minimum meson_version '>=0.47.0' but uses features which were added in newer versions:
 * 0.50.0: {'install arg in configure_file'}

superm1, 2020-09-08:
>hmm, not sure what you mean. Should I rm dep on tpm2-tss in the ebuild for UEFI?

That's something to discuss with the maintainer, sure.

>That would still pull in tpm2-tss into fwup if tpm2-tss happens to be installed.

Yes

But I think you still have a problem with your system and TPMs.
do you have `/dev/tpm0`?

>BTW, noted this meson warning:

Intentional right now.

joakim-tjernlund, 2020-09-08:
> > hmm, not sure what you mean. Should I rm dep on tpm2-tss in the ebuild for UEFI?
> 
> That's something to discuss with the maintainer, sure.

Will suggest that, the error goes away if I uninstall tpm2-tss anyway.
But why does UEFI have an optional dep on tpm2-tss when there is an tpm plugin
> 
> > That would still pull in tpm2-tss into fwup if tpm2-tss happens to be installed.
> 
> Yes

automatic deps like above is not something Gentoo likes. fwupd would break if tpm2-tss were removed
without warning.

> 
> But I think you still have a problem with your system and TPMs.
> do you have `/dev/tpm0`?

No, I do not

> 
> > BTW, noted this meson warning:
> 
> Intentional right now.

OK

superm1, 2020-09-08:
> automatic deps like above is not something Gentoo likes. fwupd would break if tpm2-tss were removed

Submitted https://github.com/fwupd/fwupd/pull/2362 for this to make it more like how you want it to work.

I guess with no other changes on your side, the TPM error will stop since `USE=-tpm`

>No, I do not

Do you have the TPM turned off in firmware setup?  Or are you missing some of those other settings I mentioned in my kernel config grep above?

joakim-tjernlund, 2020-09-08:
> Submitted #2362 for this to make it more like how you want it to work.
>
> I guess with no other changes on your side, the TPM error will stop since `USE=-tpm`
>
>
> Do you have the TPM turned off in [...]

home ATM, but I think both.

joakim-tjernlund, 2020-09-08:
> > automatic deps like above is not something Gentoo likes. fwupd would break if tpm2-tss were removed
> 
> Submitted #2362 for this to make it more like how you want it to work.
> 
> I guess with no other changes on your side, the TPM error will stop since `USE=-tpm`

For me, yes, but consider binary dists too. Would be nice for them to not throw an error like it does now.
Possibly a debug msg could do?

superm1, 2020-09-08:
> For me, yes, but consider binary dists too. Would be nice for them to not throw an error like it does now.

If it's compiled with TPM and TPM isn't working we shouldn't be hiding it I feel.  @hughsie thoughts?

closed 2020-09-08 by superm1

hughsie, 2020-09-08:
I guess I'd rather like to know why people don't want to use the TPM. If the TPM isn't working then yes, we need to tell the user.

joakim-tjernlund, 2020-09-09:
> I guess I'd rather like to know why people don't want to use the TPM. If the TPM isn't working then yes, we need to tell the user.

In this case there is more like "No TPM device present on the computer"

joakim-tjernlund, 2020-09-09:
@hughsie , any idea about the check sum failure?
Should I just delete some file and have the checksum regenerated and be done?

hughsie, 2020-09-09:
This is the root cause: https://github.com/fwupd/lvfs-website/pull/778

TL;DR: two workers were processing the metadata regeneration at the same time, both with the same `build_cnt` -- which meant we'd race on both the file write and the CDN purge.

hughsie, 2020-09-09:
Okay, I've deployed this now; can you retry please.

joakim-tjernlund, 2020-10-07:
See like this bug is back, running current master:
```
....
09:15:11:0067 FuEngine             ignoring 0.1.37 < 0.1.42
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string '6d1cf038-74ab-49b5-bb5c-6a1e99f3aea4' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string '14503b3d-73ce-5d06-8137-77c68972a341' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string '5971a208-da00-5fce-b5f5-1234342f9cf7' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string 'c6682ade-b5ec-57c4-b687-676351208742' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string '112b3acf-d42a-53cc-91c8-458c3b58f359' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string 'e2c96871-8be3-530e-a796-5652cee64eeb' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string '3545c1f3-4d81-59f7-b9d9-d25d5811e4be' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string 'fa8ac445-2c73-5bd7-99ab-6cfd94d1e40c' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string 'e2c96871-8be3-530e-a796-5652cee64eeb' was unfound
09:15:11:0067 XbSilo               ignoring for OR statement: indexed string '6f3f32bd-ac98-523e-9128-ef67941fb614' was unfound
09:15:11:0077 FuCommon             running 'wget https://cdn.fwupd.org/downloads/firmware-testing.xml.gz -O /root/.cache/fwupdmgr/firmware-testing.xml.gz'
09:15:11:0077 GLib                 posix_spawn avoided (fd close requested) (child_setup specified) 
09:15:11:0116 FuCommon             reading /root/.cache/fwupdmgr/firmware-testing.xml.gz with 44773 bytes
09:15:11:0116 FuCommon             running 'wget https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.jcat -O /root/.cache/fwupdmgr/firmware-testing.xml.gz.jcat'
09:15:11:0116 GLib                 posix_spawn avoided (fd close requested) (child_setup specified) 
09:15:11:0152 FuCommon             reading /root/.cache/fwupdmgr/firmware-testing.xml.gz.jcat with 2236 bytes
09:15:11:0152 FuMain               updating lvfs-testing
checksum failure: failed to verify data, expected 0cbcf8ba6cf9004a8bcafb7e95406dc8ab95501c
09:15:11:0156 FuPlugin             destroy(linux_swap)
09:15:11:0157 FuPlugin             destroy(upower)
09:15:11:0158 FuPlugin             destroy(linux_lockdown)
09:15:11:0159 FuPlugin             destroy(platform_integrity)
09:15:11:0159 FuPlugin             destroy(uefi)
09:15:11:0161 FuPlugin             destroy(linux_tainted)
09:15:11:0161 FuPlugin             destroy(logind)
```

hughsie, 2020-10-07:
@joakim-tjernlund good catch! Mea culpa.

commit 4e0a62ce (2020-09-08): Make TPM more optional (Fixes: #2360)

- Rename the `plugin_tpm` option to `tpm` and when it's disabled
remove TPM support from all plugins
- If enabled then require the TSS to be installed
commit e6378246 (2020-09-08): Make TPM more optional (Fixes: #2360)

- Rename the `plugin_tpm` option to `tpm` and when it's disabled
remove TPM support from all plugins
- If enabled then require the TSS to be installed
commit cffa31a6 (2020-09-08): Make TPM more optional (Fixes: #2360)

- Rename the `plugin_tpm` option to `tpm` and when it's disabled
remove TPM support from all plugins
- If enabled then require the TSS to be installed
commit 1ed29147 (2020-09-08): Make TPM more optional (Fixes: #2360)

- Rename the `plugin_tpm` option to `tpm` and when it's disabled
remove TPM support from all plugins
- If enabled then require the TSS to be installed