fwupd/fwupd #2363: Rename dbxtool

FFY00, 2020-09-08:
**Is your feature request related to a problem? Please describe.**
It looks like the latest fwupd version installs a dbxtool binary, which conflicts with https://github.com/rhboot/dbxtool.

**Describe the solution you'd like**
Rename the executable. A suggestion would be `fwupd-dbxtool`.

**Describe alternatives you've considered**
I can rename it manually in the downstream package, but I think it would be better solved in the upstream.

hughsie, 2020-09-08:
It's deliberately a replacement of rhboot/dbxtool -- with the blessing of Peter the original maintainer. In Fedora, fwupd now obsoletes dbxtool.

closed 2020-09-08 by hughsie

FFY00, 2020-09-08:
Thanks, I will adjust the arch package.

@vathpela would be possible to add this information in the project README?