fwupd/fwupd #2364: vli: Hyper Hub+PD Update

memily, 2020-09-09:
**Describe the bug**
After adding a hyper quirk file we can update the parent VL817 hub; but are unable to update the BB/VL103 PD child. 

**Steps to Reproduce**
Tried fwupdtool install command on corresponding hyper *.cab file. 

**Expected behavior**
Same setup and update protocol as current Lenovo TR Gen2 Hub.

**fwupd version information**
1.4.1, (dnf)

**fwupd device information**
Attached install verbose log, red error twiced on:

"ignoring device eed3f7906b7b04db3244b7233989fd12f141d2fc [vli] existing device c843d6b450bf0a70c9b3e794904887ee165c52da [vli] already exists"

[verbose.txt](https://github.com/fwupd/fwupd/files/5194173/verbose.txt)

hughsie, 2020-09-09:
> ignoring device eed3f7906b7b04db3244b7233989fd12f141d2fc [vli] existing device c843d6b450bf0a70c9b3e794904887ee165c52da [vli] already exists

So we have a

* USB HUB `USB\VID_2D01&PID_0385` with that gets the device_id of `b0c12d93b90105a3a7d0860fbecfb9fca76e0fc8` which has a VL103 child of `USB\VID_2D01&PID_0380` which has device_id `c843d6b450bf0a70c9b3e794904887ee165c52da`
* USB HUB `USB\VID_2D01&PID_0382` with that gets the device_id of `4295296d98b3ba38c72f6baa33d24f03a1d428f6` which has a VL103 child of `USB\VID_2D01&PID_0380` which has device_id `eed3f7906b7b04db3244b7233989fd12f141d2fc`

Are there really two USB hubs on this one physical device?

hughsie, 2020-09-09:
Either way I think something like this is a good idea:
[usbhub-pd.txt](https://github.com/fwupd/fwupd/files/5194749/usbhub-pd.txt)

closed 2020-09-09 by hughsie