fwupd/fwupd #2370: Support admin password protected bios updates

SEJeff, 2020-09-09:
**Is your feature request related to a problem? Please describe.**
As someone who manages a fleet of linux desktops and servers, I'd like to update the bios regularly just like we do with our windows desktops and servers.  Our desktops all have admin bios passwords set to prevent someone from mucking around with the boot order or disabling things like secure boot. The users do not have the admin bios passwords for their desktops.

**Describe the solution you'd like**
When updating the bios, the admin password needs to be entered to allow the update to happen. We would like a way to pass the admin password via fwupd to update the bios. Ideally, not via plaintext on disk or in ps output.

As an example, dell's firmware / bios update tool includes this command line executable on windows to do something like:
```
dcu-cli.exe /configure -biosPassword="Test1234" # The password can be passed via an env var
```

**Describe alternatives you've considered**
The only alternative is to share the admin bios password with all users and ask them to type it in to update the bios on their desktops. This is not a reasonable way to manage things in a fleet.

**Additional context**
[Dell Command Update docs](https://www.dell.com/support/manuals/us/en/04/command-update-v3.1/dellcommandupdate_3.1_ug/command-line-interface-reference?guid=guid-92619086-5f7c-4a05-bce2-0d560c15e8ed&lang=en-us). The `-encryptedPassword` and `-encryptedPasswordFile` flags seem super useful. If we could generate an encrypted password, then copy that out to the fleet as root:root 600 and then use that to update the firmware, it would be amazing.

superm1, 2020-09-09:
For Dell the desktop/laptop and server firmware upgrade implementations are different.  fwupd support is (currently) only available on the desktop/laptops.

Can you please clarify the following:
1. Is this request for desktop/laptop indeed?
2. What happens when you run an upgrade from fwupd when admin password is present today?
3. What happens when you run a BIOS upgrade from Windows update (if possible)?

hughsie, 2020-09-09:
@superm1?

SEJeff, 2020-09-09:
@superm1 sure here you go.

1. Our use case for this request is only for desktops/laptops
2. We don't currently run it. We'd like to change that. I should be able to reproduce it and give this a shot in the next week or so.
3. We use [Microsoft Endpoint Configuration Manager](https://www.microsoft.com/en-us/microsoft-365/microsoft-endpoint-manager) to control what and when updates apply. We have a task that runs the bios update with the password flag. We are not using upstream windows update directly as a result. It is 100% hands off for the user. They don't type in any password.

superm1, 2020-09-09:
OK thanks for confirming it's for desktops/laptops.  My expectation is that admin password is not needed when doing a UEFI capsule update (which is what fwupd and WU both do).  Running BIOS upgrades "within" Windows is a different story for how it works.

I checked on my side using the following:

0. Kernel with this patch in place: https://patchwork.kernel.org/patch/11757461/
1. Check admin password is off:
```
$ cat /sys/devices/platform/dell-wmi-sysman/attributes/Admin/is_password_set
0
```
2. Enable admin password:
`$ echo "dell1234" | sudo tee /sys/devices/platform/dell-wmi-sysman/attributes/Admin/new_password`
3. Verify it's set. 
```
$ cat /sys/devices/platform/dell-wmi-sysman/attributes/Admin/is_password_set
1
```
4. Reinstall current firmware
`$ fwupdmgr reinstall`
5. Answer yes to reboot system
6. See Dell's BIOS upgrade process take over again.

This is on and XPS 7390, with secure boot turned off.  If you also use XPS 7390 and do it with secure boot turned on, there is a bug in Dell's firmware that is [not yet fixed](https://github.com/fwupd/firmware-dell/issues/18).  I'm not aware that problem on any other machines right now (but if I'm wrong, please mention on that bug).

SEJeff, 2020-09-09:
@superm1 That's really great! I'm not sure if that is going to work for us however. Most of the clients are RHEL7 / RHEL8 with a 4.10/4.18.x kernel. My test box for this is a Precision 5820 Tower if that matters.

Do you happen to know if there is anything that actually requires the bios password? I guess it is outside of the scope for fwupd, but we'd like to standardize and manage the settings as well. I'd assume that would require the bios password as it does if you boot into it manually. With servers, you can do really all of this with catalog updates + omsa / racadm / redfish.

superm1, 2020-09-09:
>@superm1 That's really great! I'm not sure if that is going to work for us however. Most of the clients are RHEL7 / RHEL8 with a 4.10/4.18.x kernel. My test box for this is a Precision 5820 Tower if that matters.

The patch I linked is just one way to enable/disable a BIOS admin password.  You can certainly use the Dell command suite or libsmbios with RHEL.  Aspirationally once it's merged upstream I think it will displace all those other tools though in favor of simple shell scripts.

>Do you happen to know if there is anything that actually requires the bios password? I guess it is outside of the scope for fwupd, but we'd like to standardize and manage the settings as well. I'd assume that would require the bios password as it does if you boot into it manually. 

So there are a few different types of passwords used by Dell:

* Admin password: This is what you need to change items in BIOS setup.  It's what you were asking about.  You can optionally lock it down so you need it to even _view_ items in BIOS setup too.
* System password: This is required every time the system turns on
* HDD password: Unlock something supported by the Opal specification

SEJeff, 2020-09-09:
Yes, this is in reference to the Admin password. We are ok with read access, but would rather users not change things we set.

I also noticed that the bios in the lvmfs remote is 0.1.13.0 for a Precision 5820, but the latest is 2.4.0 per support.dell.com. Is there a better repository to use, or is lvmfs (remote: https://cdn.fwupd.org/downloads/firmware.xml.gz) the best one?

Thanks again for your time, I really appreciate it.

superm1, 2020-09-09:
@SEJeff The 2.x.y firmware was on LVFS for 5820 but appears to have been removed recently.  Let me ask why that happened, I expect it should be sorted out by your trial next week.

superm1, 2020-09-10:
@SEJeff I got word it's been intentionally pulled because of some failures.  You can find a lot of reports about failures on Dell's forums:
https://www.dell.com/community/Precision-Fixed-Workstations/5820-bios-update-2-0-2-Jan-20-stuck-at-0/td-p/7467683

I'm checking if we have something public explaining the details for the failures and what you can do about them.  For now to enable your test with admin password though I wanted to check - what kind of monitor (resolution) would you have connected to your various systems you wanted to test?  It may impact what we can do.

SEJeff, 2020-09-10:
I’ve got one of the Dell 49” ultrawides hooked up to it, so probably not the best for testing this. I believe the native resolution is 5120x1440

superm1, 2020-09-10:
Would it be OK to do the test for the lack of admin password with secure boot turned off and legacy option ROM turned on?  I think that may be able to work around the issue.

SEJeff, 2020-09-10:
Most all of our older builds were legacy boot mode, but all newish ones in the past year or so are uefi. We can’t use secure boot due to development of custom kernel modules so that’s fine.

I’ll be able to test that hopefully soon.

hughsie, 2020-10-22:
Hey @SEJeff I wondered what you wanted to do with this issue?

SEJeff, 2020-11-22:
I guess we can close it since it is a noop. Thanks!

closed 2020-11-22 by SEJeff