fwupd/fwupd #2373: Thunderbolt plugin fails to read NVM version

djcampello, 2020-09-09:
During fwuptool get-updates I see this output on verbose invocation. Each attempt takes .2 seconds delaying the command.

```
23:30:19:0882 FuPluginThunderbolt  Attempt 0: Failed to read NVM version
23:30:20:0082 FuPluginThunderbolt  Attempt 1: Failed to read NVM version
23:30:20:0282 FuPluginThunderbolt  Attempt 2: Failed to read NVM version
23:30:20:0483 FuPluginThunderbolt  Attempt 3: Failed to read NVM version
23:30:20:0683 FuPluginThunderbolt  Attempt 4: Failed to read NVM version
23:30:20:0883 FuPluginThunderbolt  Attempt 5: Failed to read NVM version
23:30:21:0083 FuPluginThunderbolt  Attempt 6: Failed to read NVM version
23:30:21:0283 FuPluginThunderbolt  Attempt 7: Failed to read NVM version
23:30:21:0483 FuPluginThunderbolt  Attempt 8: Failed to read NVM version
23:30:21:0683 FuPluginThunderbolt  Attempt 9: Failed to read NVM version
23:30:21:0884 FuPluginThunderbolt  Attempt 10: Failed to read NVM version
23:30:22:0084 FuPluginThunderbolt  Attempt 11: Failed to read NVM version
23:30:22:0284 FuPluginThunderbolt  Attempt 12: Failed to read NVM version
23:30:22:0484 FuPluginThunderbolt  Attempt 13: Failed to read NVM version
23:30:22:0684 FuPluginThunderbolt  Attempt 14: Failed to read NVM version
23:30:22:0884 FuPluginThunderbolt  Attempt 15: Failed to read NVM version
23:30:23:0084 FuPluginThunderbolt  Attempt 16: Failed to read NVM version
23:30:23:0284 FuPluginThunderbolt  Attempt 17: Failed to read NVM version
23:30:23:0485 FuPluginThunderbolt  Attempt 18: Failed to read NVM version
23:30:23:0685 FuPluginThunderbolt  Attempt 19: Failed to read NVM version
23:30:23:0885 FuPluginThunderbolt  Attempt 20: Failed to read NVM version
23:30:24:0085 FuPluginThunderbolt  Attempt 21: Failed to read NVM version
23:30:24:0285 FuPluginThunderbolt  Attempt 22: Failed to read NVM version
23:30:24:0485 FuPluginThunderbolt  Attempt 23: Failed to read NVM version
23:30:24:0686 FuPluginThunderbolt  Attempt 24: Failed to read NVM version
23:30:24:0886 FuPluginThunderbolt  Attempt 25: Failed to read NVM version
23:30:25:0086 FuPluginThunderbolt  Attempt 26: Failed to read NVM version
23:30:25:0286 FuPluginThunderbolt  Attempt 27: Failed to read NVM version
23:30:25:0486 FuPluginThunderbolt  Attempt 28: Failed to read NVM version
23:30:25:0686 FuPluginThunderbolt  Attempt 29: Failed to read NVM version
23:30:25:0887 FuPluginThunderbolt  Attempt 30: Failed to read NVM version
23:30:26:0087 FuPluginThunderbolt  Attempt 31: Failed to read NVM version
23:30:26:0287 FuPluginThunderbolt  Attempt 32: Failed to read NVM version
23:30:26:0487 FuPluginThunderbolt  Attempt 33: Failed to read NVM version
23:30:26:0687 FuPluginThunderbolt  Attempt 34: Failed to read NVM version
23:30:26:0887 FuPluginThunderbolt  Attempt 35: Failed to read NVM version
23:30:27:0087 FuPluginThunderbolt  Attempt 36: Failed to read NVM version
23:30:27:0288 FuPluginThunderbolt  Attempt 37: Failed to read NVM version
23:30:27:0488 FuPluginThunderbolt  Attempt 38: Failed to read NVM version
23:30:27:0688 FuPluginThunderbolt  Attempt 39: Failed to read NVM version
23:30:27:0888 FuPluginThunderbolt  Attempt 40: Failed to read NVM version
23:30:28:0088 FuPluginThunderbolt  Attempt 41: Failed to read NVM version
23:30:28:0288 FuPluginThunderbolt  Attempt 42: Failed to read NVM version
23:30:28:0489 FuPluginThunderbolt  Attempt 43: Failed to read NVM version
23:30:28:0689 FuPluginThunderbolt  Attempt 44: Failed to read NVM version
23:30:28:0889 FuPluginThunderbolt  Attempt 45: Failed to read NVM version
23:30:29:0089 FuPluginThunderbolt  Attempt 46: Failed to read NVM version
23:30:29:0289 FuPluginThunderbolt  Attempt 47: Failed to read NVM version
23:30:29:0489 FuPluginThunderbolt  Attempt 48: Failed to read NVM version
23:30:29:0689 FuPluginThunderbolt  Attempt 49: Failed to read NVM version
23:30:29:0890 FuPluginThunderbolt  failed to read version
23:30:29:0890 FuPluginThunderbolt  failed to get Vendor ID
23:30:29:0890 FuPluginThunderbolt  failed to get Device ID
```
My guess is that the plugin is attempting to read nvm_version vendor and device sysfs files for device 0-0

```
localhost /sys/bus/thunderbolt/devices/0-0 # ls
0-0:3.1  0-3  authorized  generation  power  subsystem  uevent  unique_id  wakeup
```

djcampello, 2020-09-09:
@superm1 would you know if the plugin attempts to scan for all devices under /sys/bus/thunderbolt ?

superm1, 2020-09-09:
I guess that's a USB4 device running in SW CM mode, so no nvm_version exports. I think we need to look for this case and not build a fwupd device lost likely. There's no point to fwupd exporting it if it can't tell you a version for it or something useful about system security (in 1.5.x).

In probe I would suggest to directly look for that file being present and not letting it probe successfully if not present.

commit 65b71303 (2020-09-10): thunderbolt: ignore non-updatable host controller (Fixes: #2373)

nvm_version will not export in Intel USB4 host controller running in
SW CM, there is no point in fwupd displaying a device for it.
commit 64f38577 (2020-09-10): thunderbolt: ignore non-updatable host controller (Fixes: #2373)

nvm_version will not export in Intel USB4 host controller running in
SW CM, there is no point in fwupd displaying a device for it.
closed 2020-09-10 by superm1