fwupd/fwupd #2374: thunderbolt update for W19TB does not complete

djcampello, 2020-09-10:
```
localhost /sys/bus/thunderbolt/devices/0-3 # fwupdtool update     
Decompressing?           [***************************************]
Decompressing?           [***************************************]
Loading?                 [***************************************]
Thunderbolt controller in Dell dock must remain connected for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Decompressing?           [***************************************]
Installing on Package level of Dell dock?                        ]
Restarting device?       [***************************************]
Installing on VMM5331 in Dell dock?                              ]
Restarting device?       [***************************************] Less than one minute remaining?
Installing on Thunderbolt controller in Dell dock?               ]
Writing?                 [***************************************]
Installing on WD19TB?    [ -                                     ]
Writing?                 [***************************************]
Updates will complete after the dock is unplugged.  The dock will be unavailable for approximately 1 minute.
? Package level of Dell dock has the latest available firmware version
? RTS5413 in Dell dock has the latest available firmware version
? RTS5487 in Dell dock has the latest available firmware version
? VMM5331 in Dell dock has the latest available firmware version
? WD19TB has the latest available firmware version
localhost /sys/bus/thunderbolt/devices/0-3 # fwupdtool get-updates
Decompressing?           [***************************************]
Decompressing?           [***************************************]
Loading?                 [***************************************]
??Thunderbolt controller in Dell dock:
  ?   Device ID:           201af6dc56a22742de828b6bab212f7d3e29254c
  ?   Summary:             Thunderbolt controller
  ?   Current version:     40.00
  ?   Vendor:              Dell Inc. (TBT:0x00D4)
  ?   Install Duration:    22 seconds
  ?   GUIDs:               48711a14-0aaa-55b8-a17f-f7c494b318c6 ? THUNDERBOLT\VEN_00D4&DEV_B070&REV_00
  ?                        08a8c886-2818-544c-b1a3-588eb07ae487 ? THUNDERBOLT\VEN_00D4&DEV_B070
  ?                        c94770ca-1773-592c-b20a-e87243bc7cd0 ? TBT-00d4b070
  ?   Device Flags:        ? Updatable
  ?                        ? Requires AC power
  ?                        ? Supported on remote server
  ?                        ? Device update needs activation
  ?                        ? Device stages updates
  ?                        ? Device is usable for the duration of the update
  ? 
  ??Thunderbolt controller Update:
        New version:       43.00
        Remote ID:         vendor-directory
        Summary:           Update for the Thunderbolt controller within a Dell dock
        License:           Proprietary
        Created:           2019-10-23
        Urgency:           High
        Vendor:            Dell Inc.
        Duration:          22 seconds
        Flags:             is-upgrade
```

fwupdtool activate or unplug/replug has no effect.

Doing a manual ``echo 1 > nvm_authenticate`` successfully applies the update

superm1, 2020-09-10:
Are the kernel patches for authenticate on disconnect backported in this SUT?

superm1, 2020-09-10:
I'm guessing that's a yes since I see https://chromium-review.googlesource.com/c/chromiumos/third_party/kernel/+/2314782 as merged and presume you're running a recent enough build.  Is it still possible to put the SUT into FW CM?  I am wondering if the access of the register for force power isn't working properly in SW CM.

@westeri is that a reasonable possibility here?

superm1, 2020-09-10:
From my side I just double checked with fwupd master and kernel 5.9-rc4 w/ a Titan Ridge host controller (from an XPS 7390) and had no problems.
Command line output:
[test_run.txt](https://github.com/fwupd/fwupd/files/5198884/test_run.txt)

In addition to checking with FW CM (if possible), can you please share --verbose output from the full test run?

superm1, 2020-09-10:
Also I double checked on a TGL system (integrated USB4 w/ FW CM), using same dock.  I downgraded to NVM 40, and then pulled updated package from lvfs-testing and ran exact same check and got same success result that upgrade finished on unplug.

westeri, 2020-09-10:
The LC power trick is done on the device side so I don't think it matters which CM (FW/SW) is used.

superm1, 2020-09-10:
I knew it was device side, but I was wondering if being in different CM meant that actual access to that TB switch in the remote device could be failing.  If that's not a possibility, @djcampello is there anything else that has landed in the chrome kernel tree in `drivers/thundrebolt` that's **not** in 5.9-rc4 and could be impacting this?  A full verbose log from fwupd would be really helpful (feel free to santitize for any codename data or share privately if necessary).

westeri, 2020-09-10:
Does the issue reproduce if you only use the kernel interface? Also is there some firmware for the WD19TB dock that I could try? I can try with both FW and SW CM if it reproduces.

superm1, 2020-09-10:
>Does the issue reproduce if you only use the kernel interface?

I think specifically it won't be obvious how to do the update on disconnect here (it needs an event sent to Dell's dock EC from fwupd currently).   The easiest you can get is probably to run fwupd's flow first, and then run the kernel flow manually (a second time).

>Also is there some firmware for the WD19TB dock that I could try? I can try with both FW and SW CM if it reproduces.

@westeri you can go between https://fwupd.org/downloads/e66d7258d272c78ee46a8b1770474d046a46df8e-WD19FirmwareUpdateLinux_01.00.04.cab and https://fwupd.org/downloads/c05bacfd8f73f30812559f14245b92a069c680caf300e961c78e00c985efe3e0-WD19FirmwareUpdateLinux_01.00.14.cab which are both available from https://fwupd.org/lvfs/devices/com.dell.wd198ceeeffd.firmware

You can use `gcab` to extract them if you don't use fwupd and want to go kernel interface directly.  Should have NVM 40 and NVM 43.

djcampello, 2020-09-10:
Something to note. If the device is not authorized I believe the dell-dock plugin wont be engaged to set up the update on disconnect. On my test I had issues with both authorized and non authorized cases

superm1, 2020-09-10:
That's correct - there is an assumption on the device being authorized.  Perhaps we should be safeguarding against that and impacts it can have by not let fwupd update a device missing the authorized attribute.

djcampello, 2020-09-11:
I tried again with the patch you referenced and seems to be working now:

```
localhost ~ # echo 1 > /sys/bus/thunderbolt/devices/0-3/authorized
localhost ~ # fwupdtool get-updates
Decompressing?           [***************************************]
Loading?                 [***************************************]
? RTS5413 in Dell dock has the latest available firmware version
? RTS5487 in Dell dock has the latest available firmware version
? VMM5331 in Dell dock has the latest available firmware version
? WD19TB has the latest available firmware version
?
??Thunderbolt controller in Dell dock:
? ?   Device ID:           201af6dc56a22742de828b6bab212f7d3e29254c
? ?   Summary:             Thunderbolt controller
? ?   Current version:     40.00
? ?   Vendor:              Dell Inc. (TBT:0x00D4)
? ?   Install Duration:    22 seconds
? ?   GUIDs:               48711a14-0aaa-55b8-a17f-f7c494b318c6 ? THUNDERBOLT\VEN_00D4&DEV_B070&REV_00
? ?                        08a8c886-2818-544c-b1a3-588eb07ae487 ? THUNDERBOLT\VEN_00D4&DEV_B070
? ?                        c94770ca-1773-592c-b20a-e87243bc7cd0 ? TBT-00d4b070
? ?   Device Flags:        ? Updatable
? ?                        ? Requires AC power
? ?                        ? Supported on remote server
? ?                        ? Device update needs activation
? ?                        ? Device stages updates
? ?                        ? Device is usable for the duration of the update
? ? 
? ??Thunderbolt controller Update:
?       New version:       43.00
?       Remote ID:         vendor-directory
?       Summary:           Update for the Thunderbolt controller within a Dell dock
?       License:           Proprietary
?       Created:           2019-10-23
?       Urgency:           High
?       Vendor:            Dell Inc.
?       Duration:          22 seconds
?       Flags:             is-upgrade
?     
??Package level of Dell dock:
  ?   Device ID:           073624c16dd99abe01ba1da223a70321e4f29beb
  ?   Summary:             A representation of dock update status
  ?   Current version:     01.00.13.00
  ?   Vendor:              Dell Inc. (USB:0x413C)
  ?   Install Duration:    5 seconds
  ?   GUID:                8ceeeffd-51b6-580c-9b75-69143227aff8 ? USB\VID_413C&PID_B06E&hub&status
  ?   Device Flags:        ? Updatable
  ?                        ? Supported on remote server
  ?                        ? Device can recover flash failures
  ?                        ? Device is usable for the duration of the update
  ? 
  ??Dell WD19/WD19DC/WDTB Update:
        New version:       01.00.13.01
        Remote ID:         vendor-directory
        Summary:           Update package for the WD19 family
        License:           Proprietary
        Created:           2019-10-23
        Urgency:           High
        Vendor:            Dell Inc.
        Duration:          5 seconds
        Flags:             is-upgrade
      
localhost ~ # fwupdtool update     
Decompressing?           [***************************************]
Loading?                 [***************************************]
Thunderbolt controller in Dell dock must remain connected for the duration of the update to avoid damage. Continue with update? [Y|n]: 
Decompressing?           [***************************************]
Installing on Package level of Dell dock?                        ]
Restarting device?       [***************************************]
Installing on Thunderbolt controller in Dell dock?               ]
[... 29 lines cut ...]
        Summary:           Update for the Thunderbolt controller within a Dell dock
        License:           Proprietary
        Created:           2019-10-23
        Urgency:           High
        Vendor:            Dell Inc.
        Duration:          22 seconds
        Flags:             is-upgrade
```
Note that the update is still available. Can we mark it as pending activation to no re-trigger it on another fwupdtool update?
Here I unplugged and re-plugged:

```      
localhost ~ # fwupdtool get-updates
Decompressing?           [***************************************]
Loading?                 [***************************************]
localhost ~ # echo 1 > /sys/bus/thunderbolt/devices/0-3/authorized
localhost ~ # fwupdtool get-updates   
Decompressing?           [***************************************]
Loading?                 [***************************************]
? Thunderbolt controller in Dell dock has the latest available firmware version
? Package level of Dell dock has the latest available firmware version
? RTS5413 in Dell dock has the latest available firmware version
? RTS5487 in Dell dock has the latest available firmware version
? VMM5331 in Dell dock has the latest available firmware version
? WD19TB has the latest available firmware version

superm1, 2020-09-11:
>I tried again with the patch you referenced and seems to be working now:

Well glad it worked now.

> Note that the update is still available. Can we mark it as pending activation to no re-trigger it on another fwupdtool update?

Hmm, this is a situation that with the daemon works properly because it stays in memory but we don't flush anything to the history database about it, but I'm not sure between fwupdtool runs how it can be accomplished.  @hughsie thoughts?

djcampello, 2020-09-11:
Experimenting with the latest patch I found out that fwupdtool activate fails to apply the update (i.e., authenticate). On the other hand, unplugging (without an activate in between) seems to work fine

djcampello, 2020-09-11:
I commented a little bit to late. Can we reopen this issue?

superm1, 2020-09-11:
Can you please show me the specific message you get in this failure circumstance?  I'm reproducing a failure in the same way, but I am concerned it's actually a stale history database and not the same failure as you.

djcampello, 2020-09-11:
I don't think I get a failure at all: the device does reset is just that it comes back with the same version. Also I just noticed that if using the install command the first attempt does not take the composite device but only the thunderbolt device (something wrong with plugin priority maybe?)

```
localhost ~ # fwupdtool install /usr/share/fwupd/remotes.d/vendor/firmware/e66d7258d272c78ee46a8b1770474d046a46df8e-WD19FirmwareUpdateLinux_01.00.04.cab --allow-reinstall --allow-older
Decompressing?           [***************************************]
Decompressing?           [***************************************]
Loading?                 [***************************************]
Decompressing?           [***************************************]
Installing on Thunderbolt controller in Dell dock?               ]
Writing?                 [***************************************]
localhost ~ # fwupdtool install /usr/share/fwupd/remotes.d/vendor/firmware/e66d7258d272c78ee46a8b1770474d046a46df8e-WD19FirmwareUpdateLinux_01.00.04.cab --allow-reinstall --allow-older
Decompressing?           [***************************************]
Decompressing?           [***************************************]
Loading?                 [***************************************]
Decompressing?           [***************************************]
Installing on Package level of Dell dock?                        ]
Restarting device?       [***************************************]
Installing on VMM5331 in Dell dock?                              ]
Restarting device?       [***************************************] Less than one minute remaining?
Installing on RTS5413 in Dell dock?                              ]
Restarting device?       [***************************************] Less than one minute remaining?
Installing on RTS5487 in Dell dock?                    /         ]
Restarting device?       [***************************************]
Installing on Thunderbolt controller in Dell dock?          -    ]
Writing?                 [***************************************]
Installing on WD19TB?    [ -                                     ]
Writing?                 [***************************************]
Updates will complete after the dock is unplugged. The dock will be unavailable for approximately 1 minute.
```

on activate I do not see any errors:

```
...
19:28:49:0841 FuEngine             Activating Thunderbolt controller in Dell dock
19:28:49:0841 FuPlugin             running superclassed activate() on thunderbolt
...
19:28:49:0848 FuEngine             Activating WD19TB
19:28:49:0848 FuPlugin             running superclassed activate() on dell_dock
```

superm1, 2020-09-11:
OK that's definitely different then I was seeing, so my failure is something else.

With your failure, I have a suspicion it's timing between the activate of thunderbolt and dell-dock happening back to back.  Activation normally takes about 2 seconds and your two activations happened effectively back to back.
Here's my hypothesis:
* Running activate on thunderbolt starts activation process on the controller (writes 1 into kernel sysfs)
* Running activate on dell-dock tries to start activation process on dock HW.  The first call from kernel hasn't finished, but this one fails, and then dell-dock reboots the dock.

To prove that hypothesis, can you instead try to send activate only to the device ID corresponding to the Thunderbolt controller?
`fwupdtool activate $ID`

superm1, 2020-09-11:
If that trial works, then this may be the solution.  It will prevent putting the thunderbolt plugin into needs activation, and only let activation happen from dell-dock.

```
diff --git a/plugins/thunderbolt/fu-thunderbolt-device.c b/plugins/thunderbolt/fu-thunderbolt-device.c
index de106aac..7db73a1a 100644
--- a/plugins/thunderbolt/fu-thunderbolt-device.c
+++ b/plugins/thunderbolt/fu-thunderbolt-device.c
@@ -742,7 +742,6 @@ fu_thunderbolt_device_write_firmware (FuDevice *device,
        /* using an active delayed activation flow later (either shutdown or another plugin) */
        if (fu_device_has_flag (device, FWUPD_DEVICE_FLAG_SKIPS_RESTART)) {
                g_debug ("Skipping Thunderbolt reset per quirk request");
-               fu_device_add_flag (device, FWUPD_DEVICE_FLAG_NEEDS_ACTIVATION);
                return TRUE;
        }
```

superm1, 2020-09-11:
Checked and elsewhere dell-dock was looking for that.  So that patch won't do it.

djcampello, 2020-09-11:
fwupdtool activate on the controller does not seem to be doing anything:

```
localhost ~ # fwupdtool activate 201af6dc56a22742de828b6bab212f7d3e29254c
Decompressing?           [***************************************]
Decompressing?           [***************************************]
Decompressing?           [***************************************]
Loading?                 [***************************************]
Activating firmware update Thunderbolt controller in Dell dock?
```

But the device does not reset and keeps reporting the old version

I am looking at the commit message for https://github.com/fwupd/fwupd/commit/834b28009d7e8fab9d37621456973a51a521c284 and I'm wondering if the following is problematic:

```
2 > nvm_authenticate_on_disconnect
1 > nvm_authenticate
```

djcampello, 2020-09-11:
never mind my last comment cause given the first comment a manual write to nvm_authenticate works

superm1, 2020-09-11:
I think that's comment above is actually on the right train of thought.

Writing 1 to "nvm_authenticate_on_disconnect" only sets up force power for an external activation, doesn't actually do anything.
So the vfunc to call activate was just writing force power again, which was done already anyways and behaved like a no-op at best.

Making it explicit should sort this.  PR incoming.

superm1, 2020-09-14:
@djcampello I think this is complete now with what's in master and we can close right?  

Given the most recent discussion in the PR what was happening with your failures in the first place was a failed activation call that was throwing off the authenticate on disconnect functionality, which explains why I wasn't seeing the original failure.

djcampello, 2020-09-14:
Yes, this should be closed now. Thanks for the help here!

PR #2385 (2020-09-11): dell-dock: mirror updatable flag into thunderbolt
This makes sure that if an update is pending (as stored in the EC), Thunderbolt
won't show in a needing update state again.

Fixes: #2374

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 6a1a49ea (2020-09-11): dell-dock: mirror updatable flag into thunderbolt

This makes sure that if an update is pending (as stored in the EC), Thunderbolt
won't show in a needing update state again.

Fixes: #2374
closed 2020-09-14 by djcampello