fwupd/fwupd #2376: vli: Hyper+Lenovo 07 TR Gen2

memily, 2020-09-10:
**Describe the bug**
We’re soooo, very, terribly, sorry…
 
As a sanity check we went back and tried to update/downgrade Proj 07, Lenovo Travel Gen2 (VL817+VL103Q4) and the BB/VL103 PD is not updating on that device either.  In theory the BB lives inside the VL817 hub and can only be accessed by the hub. Going along that assumption, fu-vli-usbhub-device.c, function “fu_vli_usbhub_device_update_v2()” should turn over control to "fu-vli-usbhub-pd-device.c" for writing pd fw, we think? Also hd2_fw_addr and hd2_fw_offset seem to be returning the wrong values for PD addr and offset (0x9000 and 0x2000). 

**Steps to Reproduce**
Attempt to fwupdtool update/downgrade install Lenovo TR Gen2 hub for sanity check, with manual power cycling. (Should we add flag "skips-restart" for PD child in quirk?)

**Expected behavior**
Hub+PD update...

**fwupd version information**
1.4.1 (dnf, most recent pull)

**fwupd device information**
Included updated hyper verbose log for reference as needed since it was same as TR Gen2.
Sorry! >x<
[verbose.txt](https://github.com/fwupd/fwupd/files/5199187/verbose.txt)

hughsie, 2020-09-10:
> fwupd version information
> 1.4.1

Are you sure you're using 1.4.1? If so you're missing out on a lot of fixes to the VLI plugin:

    git diff 1.4.1..1.4.6 ../plugins/vli/ | diffstat
     README.md                      |    3 
     fu-vli-common.c                |    4 +
     fu-vli-common.h                |    1 
     fu-vli-device.c                |    1 
     fu-vli-pd-device.c             |  167 +++++++++++++++++++++++++++++-------------------
     fu-vli-usbhub-device.c         |  216 ++++++++++++++++++++++++++++++++------------------------------
     fu-vli-usbhub-device.h         |   21 ------
     fu-vli-usbhub-i2c-common.h     |   12 ---
     fu-vli-usbhub-i2c-device.h     |   19 -----
     fu-vli-usbhub-msp430-device.c  |  192 +++++++++++++++++++++++++++++++++++++------------------
     fu-vli-usbhub-msp430-device.h  |   19 +++++
     fu-vli-usbhub-pd-device.c      |   18 +++++
     fu-vli-usbhub-rtd21xx-device.c |  524 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
     fu-vli-usbhub-rtd21xx-device.h |   19 +++++
     meson.build                    |    3 
     vli-pd.quirk                   |    6 -
     vli-usbhub-lenovo.quirk        |   28 ++++++--
     17 files changed, 961 insertions(+), 292 deletions(-)

Can you try with 1.4.5 or 1.4.6 please? If that still fails, can you get the verbose log with the newer version please.

memily, 2020-09-11:
Follow-up: with fwupd 1.4.6, update/downgrade for both TR Gen2 PD and Hyper PD fail. I've attached the verbose log for hyper. Let me know if there's anything else I can help with.
[verbose.txt](https://github.com/fwupd/fwupd/files/5208264/verbose.txt)

hughsie, 2020-09-11:
> Also hd2_fw_addr and hd2_fw_offset seem to be returning the wrong values for PD addr and offset (0x9000 and 0x2000).

What should they be? At the moment the FwOffset https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-common.c#L212 and FwSize https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-common.c#L160 are hardcoded based on it being a VL103.

The `H2Hdr@0x1000` is read out of the existing firmware (and seems to pass a checksum test, so it's probably legitimate). Is it possible that fwupd did one update on the hardware and wrote out a valid, but incorrect HD2 header? The address we use for HD2 is calculated here: https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-usbhub-device.c#L929

memily, 2020-09-14:
> What should they be? At the moment the FwOffset https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-common.c#L212 and FwSize https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-common.c#L160 are hardcoded based on it being a VL103.
> 
The values listed for FwOffset and FwSize inside /plugins/vli/fu-vli-common.c are correct. 

> The `H2Hdr@0x1000` is read out of the existing firmware (and seems to pass a checksum test, so it's probably legitimate). Is it possible that fwupd did one update on the hardware and wrote out a [...]

We are still looking into the problem, will get back to you soon as we can. 

Point of clarification, what fwupd version is the master branch? We ninja compiled and got 1.4.1 client/daemon?

hughsie, 2020-09-14:
> what fwupd version is the master branch?

If you're on master, you should always get `1.5.0` but the stable 1_4_X branch is currently `1.4.7`

memily, 2020-09-16:
> If you're on master, you should always get `1.5.0` but the stable 1_4_X branch is currently `1.4.7`

[On master] With the most up-to-date libxmlb and libjcat, we tried rpm and dnf update/builddep to fwupd 1.4.6 and we were not able to ninja compile to version 1.4.6, we’re still getting 1.4.1? If we download the tar release of 1.4.6 we can compile to fwupdtool and fwupdmgr 1.4.6. It is odd and we’re not sure what we’re doing wrong.  

We think one of the issues might be in  https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-usbhub-device.c#L538 where the VL103 PD is set-up as a child. The flag "has-shared-spi-pd" takes precedent over parent-child tiering. In theory VL817 is the proxy, acting on behalf of VL103 PD (same tier, invisible PD). 

We are also not seeing fwupd call write_firmware at https://github.com/fwupd/fwupd/blob/master/plugins/vli/fu-vli-usbhub-pd-device.c#L151, only from fu-vli-usbhub-device.c, is this expected behavior? Thanks.

hughsie, 2020-09-18:
> where the VL103 PD is set-up as a child

So that adds the `VL103` PD as a child to the `VL817`, right? The former is created using `fu_vli_usbhub_pd_device_new()` -- which sets up `klass_device->write_firmware = fu_vli_usbhub_pd_device_write_firmware` and that uses `FuVliUsbhubDevice *parent` to write, for instance `fu_vli_device_spi_write (FU_VLI_DEVICE (parent)`

Can you share the `fwuptool` log of trying to install onto the VL103? either `install` or `install-blob` is fine. Also, the GUID for the install is supposed to be the GUID on the VL103 device, not the hub device.

memily, 2020-09-21:
Yes, that adds the VL103 PD as a child to the VL817. The bin files install with install-blob but the hyper cab file is not going into `fu_vli_usbhub_pd_device_write_firmware`. 

Log for Lenovo TR Gen2:
[verbose_P07_VL103.txt](https://github.com/fwupd/fwupd/files/5254753/verbose_P07_VL103.txt)

Log for Hyper: 
[verbose_hyper.txt](https://github.com/fwupd/fwupd/files/5254754/verbose_hyper.txt)

hughsie, 2020-09-22:
> Log for Lenovo TR Gen2:

memily, 2020-09-25:
>I only have the "Mini Dock" so if you can reproduce with that I'd appreciate it.

Sure, thank you fer waiting.

[verbose_MiniDock_VL103.txt](https://github.com/fwupd/fwupd/files/5281673/verbose_MiniDock_VL103.txt)

[Mini_Dock_cabs.zip](https://github.com/fwupd/fwupd/files/5281674/Mini_Dock_cabs.zip)

For Mini Dock we haven't changed the GUIDs at all, so yes U2 PD is `USB\VID_17EF&PID_721C&APP_12`; Mini Dock U3 Hub is `USB\VID_17EF&PID_3094`; Mini Dock U2 Hub is `USB\VID_17EF&PID_3095`.

For Hyper U3 Hub `USB\VID_2D01&PID_0385`, U2 Hub `USB\VID_2D01&PID_0382` and U2 PD `USB\VID_2D01&PID_0380`.

hughsie, 2020-09-28:
@memily I get this in the history log:

    └─VL103:
      │   Device ID:          fe008de085345975906d64be2af7cc99f36724ca
      │   Update Error:       device version not updated on success, 138.4.18.26 != 138.4.24.38

Is the VL103 firmware version actually that version?

    $ fwupdtool firmware-parse VL103_App5-R8A_04_18_26_20191206_LuxShare.bin vli-pd
    FuVliPdFirmware:
    Version:                138.4.24.38
    DeviceKind:             VL103
    VID:                    0x17ef
    PID:                    0x721c
      FuFirmwareImage:
      Data:                 0x8000

but I see:

    <release urgency="medium" version="138.4.18.26" date="2019-12-06">
    <checksum filename="VL103_App5-R8A_04_18_26_20191206_LuxShare.bin" target="content"/>

...in the metainfo.xml file...

memily, 2020-09-29:
The issue is related to number format and our mistype!
 
[new/old] in hex is: `8A.04.18.26/8A.04.17.26`
[new/old] in dec is: `138.4.24.38/138.4.23.38`

Fixed cab files attached. The `write_firmware` problem remained after testing.
[Lenovo_Mini Dock.zip](https://github.com/fwupd/fwupd/files/5298172/Lenovo_Mini.Dock.zip)

hughsie, 2020-09-29:
With that branch:

    $ ./src/fwupdmgr install /home/hughsie/Downloads/Lenovo-Mini_Dock_new.cab --allow-reinstall --allow-older
    Decompressing…           [***************************************]
    Authenticating…          [***************************************]
    Installing on Lenovo USB-C Mini Dock…                            ]
    Restarting device…       [***************************************]
    Installing on VL103…     [***************************************]
    Installing on Lenovo USB-C Mini Dock…****************************]
    Restarting device…       [***************************************]
    Restarting device…       [***************************************] Less than one minute remaining…
    Successfully installed firmware

Phew! That was a complicated issue to debug. Fingers crossed it fixes the issue for you.

memily, 2020-09-30:
>Phew! That was a complicated issue to debug. Fingers crossed it fixes the issue for you.

.....i come, bearing bad tidings.
PD will update to mini_dock_new fw version but the hub will not update to mini_dock_new fw version (`write_firmware` function is called correctly from `fu-vli-usbhub_pd_device.c`). For downgrade, neither will downgrade...... ʷᵒᵉ

Will continue testing and attached verbose_log, (we're going on chinese holiday): 
[verbose_MiniDock_VL103.txt](https://github.com/fwupd/fwupd/files/5303766/verbose_MiniDock_VL103.txt)

hughsie, 2020-10-01:
> Will continue testing and attached verbose_log

Ahh! You're using fwupdtool, I was using fwupd+fwupdmgr. I'll test with that too now.

hughsie, 2020-10-01:
So, testing here I managed to get VL211, VL103 and then VL817 updated, but then it failed on coming back from GPIO reset. My log tail:

    Restarting device…       [ -                                     ]10:20:14:0849 FuPluginVliUsbhub    using GPIO reset for da77984c82b59c6fc69516431f467fd9a8d39a7f
    10:20:14:0851 FuDeviceList         waiting 10000ms for replug
    Restarting device…       [  -                                    ]10:20:14:0918 FuEngine             auto-removing GUsbDevice
    10:20:14:0918 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    10:20:14:0918 FuEngine             auto-removing GUsbDevice
    10:20:14:0918 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    10:20:14:0926 FuEngine             auto-removing GUsbDevice
    10:20:14:0926 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    10:20:14:0926 FuEngine             auto-removing GUsbDevice
    10:20:14:0926 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    10:20:14:0929 FuEngine             auto-removing GUsbDevice
    10:20:14:0929 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    10:20:14:0929 FuEngine             auto-removing GUsbDevice
    10:20:14:0929 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    Restarting device…       [                    \                  ]10:20:15:0640 FuEngine             auto-removing GUsbDevice
    10:20:15:0640 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    10:20:15:0640 FuEngine             auto-removing GUsbDevice
    10:20:15:0640 FuDeviceList         waiting 10000ms for Lenovo USB-C Mini Dock device removal
    Restarting device…       [                   -                   ]10:20:24:0852 FuDeviceList         device did not replug
    10:20:24:0852 FuHistory            modifying device Lenovo USB-C Mini Dock [da77984c82b59c6fc69516431f467fd9a8d39a7f]
    10:20:24:0863 FuPlugin             composite_cleanup(dell_dock)
    10:20:24:0863 FuIdle               uninhibiting: update
    failed to wait for detach replug: device da77984c82b59c6fc69516431f467fd9a8d39a7f did not come back
    10:20:24:0870 FuPlugin             destroy(tpm_eventlog)

So you can see we did the GPIO reset which powered down the entire hub so that all the USB devices disappeared, but the hub really didn't turn back on! The kernel has:

    [271331.109526] usb 3-1: USB disconnect, device number 8
    [271331.109531] usb 3-1.4: USB disconnect, device number 9
    [271331.109534] usb 3-1.4.3: USB disconnect, device number 10
    [271331.161136] usb 3-1.5: USB disconnect, device number 11
    [271331.846830] usb 4-1: USB disconnect, device number 4
    [271331.846835] usb 4-1.1: USB disconnect, device number 5

and no USB enumeration at all! When I manually remove the usb-c plug and reinsert it, the device enumerates fine. If I "hack the system" and manually unplug and replug really fast for the last GPIO reset I can get all devices to update successfully every time.

My experience and yours kinda point to the GPIO reset being unreliable for some reason. I can do:

    sudo ./src/fwupdtool attach --verbose aed12149-2f7c-5ba0-aae7-05f5a7a21344

to manually reset the VL103 using the GPIO parent device, and that works every time. Is there some reason the device might take more than 10 seconds to "come back" after a GPIO reset, or is the GPIO reset rate-limited in some way? I'm wondering how the device managed to get itself in a state where a manual unplug+replug was required.

memily, 2020-10-06:
>So you can see we did the GPIO reset which powered down the entire hub so that all the USB devices disappeared, but the hub really didn't turn back on! The kernel has...

We need two points of clarification:
1.	When the USB device disappears; at what stage of the update process does it disappear? During or after update does the OS lose the device? We’re wondering if the problem is that the hub went into suspend/sleep mode. 
2.	Is the wait target device the VL817 hub guid? We're wondering if fwupd is unable to find VL817.  

>to manually reset the VL103 using the GPIO parent device, and that works every time. Is there some reason the device might take more than 10 seconds to "come back" after a GPIO reset, or is the GPIO reset rate-limited in some way?  

The GPIOB reset works exactly the same as a chip reset, so we're aboard the same HMS Baffled... Would it be possible to execute the GPIOB function on the device? No update, just the GPIOB command through fwupd? If the hub disappears after the command is issued then we can know for certain the problem is the GPIOB reset pin.

hughsie, 2020-10-06:
> When the USB device disappears; at what stage of the update process does it disappear

After the VL211, VL103 and VL817 had completed, i.e. the very last thing.

> Is the wait target device the VL817 hub guid

Yes, it's the proxy GUID.

> Would it be possible to execute the GPIOB function on the device

Yes, you can do the GPIO reset using "fwupdtool attach" for the VL807 -- but that works for me every time flawlessly. The only time it seems to not re-enumerate is after doing the three firmware updates. HMS Baffled indeed.

memily, 2020-10-07:
Just to be sure, my manager would like to know if updating and downgrading produces the same results as reinstalling (new->new or old->old)?

We’ve asked Lenovo to send us a Mini Dock to test. On our current board, with fwupd/fwupdmgr, VL817 Hub FW does not get written onto spi flash, only VL103 PD FW is written with the locker returning NULL for parent device after sector_erase in `fu_vli_usbhub_pd_device_write_firmware` function. Will keep you posted!
[verbose_MiniDock_VL103.txt](https://github.com/fwupd/fwupd/files/5339292/verbose_MiniDock_VL103.txt)

memily, 2020-10-13:
Running fwupd/fwupdmgr 1.5.0 install to downgrade [new->old] Mini Dock version 4.154 to 4.94; `write_firmware` is not called from fu-vli-usbhub-pd-device.c however `write_firmware` is called from when updating [old->new] Mini Dock from 4.94 to 4.154. The behavior is strange and repeats itself after carefully checking fw versions...

I've cut out the VL211 portion since the bug comes from the VL817 share spi section. 
[verbose_MiniDock_VL103.txt](https://github.com/fwupd/fwupd/files/5370689/verbose_MiniDock_VL103.txt)

memily, 2020-10-14:
**Describe the bug**
First an update: after `fwupdmgr install` update/downgrade we use VLI Tool to read back the contents of SPI Flash and can confirm that inside SPI Flash is the correct firmware expected after a update/downgrade so yes fwupd is erasing and writing as intended for both Hub and PD, GPIOB is also working as intended using the attach method. 

We noticed running `fwupdmgr get-devices` directly after `fwupdmgr install *.cab` that fwupdmgr returned incorrect, pre-install information. Not the actual result; firmware install success. It was only after power-cycling Mini Dock and "flushing" `fwupdmgr get-devices`, that the next iteration of `fwupdmgr get-devices` returned the correct fw version for Hub+PD [VL817/Vl103]. 

**Steps to Reproduce Incorrect FW Version Result**
1. fwupdmgr install *.cab
2. fwupdmgr get-devices 

**Steps to Reproduce Correct FW Version Result**
1. fwupdmgr install *.cab
2. manually remove device from port
3. fwupdmgr get-devices (make sure device is gone)
4. manually replug device into port
5. fwupdmgr get-devices

Do you by chance, know why this is happening? 

**fwupd version information**
Running fwupd/fwupdmgr 1.5.0 with Lenovo Mini Dock
Attached cab files for Mini Dock:
[08_Mini dock.zip](https://github.com/fwupd/fwupd/files/5377251/08_Mini.dock.zip)

hughsie, 2020-10-16:
> fwupdmgr returned incorrect, pre-install information

I think what's happening is that:

* FuVliUsbhubDevice creates the FuVliUsbhubPdDevice in fu_vli_usbhub_device_setup()
* We call fu_vli_usbhub_pd_device_write_firmware() on the FuVliUsbhubPdDevice
* Nothing ever calls back to fu_vli_usbhub_device_pd_setup() to reload the FuVliPdHdr `hdr` block

This is all wonky because the FuVliUsbhubDevice is responsible for reading out the block of SPI handled by the PD device. I'll have a think; it seems very odd that we add the GUIDs in probe given we have to have the parent device open.

SherlockChu219, 2020-10-19:
Hi Richard, today Emily tested Mini-dock (Hub1+PD, Hub2), it works fine. But in TR-Hub(Hub+PD),  seems PD isn't updated. We'll continue check tomorrow.
I have a suggestion for all dock/dongle: after FW updated, can we use "power-cycling" to reload new FW?  There are so many devices with different design & combination, power-cycling is the easiest way.

hughsie, 2020-10-19:
> But in TR-Hub(Hub+PD), seems PD isn't updated

Can you attach old and new.cab files please, and I can test here.

> can we use "power-cycling" to reload new FW

Sure, in same cases we do that already, e.g. in the GPIOB case.

memily, 2020-10-20:
Here are the TR_Gen2 hub cab files! (configuration reminder, 07_TR_Gen is: VL817+VL103 Share SPI)
[07_TR_Gen2.zip](https://github.com/fwupd/fwupd/files/5405757/07_TR_Gen2.zip)

hughsie, 2020-10-20:
> Here are the TR_Gen2 hub cab files

Doh, I thought you said TB_Gen2 -- sorry. I've got the TB3 gen2 dock, but I never got the TR-Hub. Is it this thing? https://support.lenovo.com/de/en/accessories/usb_c_travel_hub -- if so it must be the only Lenovo docking station thing I've _not_ got... I'd appreciate adding it into the test matrix if you can arrange for one or two to be sent to me in the UK.

In the meantime, can you attach the --verbose from the daemon when you're trying to install with the latest code from master, from old to new. Thanks.

hughsie, 2020-10-20:
> blue demo board sent to you that you can use for TR_Gen2 testing

Ahh, a green one perhaps? VL0430B perhaps? I'll try that now.

hughsie, 2020-10-20:
> VL0430B perhaps

Are you sure that's the same topology as the TR_Gen2? I see the VL103 enumerating on the USB bus (as a com.vli.pd)? With VL817_V2_BC_Apple2A_Q7_0464_Luxshare_TravelHub_20190731.bin installed on the VL817 I also see a shared-SPI VL103 with the protocol com.vli.usbhub -- is that expected?

memily, 2020-10-21:
>with the protocol com.vli.usbhub -- is that expected?

I’ve been meaning to ask you about VL103 appearing as both pd and usbhub for TR_Gen2 in the quirk files…

Originally, our logic for share spi is to ignore the vid/pid of the downport shared device (pd/hub) and let the up port hub handle all control transfer requests on behalf of the shared device. How should the quirk file be configured to reflect this?

SherlockChu219, 2020-10-21:
Hi, Richard, 
I recommend the rule for TR-Gen2 is "If fwupdmgr detects the ID = 17EF/721D, then update hub+pd FW".
(1) With our test board, VL0430B, VL817 has its own billboard device on bus. And VL103 is attached to hub's down-port, so, you might see VL103 on USB bus.
(2) With TR-Gen2 device, VL817 also has its own billboard device on bus. But VL103 is not attached to hub's down-port,  and you won't see VL103 on USB bus.
Therefore, I suggest the best rule is to only check the hub's ID. If fwupdmgr sees TR-Hub ID, then update the hub+pd fw together.

hughsie, 2020-10-21:
> I’ve been meaning to ask you about VL103 appearing as both pd and usbhub for TR_Gen2 in the quirk files…

The quirk files for "physical" devices (e.g. with USB D+D- connected) are all opt-in in fwupd; if the quirk entry that adds support for the device is removed then it won't be probed and it's as if the device isn't there. I think `VID_17EF&PID_721C` is unusual as it's being claimed once in vli-pd.quirk and once in vli-usbhub-lenovo.quirk; the latter with much more precision. I've removed the former generic match in https://github.com/fwupd/fwupd/commit/f6a765d251f3ba4e314eab03184e42afb1dfc6c6

hughsie, 2020-10-21:
Okay, I've got all this working now on the VL0430B (fake TR_Gen2) -- you need to change both `tier3.metainfo.xml` from `<value key="LVFS::UpdateProtocol">com.vli.pd</value>` to `<value key="LVFS::UpdateProtocol">com.vli.usbhub</value>` as you want the writes to go through the USB Hub. The other fix needed is 07326182ca95c5bc43f054082c024222b888d86c and the cosmetic fix was also 0d778dbd50901271e92ba79dab2dd504134bcee5 -- can you test with git master please.

memily, 2020-10-22:
> cosmetic fix was also 0d778db 

Can you please explain the purpose of this fix and how it relates to U2 recovery?

>-- can you test with git master please.

After initial testing with both fwupdtool/fwupdmgr we were able to upgrade/downgrade the Mini Dock and TR_Gen2/Hyper. Will do more testing in the following days --adding Hyper support-- but wanted to update you on the progress!

hughsie, 2020-10-22:
> >    cosmetic fix was also 0d778db
> Can you please explain the purpose of this fix

In fwupdmgr it looked like two devices when really it's one physical device. Setting the ParentGuid allows fwupd to set up the "tree" of devices in the right logical order. It's really just cosmetic.

> we were able to upgrade/downgrade the Mini Dock and TR_Gen2/Hyper

Great news!

hughsie, 2020-10-22:
> Great news!

Also, close this bug now perhaps?

memily, 2020-10-23:
>Also, close this bug now perhaps? 
Yes, we will PR some changes needed for Hyper later.

closed 2020-10-23 by hughsie