fwupd/fwupd #2377: Composite out-of-order updates fail when

hughsie, 2020-09-10:
Imagine this scenario:

    Thunderbolt dock
      - TBT controller
        - USB 3.1 HUB
          - Audio device

We then try to install firmware on the TBT controller and Audio device, one after another in a **composite transaction** with multiple firmware types in one .cab file. In all, three devices get removed, some multiple times. Lets say the remove delay on the Audio device is 10 seconds. The TBT takes 30 seconds to detach, write, attach the hub:

 * prepare TBT firmware
 * detach TBT controller
 * write firmware 100%
 * reboot controller (takes ~6 seconds) which causes the USB hub and audio device to "fall off the bus" and the 10s disconnection timeout starts for Audio and HUB
 * wait for the tbt controller to come back
 * move onto writing the audio device
 * fail, as it's not yet been reenumerated
 * a few seconds later, the audio device enumerates -- TOO LATE

The same thing happens if you generate a composite update for the HUB and Audio device; after installing to the hub, we wait for the hub to come back then try to install onto the audio device which is still yet to enumerate.

I'm not sure the best way to fix this with minimum extra code and magic.  I think the only thing we can do is enforce the remove delay of the children to be always bigger that the write time of the parent, e.g.:

 * Make the *effective* remove_delay for a device is a sum of the priv->remove_delay and the MAX(estimated write time of all the parents), although sometimes we don't know the detach+write+attach time, so to speak
 * Set the remove_delay to infinite for all the children when the parent is being updated -- which has the effect that they're never going to be removed if the parent doesn't come back
 * Just set the RemoveDelay of the Audio and HUB devices to be 1.5*(detach+write+attach time of TBT controller) in a quirk for the dock

What we also need to do (which I think is the bug we're seeing) is wait for the child to re-attach if it's in the removed state. Either:

 * Enforce that we do the grandchild -> child -> parent device to workaround the bug -- which is already done using `fu_main_install_task_sort_cb()` and the device order. Maybe this is just a case where we don't set the device order correctly? We do the order promotion dance (using FWUPD_DEVICE_FLAG_INSTALL_PARENT_FIRST) when using `fu_device_add_child()` but seemingly not when using `fu_device_set_parent()` from something like `ParentGuid`

* Manually wait for the removed device as the very first thing in `fu_engine_install()`

Other ideas (and sanity check) most welcome. Thanks!

superm1, 2020-09-10:
> I'm not sure the best way to fix this with minimum extra code and magic

This is exactly the reason for why
https://github.com/torvalds/linux/commit/1cb36293833766e048cba2026dd860687a2851d9#diff-64d48cd31028512e72f9431ab48a0600
and
https://github.com/torvalds/linux/commit/4b794f8066e84818c172c81024f1d61071f14710#diff-64d48cd31028512e72f9431ab48a0600
were done.  You don't want to have the thunderbolt controller going away during the wrong time.  If you control all pieces of the flow you can do what you want when you want.

I think hacks around timing are asking for more problems.  You really need something that is orchestrating this combination of devices.  My suggestion is to use that kernel code from the second commit and delay the Thunderbolt's activation step until the very end after everything else has finished.

hughsie, 2020-09-10:
>  You don't want to have the thunderbolt controller going away during the wrong time

Agree, but it's still going to happen with the USB hub and the Audio device.

> I think hacks around timing are asking for more problems.

I agree. I think I need to verify the topology of the dock is set up correctly, as it looks like `ParentGuid` actually does affect the `order` too.

superm1, 2020-09-10:
> Agree, but it's still going to happen with the USB hub and the Audio device.

With those I think parent relationship and controlling order of who installs first makes sense.

>I agree. I think I need to verify the topology of the dock is set up correctly, as it looks like ParentGuid actually does affect the order too.

Right.

>thunderbolt controller going away during the wrong time

As a thought to improve Thunderbolt generally, I had an idea.
1. We can introduce a new client flag `PROMPTS_ACTIVATION`.  
2. Add a new kernel version field to `thunderbolt.conf` to indicate minimum for delayed activation flow (default to 5.9)
3. Change `DelayedActivation` in `thunderbolt.conf` to `auto`, where it's only used if:
a. Kernel is new enough
b. Client supports prompting activation or device activates on disconnect

Then on older clients or older kernel you get current behavior, but newer clients/kernels you're able to delay the activation until after everything else finishes.  You can prompt the user to finish firmware installation, warning them the chain of devices connected will disappear momentarily.

superm1, 2020-09-14:
> Enforce that we do the grandchild -> child -> parent device to workaround the bug -- which is already done using fu_main_install_task_sort_cb() and the device order. Maybe this is just a case where [...]

So with the recent device re-ordering stuff that you did and landed to master, is this still actually problematic?

I think we should avoid a timing relationship if at all possible, and it would be better to wait for events between devices or to push off events (such as activate Thunderbolt at a later time, maybe as part of the composite transaction cleanup).

hughsie, 2020-09-14:
> is this still actually problematic

I think the issue is fixed, although it's not trivial to reproduce. I've asked the ODM to test with this new build.

closed 2020-09-14 by hughsie