fwupd/fwupd #2378: /usr/bin/fwupdtool refresh SEGV

joakim-tjernlund, 2020-09-10:
**Describe the bug**
fwupdtool refresh SEGV on master
This is on an older HP laptop  

**Steps to Reproduce**
See above

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
```
```
fwupdtool get-devices --show-all-devices
Loading…                 [-                                      ]
OUI quirk required, please see https://github.com/fwupd/fwupd/wiki/ATA-Disk:-OUI-Quirk-Required!
---
[DeviceInstanceId=OUI\00080d]
Vendor = Toshiba
VendorId = ATA:0x1179
---
Loading…                 [***************************************]
HP EliteBook Folio 1040 G2
│
├─HD Graphics 5500:
│     Device ID:          bbbf1ce3d1cf15550c3760b354592040292415bb
│     Current version:    09
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              19e1f63e-bef0-5582-b236-364069bcd0d5 ← PCI\VEN_8086&DEV_1616&REV_09
│                         64c0ea55-d861-5dd0-aa6b-bcc307f57ed4 ← PCI\VEN_8086&DEV_1616
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-5600U CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x0000002f
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         bcad68bd-c33a-56cd-a8a6-b9a78d62ee6e ← CPUID\PRO_0&FAM_06&MOD_3D
│                         a702f59f-10c2-5561-976c-b94c89cb42d8 ← CPUID\PRO_0&FAM_06&MOD_3D&STP_4
│     Device Flags:       • Internal device
│   
├─System Firmware:
│     Device ID:          e24dbdf2c162e9342bccb3d8114b2d2fbbd7a5ca
│     Current version:    M76 Ver. 01.16
│     Vendor:             Hewlett-Packard (DMI:Hewlett-Packard)
│     Update Error:       UEFI Capsule updates not available or enabled
│     GUID:               230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│     Device Flags:       • Internal device
│   
└─THNSFJ256GDNU A:
      Device ID:          7466579651466dd8b56d7a19479bed768440026d
      Summary:            ATA Drive
      Current version:    JYLA1102
      Vendor:             Toshiba (ATA:0x1179)
      Serial Number:      17BS11G8TJ4V
      GUIDs:              f260d559-ffd1-56ce-88a6-fc38c53d8e9d ← IDE\TOSHIBA_THNSFJ256GDNU_A_________________JYLA1102
                          9bc9dfb3-2789-5a46-afff-739966fddfd9 ← IDE\0TOSHIBA_THNSFJ256GDNU_A_________________
                          9d768a22-7625-501d-bfa7-7155287899e0 ← TOSHIBA THNSFJ256GDNU A
      Device Flags:       • Internal device
                          • Updatable
                          • Requires AC power
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update

```
BT:
```
[... 123 lines cut ...]
#18 0x00007f54f261ef6f in g_hash_table_remove_all_nodes (destruction=1, notify=1, hash_table=0x55652ed9e520) at /var/tmp/portage/dev-libs/glib-2.64.5/glib-2.64.5/glib/ghash.c:1461
        __func__ = "g_hash_table_unref"
#19 g_hash_table_unref (hash_table=0x55652ed9e520) at /var/tmp/portage/dev-libs/glib-2.64.5/glib-2.64.5/glib/ghash.c:1461
        __func__ = "g_hash_table_unref"
#20 0x000055652d33b8de in  ()
#21 0x00007f54f2721f7d in g_object_unref (_object=<optimized out>) at /var/tmp/portage/dev-libs/glib-2.64.5/glib-2.64.5/gobject/gobject.c:3499
        weak_locations = <optimized out>
        old_ref = <optimized out>
        __func__ = "g_object_unref"
        object = 0x55652edaad20 [FuPluginList]
        __func__ = "g_object_unref"
#22 g_object_unref (_object=0x55652edaad20) at /var/tmp/portage/dev-libs/glib-2.64.5/glib-2.64.5/gobject/gobject.c:3391
        object = 0x55652edaad20 [FuPluginList]
        __func__ = "g_object_unref"
#23 0x000055652d329fb8 in  ()
#24 0x00007f54f2721f7d in g_object_unref (_object=<optimized out>) at /var/tmp/portage/dev-libs/glib-2.64.5/glib-2.64.5/gobject/gobject.c:3499
        weak_locations = <optimized out>
        old_ref = <optimized out>
        __func__ = "g_object_unref"
        object = 0x55652edb50c0 [FuEngine]
        __func__ = "g_object_unref"
#25 g_object_unref (_object=0x55652edb50c0) at /var/tmp/portage/dev-libs/glib-2.64.5/glib-2.64.5/gobject/gobject.c:3391
        object = 0x55652edb50c0 [FuEngine]
        __func__ = "g_object_unref"
#26 0x000055652d31ef6f in main ()

hughsie, 2020-09-10:
I think this is fixed by https://github.com/fwupd/fwupd/commit/50ff31de5690a5fda0aae7d0113980b5ebd3e926 -- can you try master please.

joakim-tjernlund, 2020-09-10:
> I think this is fixed by [50ff31d](https://github.com/fwupd/fwupd/commit/50ff31de5690a5fda0aae7d0113980b5ebd3e926) -- can you try master please.

I already had that one in my master but I rebuild again just to make sure and now it work!
Oh, well

closed 2020-09-10 by joakim-tjernlund