fwupd/fwupd #2382: PCR0 Reconstruction Invalid on NUC8i7HVK with latest BIOS

dnoliver, 2020-09-10:
**Describe the bug**
A clear and concise description of what the bug is.

I have 2 Intel NUC8i7HVK.

NUC 1 is using BIOS HNKBLi70.86A.0059.2019.1112.1124
NUC 2 is using BIOS HNKBLi70.86A.0063.2020.0827.1309

On NUC 1, the PCR 0 Reconstruction Test working pass 
On NUC 2, with the latest CSME and BIOS, the PCR 0 Reconstruction Test fails. 

On NUC 2, The output of `fwupdtpmevlog` and `tpm2_pcrread` matches

```shell
[root@nuc-2 test]# /usr/bin/fwupdtpmevlog
PCR:                     BIOS (0)
Type:                    0x7
Description:             EV_S_CRTM_CONTENTS
ChecksumSha1:            df5c4d7fe143c64df14f0ec275214181a479d58d
ChecksumSha256:          874bd7ac29fa5b9b312965f0784970068dfae30fd04513c271944320060dd370
BlobStr:                 Boot Guard Measured S-CRTM.

PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            c42fedad268200cb1d15f97841c344e79dae3320
ChecksumSha256:          d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155
BlobStr:                 ..kT..U@..N....:

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            96dee9aa396512f51ee1934fb08362a1c28ee80e
ChecksumSha256:          4269e64c670917e3d31d09655128d44bdab44bb1e3ad7a0e5ad600aaacfc96fa
BlobStr:                 ..........M.....

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            9069ca78e7450a285173431b3e52c5c25299e473
ChecksumSha256:          df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            9069ca78e7450a285173431b3e52c5c25299e473
ChecksumSha256:          df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            9ee6e6793ccec7ea5e2aa191b8839098eefe7a3d
ChecksumSha256:          6e8e3c5af4180d56eedfa479e93a084d370b489e765d5e21f7bda014b37d9cf0
BlobStr:                 a.............+.................B.o.o.t.O.r.d.e.r.........

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            0c999e5a142c58b53944513c242950c5aba48afe
ChecksumSha256:          e54303084c3c601c3915341dcaa939d5117c55c79561011861389d2ba707f279
BlobStr:                 a.............+.................B.o.o.t.0.0.0.2.....b.R.e.d. .H.a.t. .E.n.t.e.r.p.r.i.s.e. .L.i.n.u.x.....*.......................6Cy.%C.<.+..7&....4.\.E.F.I.\.r.e.d.h.a.t.\.s.h.i.m.x.6.4...e.f.i.......

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            4f57daac28bbc4d4ba3067276f9b9acdb84a9db5
ChecksumSha256:          56b252247b9ea33fad9d76afe534efd831de39824ac6725cd97011bd4a6b64b0
BlobStr:                 a.............+.................B.o.o.t.0.0.0.1.....z.u.b.u.n.t.u........u...u7K...8^l........*......................W.`E..M.y...?4.....4.\.E.F.I.\.u.b.u.n.t.u.\.s.h.i.m.x.6.4...e.f.i.......

PCR:                     BIOS Configuration (1)
[... 678 lines cut ...]
│ │                        52d7b679-db28-5bf7-bd87-41d77aeec600 ← TPM\VEN_INTC&MOD_Intel&VER_2.0
│ │   Device Flags:        • Internal device
│ │ 
│ └─Event Log:
│       Device ID:         58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:              a25657fe-b5dc-5be0-8b78-8b9dfec678ff ← system-tpm-eventlog
│       Device Flags:      • Internal device
│     
└─Unifying Receiver:
      Device ID:           fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
      Summary:             A miniaturised USB wireless receiver
      Current version:     RQR12.10_B0032
      Bootloader Version:  BOT01.02_B0015
      Vendor:              USB:0x046D
      Install Duration:    30 seconds
      GUIDs:               9d131a0c-a606-580f-8eda-80587250b8d6
                           fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
                           279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
      Device Flags:        • Updatable
```

**Additional questions**
- Operating system and version: Red Hat 8.3 Beta (Server with GUI)
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2020-09-10:
Can you get `/usr/bin/fwupdtpmevlog` on NUC-1 too please. I'd be interesting to see the difference.

dnoliver, 2020-09-10:
```shell
[test@nuc-1 ~]$ sudo /usr/bin/fwupdtpmevlog 
[sudo] password for test: 
PCR:                     BIOS (0)
Type:                    0x7
Description:             EV_S_CRTM_CONTENTS
ChecksumSha1:            a3a2d04519a9a5072b51d9e7812a82b83087cc4a
ChecksumSha256:          2366f32f33126522046b5b813cece6773bf3793d7071bb5130ccb0c9662c86e3
BlobStr:                 Boot Guard Measured S-CRTM.

PCR:                     BIOS (0)
Type:                    0x8
Description:             EV_S_CRTM_VERSION
ChecksumSha1:            c42fedad268200cb1d15f97841c344e79dae3320
ChecksumSha256:          d4720b4009438213b803568017f903093f6bea8ab47d283db32b6eabedbbf155
BlobStr:                 ..kT..U@..N....:

PCR:                     BIOS (0)
Type:                    0x1
Description:             EV_POST_CODE
ChecksumSha1:            8e6cad8bbfe7ee2a54da444241f2e757e2f1bd55
ChecksumSha256:          ae582c06dac2f568b9b2666b0788520c99064eb1c61b815a695760185774822c
BlobStr:                 ..........N.....

PCR:                     BIOS (0)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            9069ca78e7450a285173431b3e52c5c25299e473
ChecksumSha256:          df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x4
Description:             EV_SEPARATOR
ChecksumSha1:            9069ca78e7450a285173431b3e52c5c25299e473
ChecksumSha256:          df3f619804a92fdb4057192dc43dd748ea778adc52bc498ce80524c014b81119

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            bf19272589f263db569ca5709dc2311acb5430b0
ChecksumSha256:          543694c15c59e63e18937f4eacfae7627f3bc4d647f46e1b6b34ae44dc3c996e
BlobStr:                 a.............+.................B.o.o.t.O.r.d.e.r.......

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            7bba7c3af7e22e717d1b338ed8b8497c6f62ac55
ChecksumSha256:          4dd76962a2e4f90ee1aa46ff06f75724a2a4f413fdb210e1ff0cc9c2d63c0195
BlobStr:                 a.............+.........v.......B.o.o.t.0.0.0.1.....b.F.e.d.o.r.a.....*......................|...R.K.....K.2....4.\.E.F.I.\.f.e.d.o.r.a.\.s.h.i.m.x.6.4...e.f.i.......

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            185f3892c0fa7cde7088640fc198fc730a4da248
ChecksumSha256:          1ba8cbdf1a8443e8fa383e1248da2cb7ccc5c9bb7c4ef55ac60de425553a6041
BlobStr:                 a.............+.........D.......B.o.o.t.0.0.0.0.......W.i.n.d.o.w.s. .B.o.o.t. .M.a.n.a.g.e.r........u...u7K...8^l........*.............. .......jR...&B.b...C.|....F.\.E.F.I.\.M.i.c.r.o.s.o.f.t.\.B.o.o.t.\.b.o.o.t.m.g.f.w...e.f.i.......WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...}.................

PCR:                     BIOS Configuration (1)
Type:                    0x80000002
Description:             EV_EFI_VARIABLE_BOOT
ChecksumSha1:            f49f79798aedf7e25587d48afb4bcb3dd5bb3d9f
ChecksumSha256:          24974a53718e325c7bc0878c7549f6bab8165e849437be6d9b6e0f56e3fa7b1a
BlobStr:                 a.............+.................B.o.o.t.0.0.0.2.....z.u.b.u.n.t.u........u...u7K...8^l........*.........................UV.A.....*......4.\.E.F.I.\.u.b.u.n.t.u.\.s.h.i.m.x.6.4...e.f.i.......

PCR:                     Option ROMs (2)
Type:                    0x80000004
Description:             EV_EFI_BOOT_SERVICES_DRIVER
ChecksumSha1:            83ff2c5a209d7209091d4d67001d818be3421208
ChecksumSha256:          0654b4c4024c1adeaa35d16a2755f0e4b6cb39d3953c9e49dffe70ce7b28165e
BlobStr:                 ...x....(...............4............A..........................X...................
[... 545 lines cut ...]
  0 : 0x27FC0217DCA78EED67C898A6203A7AE45547986FBBDA40C5C252888F1BFEBF92
  1 : 0x28C9D84E275A7636BFB5F15A5701E9FBCCCB0C246C24F707097395EDF56FD32A
  2 : 0x85F63D955975FBD1CA4E934B32B93BF93F4D685FCBB43FE580055169B757FD6E
  3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  4 : 0xDE07FDFFDFC989717E3CE36E2A3250D6A9EE685260429F7B81AAD127BA3222BA
  5 : 0x2FDD58246A80B5917AA4E514555C6C53B1C123C5A1D6F440E68CF617DD59AD3A
  6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
  7 : 0x4D994AF4E5962E262650CA24AFDA7318BF65B9EA13F56A273161A836B48E30AD
  8 : 0x1CBCD940C532B691BAA90EDB0F5F77D1A22EE2AD39653853D51EC0D9362E6F5B
  9 : 0xFC37ABA84F58E7FBC13EBB9569F6BCD25CDB937C034F04DCAF89F94DDD11D143
  10: 0x8696389E528C92764B97E38F25365AD8CE9C218CDDAF969AF2C8F0600865DB58
  11: 0x0000000000000000000000000000000000000000000000000000000000000000
  12: 0x0000000000000000000000000000000000000000000000000000000000000000
  13: 0x0000000000000000000000000000000000000000000000000000000000000000
  14: 0x4FC186D75A67D1DF674CBD29144CA354A518553DADB66A95E749AD3DAE15867F
  15: 0x0000000000000000000000000000000000000000000000000000000000000000
  16: 0x0000000000000000000000000000000000000000000000000000000000000000
  17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  18: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  19: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  20: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  21: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  22: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
  23: 0x0000000000000000000000000000000000000000000000000000000000000000
```

hughsie, 2020-09-10:
> On NUC 2, with the latest CSME and BIOS, the PCR 0 Reconstruction Test fails.

Are both NUCs both using a new fwupd version?

For NUC1:

    Reconstructed PCRs:
      PCR 0:                 SHA1(8f265a11add0d654b546f47458b9bca6447479b3)
      PCR 0:                 SHA256(27fc0217dca78eed67c898a6203a7ae45547986fbbda40c5c252888f1bfebf92)
    sha1:
      0 : 0x8F265A11ADD0D654B546F47458B9BCA6447479B3
    sha256:
      0 : 0x27FC0217DCA78EED67C898A6203A7AE45547986FBBDA40C5C252888F1BFEBF92

    Reconstructed PCRs:
  PCR 0:                 SHA1(be2c3d6c598ea130bc6b61f974ca70bf11694235)
  PCR 0:                 SHA256(0c0062898247f8fe3085960e5b0270e7667b6f7d4cae17a503950499d45b4116)
sha1:
  0 : 0xBE2C3D6C598EA130BC6B61F974CA70BF11694235
sha256:
  0 : 0x0C0062898247F8FE3085960E5B0270E7667B6F7D4CAE17A503950499D45B4116

dnoliver, 2020-09-10:
Running with the installed version instead the compiled one:

```shell
[test@nuc-1 ~]$ /usr/bin/fwupdmgr --version
client version: 1.5.0
compile-time dependency versions
        gusb:   0.3.4
        efivar: 37
daemon version: 1.5.0

[test@nuc-1 ~]$ /usr/bin/fwupdmgr security --force
Host Security ID: HSI:0 (v1.5.0)

HSI-1
✔ CSME manufacturing mode:       Locked
✔ CSME override:                 Locked
✔ Intel DCI debugger:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ UEFI dbx:                      Found
✘ CSME v0:11.8.60.3561:          Invalid

HSI-2
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel DCI debugger:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ UEFI secure boot:              Enabled
✔ fwupd plugins:                 Untainted

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

Host Security ID attributes uploaded successfully, thanks!
```

hughsie, 2020-09-10:
So it's a regression going from 1.4.1 to 1.5.0 but just on the new BIOS version? @superm1 any ideas?

superm1, 2020-09-10:
Err I interpreted the above as nuc 1 passes with old bios on both 1.4.1 and 1.5.0.
nuc 2 fails on 1.5.0 with new BIOS, so presumably BIOS regression?

upgrading nuc 1 to new BIOS should probably reproduce the failure there too.

hughsie, 2020-09-11:
> so presumably BIOS regression

Ahh, I guess that's the more straightforward interpretation.

> upgrading nuc 1 to new BIOS should probably reproduce the failure there too

And in a similar vein, downgrading nuc1 should fix it too. @dnoliver could it be plausibly be a BIOS regression?

dnoliver, 2020-09-11:
I can't downgrade the BIOS. 

https://downloadmirror.intel.com/29832/eng/HN_0063_ReleaseNotes.pdf

```shell
Known Errata:
• Due to the Intel® ME firmware update in BIOS version 0063, you can’t downgrade to version 0059 or earlier.
```

That is good right? :)

hughsie, 2020-09-11:
Can you upgrade the other one and see if it has the same failure? Although I guess that would mean we can't even go backwards :/

dnoliver, 2020-09-11:
Updated BIOS in NUC 1 and the PCR 0 Reconstruction test fails.

So, BIOS problem? 

```shell
[root@nuc-1 test]# /usr/bin/fwupdmgr security --force
Host Security ID: HSI:1 (v1.5.0)

HSI-1
✔ CSME manufacturing mode:       Locked
✔ CSME override:                 Locked
✔ CSME v0:11.8.79.3722:          Valid
✔ Intel DCI debugger:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found
✔ UEFI dbx:                      Found

HSI-2
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel DCI debugger:            Locked
✘ TPM PCR0 reconstruction:       Invalid

HSI-3
✔ Intel BootGuard error policy:  Valid
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ UEFI secure boot:              Enabled
✔ fwupd plugins:                 Untainted

The TPM PCR0 differs from reconstruction.
 » https://github.com/fwupd/fwupd/wiki/TPM-PCR0-differs-from-reconstruction

Host Security ID attributes uploaded successfully, thanks!
```

hughsie, 2020-09-12:
> So, BIOS problem?

I think that's most plausible.

dnoliver, 2020-09-15:
Some debugging results:

The relevant part is located in https://github.com/fwupd/fwupd/blob/master/plugins/tpm-eventlog/fu-plugin-tpm-eventlog.c#L88

```cpp
static void
fu_plugin_device_registered_uefi (FuPlugin *plugin, FuDevice *device)
{
	FuPluginData *data = fu_plugin_get_data (plugin);
	GPtrArray *checksums;

	/* only the system-firmware device gets checksums */
	checksums = fu_device_get_checksums (device);
	if (checksums->len == 0)
		return;
	data->has_uefi_device = TRUE;

	for (guint i = 0; i < checksums->len; i++) {
		const gchar *checksum = g_ptr_array_index (checksums, i);
		data->reconstructed = FALSE;
		for (guint j = 0; j < data->pcr0s->len; j++) {
			const gchar *checksum_tmp = g_ptr_array_index (data->pcr0s, j);
			/* skip unless same algorithm */
			if (strlen (checksum) != strlen (checksum_tmp))
				continue;
			if (g_strcmp0 (checksum, checksum_tmp) == 0) {
				data->reconstructed = TRUE;
				break;
			}
		}
		/* check at least one reconstruction for this algorithm */
		if (!data->reconstructed)
			return;
	}
}
```

After setting some breakpoints, I was able to see PCR 0 Reconstruction for SHA1 working fine (hitting `data->reconstructed = TRUE;`). 

Based on the comment `/* check at least one reconstruction for this algorithm */`, I infer that this test should pass if at least one reconstruction was correct. But the `if (!data->reconstructed)` statement ignores the successful SHA1 reconstruction and proceed to check SHA256 reconstruction. Not sure if this was the intention. Should all reconstruction match? or at least one?

The problem occurs when SHA256 reconstruction is tested. This are the values being compared:

```shell
checksum =     "0c62898247f8fe3085960e5b0270e7667b6f7d4cae17a503950499d45b4116"
checksum_tmp = "0c0062898247f8fe3085960e5b0270e7667b6f7d4cae17a503950499d45b4116"
```

`checksum` seems to be `checksum_tmp` with the second byte truncated. The content of that byte is `00`. Is it possible that something is truncating that because it is `00`?
Anyways, `checksum` is not a valid SHA256 digest, so the test fail because of the length comparison.

@superm1, which is the expectation of this test? PCR 0 Reconstruction should match for every algorithms, or for at least one?

Also, do you think it is plausible that something is truncating that `00`?

superm1, 2020-09-15:
Well great sleuth work, thank you very much.
The intention is that both reconstructions should work, so sha1 passing and sha256 failing should flag like it does.

As for the truncation, I guess we need to dig in a little more why that is happening. @hughsie any ideas?

superm1, 2020-09-15:
Can you go over to the UEFI device and confirm `fwupdmgr get-devices` or `fwupdtool get-devices` output?  You should see the checksums on the UEFI device.  are they broken there too?  That would at least narrow down if it was in transit between UEFI and TPM devices or it's higher up the stack that it got calculated wrong.

superm1, 2020-09-15:
(you might need to add `--verbose` to see it)
Shows up like this:

```
(fwupdmgr:293561): FuMain-DEBUG: 13:38:33.823: System Firmware
  DeviceId:             b6c08fb9e5384d9d101853cc1ca20cf0ce2df2e2
  Guid:                 cbe49cca-492b-93e8-b0f4-f693501f271b
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 40d34c18-f6d5-54db-b81d-0aaa600c0540
  Plugin:               uefi
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot|can-verify|usable-during-update|md-set-verfmt
  Checksum:             SHA1(757148a099cb676056c0e3b82ed82cdc2fc1144b)
  Checksum:             SHA256(36fb28dc5aa8a5d96fed549991f59cb55d83e318df26682acd0735675520c08b)
  Vendor:               Dell Inc.
  VendorId:             DMI:Dell Inc.
  Version:              1.6.0
  VersionLowest:        1.6.0
  VersionFormat:        dell-bios
  VersionRaw:           0x00010600
  VersionLowestRaw:     0x00010600
  Icon:                 computer
  Created:              2020-09-15
```

superm1, 2020-09-15:
Here's actually a better command that narrows it down a little further:
```
# sudo fwupdtool get-devices --daemon-verbose=FuPluginUefi
```

Should see something like this:
```
19:15:24:0460 FuPluginUefi         added PCR-00=757148a099cb676056c0e3b82ed82cdc2fc1144b
19:15:24:0460 FuPluginUefi         added PCR-00=36fb28dc5aa8a5d96fed549991f59cb55d83e318df26682acd0735675520c08b
19:15:24:0464 FuPluginUefi         SecureBoot is: Disabled
19:15:24:0464 FuPluginUefi         UX Capsule support : Enabled
```

dnoliver, 2020-09-15:
It is shown truncated as well 

```shell
[test@nuc-1 fwupd]$ sudo fwupdtool get-devices --daemon-verbose=FuPluginUefi
Loading…                 [-                                      ]
19:16:33:0253 FuPluginUefi         added PCR-00=be2c3d6c598ea130bc6b61f974ca70bf11694235
19:16:33:0253 FuPluginUefi         added PCR-00=0c62898247f8fe3085960e5b0270e7667b6f7d4cae17a503950499d45b4116
19:16:33:0258 FuPluginUefi         SecureBoot is: Disabled
19:16:33:0258 FuPluginUefi         UX Capsule support : Enabled
Loading…                 [***************************************]
```

superm1, 2020-09-15:
oh wait, maybe it's this:
https://github.com/fwupd/fwupd/blob/master/plugins/uefi/fu-uefi-pcrs.c#L170

Take that check out, does it fix it?

dnoliver, 2020-09-15:
```shell
[test@nuc-1 fwupd]$ sudo tpm2_pcrread sha1:0+sha256:0
sha1:
  0 : 0xBE2C3D6C598EA130BC6B61F974CA70BF11694235
sha256:
  0 : 0x0C0062898247F8FE3085960E5B0270E7667B6F7D4CAE17A503950499D45B4116
```

superm1, 2020-09-15:
If so the check was my fault https://github.com/fwupd/fwupd/commit/069449e4276c34ab539a11c5862a1ed4a945dd66, but I guess we should be checking explicitly for all 0's not individual bytes being zero. doh.

hughsie, 2020-09-15:
Ohh no.

			gint64 val = pcr_values->digests[i].buffer[j];
			if (val > 0)
				g_string_append_printf (str, "%02x", pcr_values->digests[i].buffer[j]);

dnoliver, 2020-09-15:
yes, that solves the issue

```shell
[test@nuc-1 fwupd]$ git diff
diff --git a/plugins/uefi/fu-uefi-pcrs.c b/plugins/uefi/fu-uefi-pcrs.c
index 9396c80a..b9691438 100644
--- a/plugins/uefi/fu-uefi-pcrs.c
+++ b/plugins/uefi/fu-uefi-pcrs.c
@@ -167,7 +167,7 @@ fu_uefi_pcrs_setup_tpm20 (FuUefiPcrs *self, GError **error)
                str = g_string_new (NULL);
                for (guint j = 0; j < pcr_values->digests[i].size; j++) {
                        gint64 val = pcr_values->digests[i].buffer[j];
-                       if (val > 0)
+                       //if (val > 0)
                                g_string_append_printf (str, "%02x", pcr_values->digests[i].buffer[j]);
                }
                if (str->len > 0) {

[test@nuc-1 fwupd]$ sudo ./build/src/fwupdtool security --forceLoading…                 [***************************************]Host Security ID: HSI:2+! (v1.5.0)
HSI-1
✔ CSME manufacturing mode:       Locked
✔ CSME override:                 Locked
✔ CSME v0:11.8.79.3722:          Valid
✔ Intel DCI debugger:            Disabled
✔ SPI BIOS region:               Locked
✔ SPI lock:                      Enabled
✔ SPI write:                     Disabled
✔ TPM v2.0:                      Found

HSI-2
✔ IOMMU:                         Enabled
✔ Intel BootGuard:               Enabled
✔ Intel BootGuard ACM protected: Valid
✔ Intel BootGuard OTP fuse:      Valid
✔ Intel BootGuard verified boot: Valid
✔ Intel DCI debugger:            Locked
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Intel BootGuard error policy:  Valid
✘ Intel CET Enabled:             Not supported
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✔ Intel SMAP:                    Enabled
✘ Encrypted RAM:                 Not supported

Runtime Suffix -U
✘ Firmware updates:              Not supported

Runtime Suffix -A
✘ Firmware attestation:          Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
✘ Linux kernel lockdown:         Disabled
✘ UEFI secure boot:              Disabled

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```

commit 88f0d422 (2020-09-15): uefi: correct a logic error in TPM PCR reading (Fixes: #2382)

069449e tried to avoid issues with all zero TPM PCR's, but caused
a problem with any byte was zero.
commit a3d85438 (2020-09-15): uefi: correct a logic error in TPM PCR reading (Fixes: #2382)

069449e tried to avoid issues with all zero TPM PCR's, but caused
a problem with any byte was zero.
commit 0d091eff (2020-09-15): uefi: correct a logic error in TPM PCR reading (Fixes: #2382)

069449e tried to avoid issues with all zero TPM PCR's, but caused
a problem with any byte was zero.
closed 2020-09-15 by superm1