fwupd/fwupd #2390: Transifex - please authorize join to team requests

p-bo, 2020-09-14:
Hello,
requested for joining to Czech team long ago, but nobody noticed - looking at state of other languages, I'm probably not alone. May I ask you to do it please?
(and maybe, due lack of capacities, to authorize manually, to allow joining automatically?)
Thanks
Pavel

hughsie, 2020-09-14:
The transifex project is set up as assigned to the freedesktop.org team -- and I agree this isn't working well as the messages to the team are not being processed. It seems most other software has moved to managing it's own team of translators -- which I'm not super-keen on doing if I can avoid it. Can you tell me your transifex username and I'll try to add you manually perhaps?

p-bo, 2020-09-14:
Thanks for insight (and intervention) - my transifex username is pavelb

piotrdrag, 2020-09-15:
So, I added Richard as a maintainer on Transifex but also I accepted Pavel to the team. :)

hughsie, 2020-09-15:
Thanks @piotrdrag, much appreciated.

closed 2020-09-15 by hughsie