fwupd/fwupd #2392: Mitigate thunderspy vulnerabilities?

hughsie, 2020-09-14:
A recent research paper ( https://thunderspy.io/ ) outlines a number of thunderbolt vulnerabilities which focus on host and attached device firmware attacks.

This particular RFE is specifically about host firmware verification.  If an attacker with physical access is able to modify the firmware on the thunderbolt firmware controller while the device is off or asleep, the system will continue to work with the assumption that the firmware is valid and working.

When uploading via 'software' methods (ie, dd if=firmware.bin of=/sys/bus/thunderbolt/devices/0-0/nvm_non_active0/nvmem), the firmware is verified (in hardware), however, it is not verified at power on, or resume from sleep.   This is the attack vector that maybe fwupd can close.

superm1, 2020-09-14:
The "active" NVM can't be read/dumped from the kernel interface.  So i'm not sure how fwupd can help to close this attack vector?

hughsie, 2020-09-15:
>  So i'm not sure how fwupd can help to close this attack vector

I know we can't dump the firmware, but could we ask the controller for a checksum of the firmware contents? Even as a RFE to Intel?

YehezkelShB, 2020-09-15:
The description I find in the linked site is:
> we demonstrate the ability to create arbitrary Thunderbolt device identities, clone user-authorized Thunderbolt devices, and finally obtain PCIe connectivity to perform DMA attacks. In addition, we show unauthenticated overriding of Security Level configurations, including the ability to disable Thunderbolt security entirely, and restoring Thunderbolt connectivity if the system is restricted to exclusively passing through USB and/or DisplayPort. We conclude with demonstrating the ability to permanently disable Thunderbolt security and block all future firmware updates

Sounds like nothing is new here. Everyone knew for a long time that BIOS setting for Security levels is easy to override, which I guess was part of the reason to push for IOMMU-based protection.

Watching the video, it is about changing the FW directly by attaching the required SPI programming device, but this sounds ridiculous to me. I'm almost positive that we know how to disable Security levels with much simpler methods that don't require opening the chassis. And anyway, if attacker has enough physical access to be able to open the chassis, attach SPI programmer and do all this process, the user security has deep issues that SW-based verification of the FW will never solve.

In addition, the change there is to the NVM "headers", which can't be read by SW (which is probably what @superm1 meant).

Another point about SW-based verification is that if the attacker already has this level of access, working around SW based solution sounds like non issue. The SW can be disabled as soon as the rough device is attached, and before it has any chance to verify the FW (and do what? It can't shutdown the already active PCI connection, as much as I understand).

Also, if the controller FW has been attacked, how can we trust the checksum returned from it? The attacker can change the FW to return the original checksum instead.

What am I missing?

hughsie, 2020-09-15:
> that SW-based verification of the FW will never solve

Right, but I was surprised there was no code signing requirement with a key baked into the tbt controller...

> how can we trust the checksum returned from it

Valid point.

> What am I missing?

I opened this bug based on a downstream report for RHEL, (as we've seen) as the people that know about this stuff hang out here, rather than in Red Hat bugzilla :)

YehezkelShB, 2020-09-15:
Ah, this part is funny:

> tcfp enables disabling Thunderbolt security covertly - without reflecting the new state in the BIOS

Sure, the BIOS doesn't read the current SL from FW, so it still "remembers" the previous setting. But the state is still reflected in SW. So it isn't that stealth as it may sounds.

YehezkelShB, 2020-09-15:
> I was surprised there was no code signing requirement with a key baked into the tbt controller...

There is. If the change was to the FW itself, it'd go into safe mode (or fails the authentication of the new FW written to non-active NVM). My understanding is that the Security level setting isn't part of the signing, as it can be changed (e.g. from BIOS). I'm not sure how a setting controlled by user can be protected by signing (at least not without adding a secure-boot like mechanism).

hughsie, 2020-09-15:
> But the state is still reflected in SW

Do you think we should we read the state from the hardware as part of the fwupd startup or HSI tests?

YehezkelShB, 2020-09-15:
> Do you think we should we read the state from the hardware as part of the fwupd startup or HSI tests?

You mean - to compare the security level state as reported by FW/SW (what we usually get already) and the one found in BIOS settings?
This is an interesting idea. But a mismatch may happen also on BIOS setting reset (e.g. replacing the battery?), so this is another point to consider. And I'm not sure if there is a standard way (even an ad-hoc one) to find this BIOS setting anyway.

hughsie, 2020-09-15:
> to compare the security level state as reported by FW/SW

That's interesting too (but also hard, as you say) -- but I was thinking more of making the HSI tests fail runtime (like Secure Boot being disabled) if the security state is degraded.

superm1, 2020-09-15:
> Another point about SW-based verification is that if the attacker already has this level of access, working around SW based solution sounds like non issue. The SW can be disabled as soon as the [...]

As a comprehensive story you're right - but a piece of a larger onion it's more difficult to misrepresent the entire onion in your attack.  I view the fwupd security stuff as a snapshot of the entire onion.

> That's interesting too (but also hard, as you say) -- but I was thinking more of making the HSI tests fail runtime (like Secure Boot being disabled) if the security state is degraded.

I don't really feel this is a useful thing to invest effort into as it's really only for "older" systems.  Thunderbolt 3 Security levels are a legacy thing now as the focus is on pre-boot IOMMU and OS runtime IOMMU.  Many systems with USB4 controllers won't even be offering security levels in BIOS setup.

YehezkelShB, 2020-09-15:
> I was thinking more of making the HSI tests fail runtime (like Secure Boot being disabled) if the security state is degraded.

Ah, you mean by saving the original security level somewhere?
And then allowing the user to force update it if the change was intentional?

> Thunderbolt 3 Security levels are a legacy thing now as the focus is on pre-boot IOMMU and OS runtime IOMMU. Many systems with USB4 controllers won't even be offering security levels in BIOS setup.

+1
Even Intel TBT3 integrated into the CPU (like Ice Lake, IIRC) don't have security levels and IOMMU is what must be used for security.

hughsie, 2020-09-15:
> by saving the original security level somewhere

closed 2020-10-15 by hughsie

djcampello, 2020-11-12:
> The "active" NVM can't be read/dumped from the kernel interface. So i'm not sure how fwupd can help to close this attack vector?

Can't you read the contents of /sys/bus/thunderbolt/devices/0-0/nvm_non_active0/nvmem ?

superm1, 2020-11-12:
For the non-active yes.  But the actively running NVM isn't present.

djcampello, 2020-11-12:
Sorry I mean to say /sys/bus/thunderbolt/devices/0-3/nvm_active1/nvmem I did a hexdump of that file on my system and got some data back

superm1, 2020-11-12:
>Sorry I mean to say /sys/bus/thunderbolt/devices/0-3/nvm_active1/nvmem I did a hexdump of that file on my system and got some data back

The headers are what you can't get though as @YehezkelShB corrected my poor wording that I apparently did twice :)

>In addition, the change there is to the NVM "headers", which can't be read by SW (which is probably what @superm1 meant).