fwupd/fwupd #2396: fwupdmgr get-devices doesn't list Dell WD15 on a non-Dell laptop

vsht, 2020-09-17:
**Describe the bug**
I have a Thinkpad T14 AMD connected to a Dell WD15 that drives two screens
and supplies the power via USB-C. Fedora 32 is installed in UEFI mode. When I run
`fwupdmgr get-devices`, I don't see WD15 in the list, although it does appear if 
I do the same thing on a Dell Latitude 5450.

I'm wondering if this is a bug or some intentional limitation that WD15 firmware can
be upgraded only when connected to a Dell machine.

**Steps to Reproduce**
Take any non-Dell laptop with a USB-C port, connect Dell WD15 dock to 
it and run `fwupdmgr get-devices`

**Expected behavior**
`fwupdmgr get-devices` should list WD15 irrespectively of which laptop it is connected to.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): via `dnf`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
20UES00L00
│
├─AMD Ryzen 7 PRO 4750U with Radeon Graphics:
│     Device ID:           4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:     0x8600104
│     Vendor:              AuthenticAMD
│     GUID:                b9a2dd81-159e-5537-a7db-e7101d164d3f
│     Device Flags:        • Internal device
│   
├─HFM256GDHTNG-8510B:
│     Device ID:           310f45f1f223064b5c16bf6dff31146755a64480
│     Zusammenfassung:     NVM Express Solid State Drive
│     Current version:     80020C00
│     Vendor:              SK hynix (NVME:0x1C5C)
│     GUIDs:               3df71ce6-ec3c-5902-8215-7980cddffb5a
│                          f00c5df1-5b04-56b0-abc4-ec8796b0eb0b
│                          e187179f-1bdc-5520-b557-da7978a0f742
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Supported on remote server
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Integrated Camera:
│     Device ID:           54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:     58.18
│     Vendor:              Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:               cd900697-9e98-53bb-9f13-fac06bc26f77
│                          f8d483f9-9284-51b6-9363-7e6d59c40a97
│     Device Flags:        • Updatable
│   
├─Samsung SSD 970 EVO Plus 1TB:
│     Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Zusammenfassung:     NVM Express Solid State Drive
│     Current version:     1B2QEXM7
│     Vendor:              Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:               6e54c992-d302-59ab-b454-2d26ddd63e6d
│                          47335265-a509-51f7-841e-1c94911af66b
│                          f05ecac9-5cc2-5f77-b23b-a08ad8879ada
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─System Firmware:
│     Device ID:           3ba7ea853f9613adbcbb9326dbe5f0decd677a67
│     Current version:     65541
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                9bfd8b34-080b-40a1-47b3-ca1efa289e6d
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           1c8d3b6c7cb08805789e84e05fc8f3b097237f1e
│     Current version:     1057227777
│     Vendor:              DMI:LENOVO
│     GUID:                3b7f92a6-36d0-49ae-a2d1-272f9a067274
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           a1ea2f49e601f346e9540921ee09153f62464d4d
│     Current version:     16916636
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                cc484fef-8758-41ce-b42f-f402f79881ac
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           0309fe7d84b0b8e838ad1b2f22bd153906121cfc
│     Current version:     65541
│     Minimum Version:     65541
│     Vendor:              DMI:LENOVO
│     GUID:                4bea12df-56e3-4cdb-97dd-f133768c9051
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:           209551b815300e242a5521f54b82e0452f6e81f0
      Current version:     22552
      Minimum Version:     1
      Vendor:              DMI:LENOVO
      GUID:                58ac1529-d706-459c-9f98-c711bb777410
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update

```

**Additional questions**
- Operating system and version: Fedora 32 running the latest kernel
- Have you tried rebooting? Yes
- Is this a regression? I don't know, it is the first time I'm using this combination of hardware.

superm1, 2020-09-17:
This is correct, the WD15 can only be updated from Dell laptops.  The follow on product (WD19) can be updated from any machine.

closed 2020-09-17 by superm1

vsht, 2020-09-17:
Ok, good to know. Is it the same thing on Windows 10?

superm1, 2020-09-17:
Correct