fwupd/fwupd #2409: Problem updating Thinkpad T14s Prometheus fingerprint reader

TaIos, 2020-09-21:
**Describe the bug**
Issue is with Thinkpad T14s (20UH001ACK; Ryzen 7) and Fedora 32.
Can't get fingerprint from Synaptic working.

#### System can't detect fingerprint reader
```
$ fprintd-enroll
list_devices failed: No devices available
```
#### However, it is there
```
$ lsusb | grep "Synaptics
Bus 006 Device 003: ID 06cb:00bd Synaptics, Inc. Prometheus MIS Touch Fingerprint Reader
```
```
$ fwupdmgr get-devices
20UH001ACK
│
├─Integrated Camera:
│     Device ID:           54f0d9041b6c5438c7ff825f5139559c5ca1b222
│     Current version:     58.18
│     Vendor:              Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     GUIDs:               08c6b922-0e9b-5283-9e1a-2f5a549abf8e
│                          95b07a8e-2063-5025-80b5-1fcf4ca8e9e3
│     Device Flags:        • Updatable
│   
├─Prometheus:
│     Device ID:           50efef912094e784a03628350441f6927c4920b0
│     Souhrn:              Fingerprint reader
│     Current version:     02.01.10
│     Vendor:              Synaptics (USB:0x06CB)
│     Install Duration:    2 seconds
│     Update State:        failed
│     Update Error:        error status: 0x5ff
│     Last modified:       2020-09-21 22:39
│     GUIDs:               09e8bf16-3e69-50f5-bb66-c7a040248352
│                          8088f861-6318-5b1e-9ce4-fbddbedb09ac
│     Device Flags:        • Updatable
│                          • Supported on remote server
│                          • Cryptographic hash verification is available
│   
├─System Firmware:
│     Device ID:           b4718caaf1dc79bb34f276c98081532f468822aa
│     Current version:     65545
│     Vendor:              LENOVO (DMI:LENOVO)
│     GUID:                428ba8ba-40c8-4446-8d7a-394d6c28b26a
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Cryptographic hash verification is available
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           1c8d3b6c7cb08805789e84e05fc8f3b097237f1e
│     Current version:     1057227777
│     Vendor:              DMI:LENOVO
│     GUID:                3b7f92a6-36d0-49ae-a2d1-272f9a067274
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           59a792a5a27c35676b54ee1a7d8822003a9d922d
│     Current version:     16842757
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                158688af-6978-4bc3-a8a1-837e09ac4ad9
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           8c011dde02d2b5a0f20fb3636d48e2f7d2e9ec8a
│     Current version:     0
│     Vendor:              DMI:LENOVO
│     GUID:                1181d859-c3be-4b1d-985a-b28dca6ba6d8
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           0309fe7d84b0b8e838ad1b2f22bd153906121cfc
│     Current version:     65545
│     Minimum Version:     65545
│     Vendor:              DMI:LENOVO
│     GUID:                4bea12df-56e3-4cdb-97dd-f133768c9051
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─UEFI Device Firmware:
│     Device ID:           a29eb08c484db785aca98502a9f6c45bb85c6937
│     Current version:     22552
│     Minimum Version:     1
│     Vendor:              DMI:LENOVO
│     GUID:                cfa38dd6-0165-4f00-ae3c-6207803cab0b
│     Device Flags:        • Internal device
│                          • Updatable
│                          • Requires AC power
│                          • Needs a reboot after installation
│                          • Device is usable for the duration of the update
│   
├─Unifying Receiver:
│     Device ID:           1c8c4cf505e4bd20d5aeb2c2652841d6546c9e15
│     Souhrn:              A miniaturised USB wireless receiver
│     Current version:     RQR24.11_B0036
│     Bootloader Version:  BOT03.02_B0009
│     Vendor:              USB:0x046D
│     Install Duration:    30 seconds
│     GUIDs:               cc4cbfa9-bf9d-540b-b92b-172ce31013c1
│                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f
│                          279ed287-3607-549e-bacc-f873bb9838c4
│     Device Flags:        • Updatable
│   
└─WDC PC SN730 SDBQNTY-512G-1001:
      Device ID:           71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Souhrn:              NVM Express Solid State Drive
      Current version:     11130101
      Vendor:              Sandisk Corp (NVME:0x15B7)
      GUIDs:               a39943dd-3afb-54f8-b110-c5a21f071200
                           fccbb6ea-e20e-58ad-bf8a-7fb7d43ff4c2
                           06b4e2aa-91af-508b-b06e-65e3b3189e97
      Device Flags:        • Internal device
                           • Updatable
                           • Requires AC power
                           • Supported on remote server
                           • Needs a reboot after installation
                           • Device is usable for the duration of the update
```

**Steps to Reproduce**
If I try to update and get the latest version (which hopefully will get fingerprint reader to work)
```
$ fwupdmgr refresh 
Successfully downloaded new metadata: 2 local devices supported
$ fwupdmgr update
• Integrated Camera has no available firmware updates
Upgrade available for Prometheus from 02.01.10 to 10.01.3158804
Prometheus and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
...
error status: 0x5ff
```

**Expected behavior**
Upgrade Prometheus from 02.01.10 to 10.01.3158804 and use fingerprint reader.

**fwupd version information**
Please provide the version of the daemon and client.
```shell
$ fwupdmgr --version
client version:	1.4.5
compile-time dependency versions
	gusb:	0.3.4
	efivar:	37
daemon version:	1.4.5
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `dnf`

**Additional questions**
- Operating system and version: 5.8.10-200.fc32.x86_64 (Fedora 32)
- BIOS: [R1CET40W](https://support.lenovo.com/be/en/downloads/ds544980/)
- notebook: Thinkpad T14s (20UH001ACK)

**Other**
* Solution from [here](https://www.reddit.com/r/Fedora/comments/f9hgz2/x1y4f31_just_got_the_fingerprint_sensor_to_work/) was also tried. It failed at step 2, with the same `error status: 0x5ff`.

hughsie, 2020-09-22:
@vhuag can you take a look at this please.

vhuag, 2020-09-23:
0x5ff means the sensor failed to program the flash.

hughsie, 2020-09-23:
@vhuag thanks; do you think this is a hardware error? Would it happen again if the user tried again, and in which case, should we retry an action in the plugin to work around this kind of failure?

I'd also very much appreciate it if you could add the missing enums to https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-prometheus/fu-synaprom-common.c#L16 so that we can report this to the user in a better way. Thanks!

vhuag, 2020-09-24:
It looks like a hardware error and I don't think it can be recovered after retry, actually @TaIos already tried the other solution which is just a retry.

hughsie, 2020-09-24:
@vhuag do you have any idea what might have caused the problem? e.g. invalid flash? I want to be sure it's nothing that fwupd did. I'd also like to see a better error string from fwupd in case this happens again.

vhuag, 2020-09-24:
Yes, invalid flash might cause this problem, and there is nothing we can do in fwupd.
I'll check if we can add some error strings in fwupd.
Thanks

TaIos, 2020-09-25:
I have changed operation system to Windows  10 Home. Windows is also able to detect fingerprint reader. However, if I open setting and then configure fingerprint, widnow freezes. The same happened after multiple restarts and system updates.

This suggests the problem is in the fingerprint itself, now the software. I am returning the notebook.

 @vhuag adding meaningful error message would be very helpful. I spent several hours because error `0x5ff` did not show up in search engines.

hughsie, 2020-10-17:
@vhuag are you able to add the error strings for the extra error codes please?

vhuag, 2020-10-19:
Yes, we'll add the string to the code. I think you can close this issue because this is not fwupd problem.

hughsie, 2020-10-19:
> I think you can close this issue because this is not fwupd problem.

hughsie, 2020-10-19:
I think you need to update https://github.com/fwupd/fwupd/blob/master/plugins/synaptics-prometheus/fu-synaprom-common.c -- no?

TaIos, 2020-10-19:
Issue was specific to the particular notebook. Today, I used the same model and `fwupd` run withou errors.

```
$ sudo fwupdmgr update

• Integrated Camera has no available firmware updates
Upgrade available for Prometheus from 10.01.2923304 to 10.01.3158804
Prometheus and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Prometheus…     [***************************************]
Restarting device…       [***************************************]
Successfully installed firmware
Upgrade available for Prometheus IOTA Config from 0014 to 0021
Prometheus IOTA Config and all connected devices may not be usable while updating. Continue with update? [Y|n]: Y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating Prometheus IOTA Config…*********************************]
Updating Prometheus…     [***                                    ]
Restarting device…       [***************************************]
Successfully installed firmware
• SAMSUNG MZVLB1T0HBLR-000L7 has no available firmware updates
• System Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• UEFI Device Firmware has no available firmware updates
• Unifying Receiver has no available firmware updates

```

closed 2020-11-26 by stale[bot]